davidusb-geek/emhass-add-on #1: Problem running dayahead-optim

piio, 2022-04-10:
Hi,

I'm trying to run EMHASS using Home Assistant add-on - it installs correctly, I can also open User Interface (using default 0.0.0.0 as web_ui_url - when I try ot use my Home Assistant dns name or IP it can't bind to socket)
I configured my own sensors for PV and load without vars and try to run Launch Day-ahead Optimalization - using web ui it try to connect to localhost which is wrong - It should be using relative URL.

I tried to manually run Day-ahead Optimalization by running directly in docker container emhass:
curl -X POST http://localhost:5000/action/dayahead-optim

I got 400 Bad request:

docker exec -it addon_564343f0_emhass bash
root@564343f0-emhass:/# curl -X POST http://localhost:5000/action/dayahead-optim
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
<title>400 Bad Request</title>
<h1>Bad Request</h1>
<p>The browser (or proxy) sent a request that this server could not understand.</p>

How to properly run dayahead?

Logs from add-on:
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[2022-04-09 23:31:30,237] ERROR in app_server: [{'period_hp_1': [{'start': '02:54'}, {'end': '15:24'}]}, {'period_hp_2': [{'start': ' 17:24'}, {'end': ' 20:24'}]}]
 * Serving Flask app 'app_server' (lazy loading)
 * Environment: development
 * Debug mode: off
 * Running on all addresses (0.0.0.0)
   WARNING: This is a development server. Do not use it in a production deployment.
 * Running on http://127.0.0.1:5000
 * Running on http://172.30.33.8:5000 (Press CTRL+C to quit)
192.168.12.10 - - [09/Apr/2022 23:31:30] "GET / HTTP/1.1" 200 -
192.168.12.10 - - [09/Apr/2022 23:31:31] "GET /static/style.css HTTP/1.1" 200 -
192.168.12.10 - - [09/Apr/2022 23:31:50] "GET /action/dayahead-optim HTTP/1.1" 405 -
192.168.12.10 - - [09/Apr/2022 23:32:09] "POST /action/dayahead-optim HTTP/1.1" 400 -
192.168.12.10 - - [09/Apr/2022 23:32:14] "GET /action/dayahead-optim HTTP/1.1" 405 -
192.168.12.10 - - [09/Apr/2022 23:32:19] "GET / HTTP/1.1" 200 -
192.168.12.10 - - [09/Apr/2022 23:32:19] "GET /static/style.css HTTP/1.1" 304 -
192.168.12.10 - - [09/Apr/2022 23:32:34] "GET / HTTP/1.1" 200 -
192.168.12.10 - - [09/Apr/2022 23:33:29] "OPTIONS /action/dayahead-optim HTTP/1.1" 200 -
192.168.12.10 - - [09/Apr/2022 23:33:45] "GET /action/dayahead-optim HTTP/1.1" 405 -
192.168.12.10 - - [09/Apr/2022 23:34:23] "GET / HTTP/1.1" 200 -
192.168.12.10 - - [09/Apr/2022 23:34:24] "GET /static/style.css HTTP/1.1" 200 -
192.168.12.10 - - [09/Apr/2022 23:34:25] "GET /favicon.ico HTTP/1.1" 404 -
192.168.12.10 - - [09/Apr/2022 23:35:01] "OPTIONS /action/dayahead-optim HTTP/1.1" 200 -
127.0.0.1 - - [10/Apr/2022 11:00:43] "POST /action/dayahead-optim HTTP/1.1" 400 -

davidusb-geek, 2022-04-10:
Hi, what is your configuration? Are you using Home Assistant OS? 
Use the latest version of EMHASS.
The `curl` command should be something like this:
```
curl -i -H "Content-Type: application/json" -X POST -d '{}' http://localhost:5000/action/dayahead-optim
```
You should match the url to that of your HA instance.
For the web ui, use the `web_ui_url` parameter in the configuration pane to match your HA instance url.

piio, 2022-04-10:
I'm using supervised install on Debian 11 bullseye, EMHASS version is 0.1.18

root@564343f0-emhass:/# curl -i -H "Content-Type: application/json" -X POST -d {} http://localhost:5000/action/dayahead-optim
HTTP/1.1 500 INTERNAL SERVER ERROR
Server: Werkzeug/2.1.1 Python/3.9.2
Date: Sun, 10 Apr 2022 18:36:52 GMT
Content-Type: text/html; charset=utf-8
Content-Length: 290

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
<title>500 Internal Server Error</title>
<h1>Internal Server Error</h1>
<p>The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.</p>

In log I see:

s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[2022-04-10 20:21:46,035] ERROR in app_server: [{'period_hp_1': [{'start': '02:54'}, {'end': '15:24'}]}, {'period_hp_2': [{'start': ' 17:24'}, {'end': ' 20:24'}]}]
 * Serving Flask app 'app_server' (lazy loading)
 * Environment: development
 * Debug mode: off
 * Running on all addresses (0.0.0.0)
   WARNING: This is a development server. Do not use it in a production deployment.
 * Running on http://127.0.0.1:5000
 * Running on http://172.30.33.8:5000 (Press CTRL+C to quit)
[2022-04-10 20:22:09,212] ERROR in retrieve_hass: The retrieved JSON is empty, check that correct day or variable names are passed
[2022-04-10 20:22:09,213] ERROR in app: Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/usr/src/app_server.py", line 143, in action_call
    input_data_dict = setUp(config_path, base_path, costfun, params, app.logger)
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 42, in setUp
    rh.get_data(days_list, var_list,
  File "/usr/local/lib/python3.9/dist-packages/emhass/retrieve_hass.py", line 129, in get_data
    self.df_final = pd.concat([self.df_final, df_day], axis=0)
UnboundLocalError: local variable 'df_day' referenced before assignment

davidusb-geek, 2022-04-10:
Ok this time the retrieved data from Home Assistant is an empty json.
I can see three cases where this will happen:

1. The HA url is not correct. I have tested this add-on in HA OS and not on a supervised install. Inside the add-on I'm using the internal supervisor connection to retrieve the data using this url "http://supervisor/core/api" and this token "$SUPERVISOR_TOKEN". So it is possible that with the supervised install this is not the correct way  to use the core API. I can't tell as I don't have a supervised install but I will be asking this question to the developers of HA. If you already managed to install HA with the supervised method I think that you should not be using this add-on and use the EMHASS Python package directly. Got here https://github.com/davidusb-geek/emhass and you will find all you need to run EMHASS as a simple bash script that can be launch from an automation.
2. The json will be empty if the names of the variables passed are not correct. So double check the names for your `sensor_power_photovoltaics` and `sensor_power_load_no_var_loads`.
3. It will also be empty if your are trying to retrieve data from data where there is no data. I guess that you will find this problem if these variables are not in your HA recorder for history values. So check that also.

If this doesn't solve the issue I really recommend to use EMHASS directly as a Python package on your Linux install. This is the way I use it and is somewhat more reliable as I have tested the package in depth and using CI.

piio, 2022-04-13:
Hi,

Using new version v0.1.19 and command-line curl from my host on which HA is running I'm getting valid response (I'm forwarding port from container to host)

curl -i -H "Content-Type: application/json" -X POST -d '{}' http://localhost:5000/action/dayahead-optim
HTTP/1.1 201 CREATED
Server: Werkzeug/2.1.1 Python/3.9.2
Date: Wed, 13 Apr 2022 06:50:38 GMT
Content-Type: text/html; charset=utf-8
Content-Length: 45

EMHASS >> Action dayahead-optim received.

I noticed 2 things:
1. If I try to set historic_days_to_retrieve larger number of days for sensor created calculating load without vars than it exists (I have 3 days in my recorder because I just created this sensor, and set historic_days_to_retrieve larger than 3), EMHASS is returning error 500 - maybe this issue can be catched and returned error which will clearly get information to user that there is no historical values in whole data range which is defined in configuration?

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
<title>500 Internal Server Error</title>
<h1>Internal Server Error</h1>
<p>The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.</p>

2. In file templates/index.html there is hard-coded url, which would never be working if using from web browser on another system - I suggest changing to relative URL - I tested it on my install and it's working correctly. I created pull request to change this.

davidusb-geek, 2022-04-13:
Hi,

I updated to v0.1.20 following the merge on the subject of your second remark and other small fixes.
Yes, for the first remark I should improve the error message for this. Effectively if `historic_days_to_retrieve` is larger than the recorded history of your sensor you will have this error.
The current message is "The retrieved JSON is empty, check that correct day or variable names are passed", but it is not explicit enough.

Could you please confirm if with the latest v0.1.20 and the correct `historic_days_to_retrieve` you're able to launch the optimization?

davidusb-geek, 2022-04-21:
Hi, these errors were solved with the latest releases. Closing this by now.

closed 2022-04-21 by davidusb-geek

piio, 2022-04-21:
I'm sorry, I could'nt manage to check this earlier.
Thank you for addressing this issues!