davidusb-geek/emhass-add-on #3: 0.1.24 500 Internal Server Error

purcell-lab, 2022-04-21:
Still having issues with a stable environment.

0.1.24 runs for a little while and then the web server starts returning a 500 Internal Server Error and it looks as if the environment hangs.

[image]

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
 * Serving Flask app 'app_server' (lazy loading)
 * Environment: development
 * Debug mode: off
 * Running on all addresses (0.0.0.0)
   WARNING: This is a development server. Do not use it in a production deployment.
 * Running on http://127.0.0.1:5000
 * Running on http://172.30.33.4:5000 (Press CTRL+C to quit)
192.168.86.50 - - [21/Apr/2022 09:55:18] "GET / HTTP/1.1" 200 -
192.168.86.50 - - [21/Apr/2022 09:55:18] "GET /static/style.css HTTP/1.1" 304 -
[2022-04-21 09:55:58,382] WARNING in optimization: Failed LP solve with PULP_CBC_CMD solver, falling back to default Pulp
[2022-04-21 09:55:58,406] WARNING in optimization: Failed LP solve with default Pulp solver, falling back to GLPK_CMD
192.168.86.50 - - [21/Apr/2022 09:56:00] "POST /action/dayahead-optim HTTP/1.1" 200 -
192.168.86.50 - - [21/Apr/2022 09:56:32] "GET / HTTP/1.1" 200 -
192.168.86.50 - - [21/Apr/2022 09:56:32] "GET /static/style.css HTTP/1.1" 304 -
/usr/local/lib/python3.9/dist-packages/emhass/command_line.py:167: FutureWarning:
Passing method to DatetimeIndex.get_loc is deprecated and will raise in a future version. Use index.get_indexer([item], method=...) instead.
[2022-04-21 09:56:54,381] WARNING in command_line: Problem finding DataFrame index, using nearest option as a backup
/usr/local/lib/python3.9/dist-packages/emhass/command_line.py:170: FutureWarning:
Passing method to DatetimeIndex.get_loc is deprecated and will raise in a future version. Use index.get_indexer([item], method=...) instead.
192.168.86.50 - - [21/Apr/2022 09:56:54] "POST /action/publish-data HTTP/1.1" 200 -
192.168.86.50 - - [21/Apr/2022 09:58:30] "GET / HTTP/1.1" 200 -
192.168.86.50 - - [21/Apr/2022 09:58:30] "GET /static/style.css HTTP/1.1" 304 -
[2022-04-21 10:20:54,861] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1526, in full_dispatch_request
    return self.finalize_request(rv)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1545, in finalize_request
    response = self.make_response(rv)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1701, in make_response
    raise TypeError(
TypeError: The view function for 'index' did not return a valid response. The function either returned None or ended without a return statement.
192.168.86.50 - - [21/Apr/2022 10:20:54] "GET / HTTP/1.1" 500 -
[2022-04-21 13:03:48,154] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1526, in full_dispatch_request
    return self.finalize_request(rv)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1545, in finalize_request
    response = self.make_response(rv)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1701, in make_response
    raise TypeError(
TypeError: The view function for 'index' did not return a valid response. The function either returned None or ended without a return statement.
192.168.86.50 - - [21/Apr/2022 13:03:48] "GET / HTTP/1.1" 500 -
192.168.86.50 - - [21/Apr/2022 13:03:48] "GET /favicon.ico HTTP/1.1" 404 -
[2022-04-21 13:46:41,855] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1526, in full_dispatch_request
    return self.finalize_request(rv)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1545, in finalize_request
    response = self.make_response(rv)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1701, in make_response
    raise TypeError(
TypeError: The view function for 'index' did not return a valid response. The function either returned None or ended without a return statement.
192.168.86.50 - - [21/Apr/2022 13:46:41] "GET / HTTP/1.1" 500 -
[2022-04-21 15:25:05,040] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1526, in full_dispatch_request
    return self.finalize_request(rv)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1545, in finalize_request
    response = self.make_response(rv)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1701, in make_response
    raise TypeError(
TypeError: The view function for 'index' did not return a valid response. The function either returned None or ended without a return statement.
192.168.86.50 - - [21/Apr/2022 15:25:05] "GET / HTTP/1.1" 500 -
```

davidusb-geek, 2022-04-21:
Ok thank you. Yes it will be better to follow this debug here. 
I'll have a deeper look at this. I thought that these error disappeared with the latest v0.1.24.
In any case on my dev environment I don't have these anymore. Tested it yesterday on the dev-container and working just fine. 
I may still have some `if/else` conditions without proper return. But my question is about the hanging. So after a while you're not able to run optimization or publish any data at all?

davidusb-geek, 2022-04-21:
Ok, I was not properly using returns in flask. I'm very sorry about this. Still I don't understand why my dev environment wasn't returning me this error. Anyway I think is fixed in v0.1.25. I've no access right now to my dev env so I couldn't completely test this solution.

purcell-lab, 2022-04-21:
Thanks for your perseverance.

0.1.25 has regressed and won't run the day ahead now 

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
 * Serving Flask app 'app_server' (lazy loading)
 * Environment: development
 * Debug mode: off
 * Running on all addresses (0.0.0.0)
   WARNING: This is a development server. Do not use it in a production deployment.
 * Running on http://127.0.0.1:5000
 * Running on http://172.30.33.4:5000 (Press CTRL+C to quit)
192.168.86.22 - - [21/Apr/2022 22:20:27] "GET / HTTP/1.1" 200 -
192.168.86.22 - - [21/Apr/2022 22:20:27] "GET /static/style.css HTTP/1.1" 304 -
192.168.86.22 - - [21/Apr/2022 22:20:29] "GET /favicon.ico HTTP/1.1" 404 -
[2022-04-21 22:20:50,472] WARNING in optimization: Failed LP solve with PULP_CBC_CMD solver, falling back to default Pulp
[2022-04-21 22:20:50,505] WARNING in optimization: Failed LP solve with default Pulp solver, falling back to GLPK_CMD
[2022-04-21 22:20:52,067] ERROR in app: Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/usr/src/app_server.py", line 154, in action_call
    return redirect(url_for(".index"))
NameError: name 'redirect' is not defined
192.168.86.22 - - [21/Apr/2022 22:20:52] "POST /action/dayahead-optim HTTP/1.1" 500 -
```

davidusb-geek, 2022-04-21:
The problem of not testing. Just forgot some missing imports.
I think it is solved now >> v0.1.26

piio, 2022-04-21:
Hi,
I updated to 0.1.26 and I still see:
2022-04-21 18:32:20,060] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1526, in full_dispatch_request
    return self.finalize_request(rv)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1545, in finalize_request
    response = self.make_response(rv)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1701, in make_response
    raise TypeError(
TypeError: The view function for 'index' did not return a valid response. The function either returned None or ended without a return statement.

davidusb-geek, 2022-04-21:
Hi, the new version should fix it >> v0.1.27
I just took some time to debug this and come with a proper solution. I'm no flask expert and find it difficult to implement unittest and mock the add-on data. However tested it again on dev-container and no errors.
[image]

purcell-lab, 2022-04-21:
Unfortunately 0.1.27 is still crashing with flask errors :-(

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
 * Serving Flask app 'app_server' (lazy loading)
 * Environment: development
 * Debug mode: off
 * Running on all addresses (0.0.0.0)
   WARNING: This is a development server. Do not use it in a production deployment.
 * Running on http://127.0.0.1:5000
 * Running on http://172.30.33.4:5000 (Press CTRL+C to quit)
192.168.86.22 - - [22/Apr/2022 08:22:13] "GET / HTTP/1.1" 200 -
192.168.86.22 - - [22/Apr/2022 08:22:13] "GET /static/style.css HTTP/1.1" 304 -
[2022-04-22 08:22:48,990] WARNING in optimization: Failed LP solve with PULP_CBC_CMD solver, falling back to default Pulp
[2022-04-22 08:22:49,019] WARNING in optimization: Failed LP solve with default Pulp solver, falling back to GLPK_CMD
192.168.86.22 - - [22/Apr/2022 08:22:51] "POST /action/dayahead-optim HTTP/1.1" 302 -
192.168.86.22 - - [22/Apr/2022 08:22:51] "GET / HTTP/1.1" 200 -
192.168.86.22 - - [22/Apr/2022 08:23:04] "GET / HTTP/1.1" 200 -
192.168.86.22 - - [22/Apr/2022 08:23:04] "GET /static/style.css HTTP/1.1" 304 -
192.168.86.22 - - [22/Apr/2022 08:23:38] "POST /action/publish-data HTTP/1.1" 302 -
192.168.86.22 - - [22/Apr/2022 08:23:38] "GET / HTTP/1.1" 200 -
192.168.86.22 - - [22/Apr/2022 08:23:49] "GET / HTTP/1.1" 200 -
192.168.86.22 - - [22/Apr/2022 08:23:49] "GET /static/style.css HTTP/1.1" 304 -
[2022-04-22 08:29:59,790] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/usr/src/app_server.py", line 105, in index
    return make_response(source_html)
  File "/usr/local/lib/python3.9/dist-packages/flask/helpers.py", line 189, in make_response
    return current_app.make_response(args)  # type: ignore
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1701, in make_response
    raise TypeError(
TypeError: The view function for 'index' did not return a valid response. The function either returned None or ended without a return statement.
192.168.86.22 - - [22/Apr/2022 08:29:59] "GET / HTTP/1.1" 500 -
[2022-04-22 08:30:01,875] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/usr/src/app_server.py", line 105, in index
    return make_response(source_html)
  File "/usr/local/lib/python3.9/dist-packages/flask/helpers.py", line 189, in make_response
    return current_app.make_response(args)  # type: ignore
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1701, in make_response
    raise TypeError(
TypeError: The view function for 'index' did not return a valid response. The function either returned None or ended without a return statement.
192.168.86.22 - - [22/Apr/2022 08:30:01] "GET / HTTP/1.1" 500 -
[2022-04-22 08:30:32,894] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/usr/src/app_server.py", line 105, in index
    return make_response(source_html)
  File "/usr/local/lib/python3.9/dist-packages/flask/helpers.py", line 189, in make_response
    return current_app.make_response(args)  # type: ignore
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1701, in make_response
    raise TypeError(
TypeError: The view function for 'index' did not return a valid response. The function either returned None or ended without a return statement.
192.168.86.22 - - [22/Apr/2022 08:30:32] "GET / HTTP/1.1" 500 -

davidusb-geek, 2022-04-22:
That's a bummer! Any chance of more hints from the supervisor logs?

davidusb-geek, 2022-04-22:
Ok guys. I just rearranged everything in the web server script. Using directly the flask make_response method.
Hopefully it will solve the issues now >> v0.1.28

purcell-lab, 2022-04-23:
TypeError: from 0.1.28.

Maybe print the version number on startup in the logs, so we know which version we are talking about.

What can I do to assist with the debuging?

I'lll see what I can pull from syslog.

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
 * Serving Flask app 'app_server' (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
 * Running on all addresses (0.0.0.0)
   WARNING: This is a development server. Do not use it in a production deployment.
 * Running on http://127.0.0.1:5000
 * Running on http://172.30.33.4:5000 (Press CTRL+C to quit)
192.168.86.50 - - [23/Apr/2022 12:29:45] "GET / HTTP/1.1" 200 -
192.168.86.50 - - [23/Apr/2022 12:29:46] "GET /static/style.css HTTP/1.1" 304 -
[2022-04-23 12:30:20,655] WARNING in optimization: Failed LP solve with PULP_CBC_CMD solver, falling back to default Pulp
[2022-04-23 12:30:20,691] WARNING in optimization: Failed LP solve with default Pulp solver, falling back to GLPK_CMD
192.168.86.50 - - [23/Apr/2022 12:30:27] "POST /action/dayahead-optim HTTP/1.1" 201 -
192.168.86.50 - - [23/Apr/2022 12:33:43] "GET / HTTP/1.1" 200 -
192.168.86.50 - - [23/Apr/2022 12:33:43] "GET /static/style.css HTTP/1.1" 304 -
192.168.86.50 - - [23/Apr/2022 12:34:04] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [23/Apr/2022 12:34:31] "GET / HTTP/1.1" 200 -
192.168.86.50 - - [23/Apr/2022 12:34:31] "GET /static/style.css HTTP/1.1" 304 -
[2022-04-23 13:17:29,974] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/usr/src/app_server.py", line 176, in index
    return make_response(template.render(injection_dict=injection_dict))
  File "/usr/local/lib/python3.9/dist-packages/jinja2/environment.py", line 1291, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.9/dist-packages/jinja2/environment.py", line 926, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "/usr/src/templates/index.html", line 23, in top-level template code
    {% for plot in injection_dict %}
TypeError: 'NoneType' object is not iterable
192.168.86.50 - - [23/Apr/2022 13:17:29] "GET / HTTP/1.1" 500 -

davidusb-geek, 2022-04-23:
Your error is that `injection_dict` variable is defined as `None`. So either the flask_caching is not properly functioning on your system or there is a problem with the saved data after the optimization. Could you check if the optimization results are there in a csv file inside the docker container of the add-on? In the data folder. What I can do is to avoid using cache at all and managing data differently. But what is annoying is that on the dev env proposed for add-on development this works fine.

davidusb-geek, 2022-04-23:
Another question, this is failing when you hit the publish data button or when you refresh the web ui? If you try just the `curl` method to publish the data you have the same problem?

purcell-lab, 2022-04-23:
> Your error is that `injection_dict` variable is defined as `None`. So either the flask_caching is not properly functioning on your system or there is a problem with the saved data after the [...]

I can confirm after running the day ahead the optimisation is updated on the web site.

How do I locate the docker container data folder from the command line?

purcell-lab, 2022-04-23:
> Another question, this is failing when you hit the publish data button or when you refresh the web ui? If you try just the `curl` method to publish the data you have the same problem?

I have been using the web ui buttons for both the dayahead and publish actions, both seem to complete correctly and I can see the various sensors get loaded into HA, but the sensors never seem to update, which makes me think the add-on is hanging/ crashing, I presume the addon should be updating these sensors as time progresses:

[image]

I am suspecting the web ui maybe causing the issues, so I'll try and run now from just the curl method, and the calls seem to be good, but how can I check that the add-on is running without calling the web ui?

```
mark@penguin:~$ curl -i -H "Content-Type: application/json" -X POST -d '{}' http://odroid.lan:5000/action/dayahead-optim
HTTP/1.1 201 CREATED
Server: Werkzeug/2.1.1 Python/3.9.2
Date: Sat, 23 Apr 2022 08:24:07 GMT
Content-Type: text/html; charset=utf-8
Content-Length: 45

EMHASS >> Action dayahead-optim executed... 
mark@penguin:~$ curl -i -H "Content-Type: application/json" -X POST -d '{}' http://odroid.lan:5000/action/publish-data
HTTP/1.1 201 CREATED
Server: Werkzeug/2.1.1 Python/3.9.2
Date: Sat, 23 Apr 2022 08:25:11 GMT
Content-Type: text/html; charset=utf-8
Content-Length: 43

EMHASS >> Action publish-data executed... 
mark@penguin:~$ 
```

davidusb-geek, 2022-04-23:
No you have to create an automation to update the sensor values. See the docs for a snippet example

davidusb-geek, 2022-04-23:
If you just go to the add-on info pane, the little green light will indicate that the add-on server is up.

purcell-lab, 2022-04-23:
Ok got it.

We call the publish data action every five minute via the automation to
update the p_ sensors, which we then use as a trigger to switch the load.

purcell-lab, 2022-04-23:
My problem is I can have the green light but the add on web UI is
unresponsive/ overloaded.

Is the add on still available in the background when the web UI is
unresponsive? Presumably not via the curl method as that uses the web UI.

davidusb-geek, 2022-04-23:
Yes I think that the web ui looking unresponsive is just a wrong rendering. But in any case if the green light is on this means that the server is running and accepting curl requests with no problem. The curl command is not using the web ui, it is just using and endpoint route, so even if the web server may look unresponsive it will accept any curl command with no issues

purcell-lab, 2022-04-24:
Here is just using the curl method and not accessing the web UI.

You can see in the log that I can publish data a few times, but then the add on hangs and can no longer publish data.

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
 * Serving Flask app 'app_server' (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
 * Running on all addresses (0.0.0.0)
   WARNING: This is a development server. Do not use it in a production deployment.
 * Running on http://127.0.0.1:5000
 * Running on http://172.30.33.4:5000 (Press CTRL+C to quit)
[2022-04-24 09:42:34,960] WARNING in optimization: Failed LP solve with PULP_CBC_CMD solver, falling back to default Pulp
[2022-04-24 09:42:34,982] WARNING in optimization: Failed LP solve with default Pulp solver, falling back to GLPK_CMD
192.168.86.50 - - [24/Apr/2022 09:42:37] "POST /action/dayahead-optim HTTP/1.1" 201 -
192.168.86.50 - - [24/Apr/2022 09:43:27] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [24/Apr/2022 09:45:59] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [24/Apr/2022 09:46:33] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [24/Apr/2022 09:48:55] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [24/Apr/2022 09:50:07] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [24/Apr/2022 09:51:28] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [24/Apr/2022 09:52:40] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [24/Apr/2022 09:53:02] "POST /action/publish-data HTTP/1.1" 201 -
[2022-04-24 09:58:03,304] ERROR in app: Exception on /action/publish-data [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/usr/src/app_server.py", line 182, in action_call
    params = treat_data_params(data, params)
  File "/usr/src/app_server.py", line 47, in treat_data_params
    forecast_dates = get_forecast_dates(params['retrieve_hass_conf'][0]['freq'], params['optim_conf'][1]['delta_forecast'])
TypeError: 'NoneType' object is not subscriptable
192.168.86.50 - - [24/Apr/2022 09:58:03] "POST /action/publish-data HTTP/1.1" 500 -

davidusb-geek, 2022-04-24:
I think that a None is again returned from a cached object. So that flask caching module is definitely not working properly. This is at the origin of the issue with the web ui also. I will fix that as soon as I can and just use simple but robust data saving using pickle

davidusb-geek, 2022-04-24:
Done!!! Please check >> v0.1.31
I've dropped using caching at all. I'm just using plain old pythonic pickle objects.
The add-on is now using pre-built docker containers, so installation should be easier for the user.

purcell-lab, 2022-04-24:
Thanks for your persistence.

I'll have a look.

purcell-lab, 2022-04-25:
Success, had been running stable now for a couple of hours and I can see the emhass sensors updating in HA.

[image]

```[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
 * Serving Flask app 'app_server' (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
 * Running on all addresses (0.0.0.0)
   WARNING: This is a development server. Do not use it in a production deployment.
 * Running on http://127.0.0.1:5000
 * Running on http://172.30.33.4:5000 (Press CTRL+C to quit)
[2022-04-25 07:59:07,459] WARNING in optimization: Failed LP solve with PULP_CBC_CMD solver, falling back to default Pulp
[2022-04-25 07:59:07,494] WARNING in optimization: Failed LP solve with default Pulp solver, falling back to GLPK_CMD
192.168.86.50 - - [25/Apr/2022 07:59:07] "POST /action/dayahead-optim HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 07:59:13] "GET / HTTP/1.1" 200 -
192.168.86.50 - - [25/Apr/2022 07:59:13] "GET /static/style.css HTTP/1.1" 304 -
192.168.86.50 - - [25/Apr/2022 07:59:45] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 07:59:53] "GET / HTTP/1.1" 200 -
192.168.86.50 - - [25/Apr/2022 07:59:54] "GET /static/style.css HTTP/1.1" 304 -
192.168.86.50 - - [25/Apr/2022 08:00:40] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 08:05:55] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 08:11:10] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 08:16:26] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 08:21:43] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 08:26:59] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 08:32:16] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 08:37:32] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 08:42:46] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 08:48:02] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 08:53:21] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 08:58:38] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 09:03:54] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 09:09:09] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 09:14:24] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 09:19:40] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 09:24:56] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 09:30:12] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 09:35:28] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 09:40:44] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 09:46:00] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 09:51:16] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 09:56:32] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 10:01:48] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 10:07:05] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 10:12:21] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 10:17:37] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 10:22:55] "POST /action/publish-data HTTP/1.1" 201 -
192.168.86.50 - - [25/Apr/2022 10:28:11] "POST /action/publish-data HTTP/1.1" 201 -
```

closed 2022-04-25 by purcell-lab

davidusb-geek, 2022-04-25:
Great! Thanks to you also for your patience testing this.