davidusb-geek/emhass-add-on #4: Error during Add-on install

rd5802, 2022-04-22:
When trying to install add-on the following error occurs:

`The command '/bin/bash -o pipefail -c apt-get update && apt-get install -y --no-install-recommends libffi-dev python3 python3-pip python3-dev git build-essential coinor-libcbc-dev coinor-cbc libglpk-dev glpk-utils && pip3 install --no-cache-dir -U setuptools wheel && pip3 install --no-cache-dir -r requirements.txt && apt-get purge -y --auto-remove git build-essential python3-dev && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 159`

Home Assistant Supervised.

Supervisor log:

```
22-04-22 22:23:46 INFO (SyncWorker_1) [supervisor.docker.addon] Starting build for 5b918bf2/aarch64-addon-emhass:0.1.27
22-04-22 22:23:48 ERROR (SyncWorker_1) [supervisor.docker.addon] Can't build 5b918bf2/aarch64-addon-emhass:0.1.27: The command '/bin/bash -o pipefail -c apt-get update     && apt-get install -y --no-install-recommends         libffi-dev         python3         python3-pip         python3-dev         git         build-essential         coinor-libcbc-dev         coinor-cbc         libglpk-dev         glpk-utils     && pip3 install --no-cache-dir -U setuptools wheel     && pip3 install --no-cache-dir -r requirements.txt     && apt-get purge -y --auto-remove         git         build-essential         python3-dev     && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 159
22-04-22 22:23:48 ERROR (SyncWorker_1) [supervisor.docker.addon] Build log: 
Step 1/19 : ARG BUILD_FROM
Step 2/19 : FROM ${BUILD_FROM}
 ---> 0fb9de056226
Step 3/19 : WORKDIR /usr/src
 ---> Using cache
 ---> ee36489385da
Step 4/19 : COPY ./requirements.txt /usr/src/requirements.txt
 ---> Using cache
 ---> 2d7e008574d3
Step 5/19 : RUN apt-get update     && apt-get install -y --no-install-recommends         libffi-dev         python3         python3-pip         python3-dev         git         build-essential         coinor-libcbc-dev         coinor-cbc         libglpk-dev         glpk-utils     && pip3 install --no-cache-dir -U setuptools wheel     && pip3 install --no-cache-dir -r requirements.txt     && apt-get purge -y --auto-remove         git         build-essential         python3-dev     && rm -rf /var/lib/apt/lists/*
 ---> [Warning] The requested image's platform (linux/arm64/v8) does not match the detected host platform (linux/arm/v7) and no specific platform was requested
 ---> Running in 41ebe3254dcd
Removing intermediate container 41ebe3254dcd
```

davidusb-geek, 2022-04-22:
Hi, strange error. I've seen some comments saying that this was an issue at sometime with docker and that updating your docker install will fix it. You can try that for now. I'll try to compile the images for the different platforms myself and this will ease the installation procedure for everyone.. I'm working on it

davidusb-geek, 2022-04-24:
Hi, Please check >> v0.1.31
The add-on is now using pre-built docker containers, so installation should be easier as no compile is needed on the user side. The image will be downloaded and directly installed from docker-hub.

davidusb-geek, 2022-04-29:
Hi I'm closing as you should be able to install the add-on with direct image download from docker-hub for armv7 architectures. Add-on >>v0.1.38

closed 2022-04-29 by davidusb-geek