davidusb-geek/emhass-add-on #7: s6-overlay-suexec: fatal: can only run as pid 1

purcell-lab, 2022-05-13:
Upgrade/ Fresh install to 0.2.1 fails:

[image]

purcell-lab, 2022-05-13:
Issue still present upgrading to 0.2.2

davidusb-geek, 2022-05-13:
Hi, I'm working on solving this...

purcell-lab, 2022-05-13:
Upgrade to 0.2.3:

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-supervise emhass: warning: unable to spawn ./run - waiting 10 seconds
s6-supervise emhass (child): fatal: unable to exec run: Permission denied
s6-supervise emhass: warning: unable to spawn ./run - waiting 10 seconds
s6-supervise emhass (child): fatal: unable to exec run: Permission denied
s6-supervise emhass: warning: unable to spawn ./run - waiting 10 seconds
s6-supervise emhass (child): fatal: unable to exec run: Permission denied
s6-supervise emhass: warning: unable to spawn ./run - waiting 10 seconds
s6-supervise emhass (child): fatal: unable to exec run: Permission denied

davidusb-geek, 2022-05-13:
Ok, they upgraded s6 overlay from v2 to v3 and that broke a lot of add-ons out there. I applied the proposed solution. Let's see if it works.
An open discussion here: https://community.home-assistant.io/t/hello-world-example-addon-from-developer-docs-stopped-working-s6-overlay-issue/421486

davidusb-geek, 2022-05-13:
In the meantime if that doesn't work. I upgraded the main emhass core module and it can now be used with docker in standalone mode. So not as an add-on but as an external container. It should be able to communicate fine with HA.
I've added a new Model Predictive Control !!!

purcell-lab, 2022-05-13:
Thanks,

Yes very much looking forward to MPC!

purcell-lab, 2022-05-14:
Loading at start now with 0.2.5, thanks for the quick turn around with the upstream changes.

I like the additional data in the logging now as well.

I'll have a play around and check out MPC.

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
[2022-05-14 10:14:33,808] INFO in web_server: Launching the emhass webserver at: http://0.0.0.0:5000
[2022-05-14 10:14:33,809] INFO in web_server: Home Assistant data fetch will be performed using url: http://supervisor/core/api
[2022-05-14 10:14:33,809] INFO in web_server: The base path is: /usr/src
[2022-05-14 10:14:33,814] INFO in web_server: Using core emhass version: 0.3.2

closed 2022-05-14 by purcell-lab