davidusb-geek/emhass-add-on #9: 66% process CPU - WARNING:waitress.queue:Task queue depth is 4

purcell-lab, 2022-05-14:
Emhass-add-om 0.2.5

Iterates and gets stuck with:

WARNING:waitress.queue:Task queue depth is 4

With ever increasing depths.

```
INFO:web_server:Perform optimization for the day-ahead
[2022-05-14 12:18:07,721] WARNING in optimization: Failed LP solve with PULP_CBC_CMD solver, falling back to default Pulp
WARNING:web_server:Failed LP solve with PULP_CBC_CMD solver, falling back to default Pulp
[2022-05-14 12:18:07,764] WARNING in optimization: Failed LP solve with default Pulp solver, falling back to GLPK_CMD
WARNING:web_server:Failed LP solve with default Pulp solver, falling back to GLPK_CMD
WARNING:waitress.queue:Task queue depth is 1
WARNING:waitress.queue:Task queue depth is 2
WARNING:waitress.queue:Task queue depth is 3
WARNING:waitress.queue:Task queue depth is 4
WARNING:waitress.queue:Task queue depth is 5
WARNING:waitress.queue:Task queue depth is 6
WARNING:waitress.queue:Task queue depth is 7
WARNING:waitress.queue:Task queue depth is 8
WARNING:waitress.queue:Task queue depth is 9

davidusb-geek, 2022-05-14:
This is related to the other open issue. It is just that flask can't find my template html. This works perfectly fine as it is when I run the new stand-alone docker for the core emhass module. I will find a solution to this when I have access to my dev env tomorrow.

davidusb-geek, 2022-05-16:
Hi. I just tested the add-on in v0.2.7. It is much better now. There is still a small problem with the webui buttons and the CSS style not found but the optimizations are functional now. I'll patch those issues later.

purcell-lab, 2022-05-16:
Working for me here. Thanks.

purcell-lab, 2022-05-21:
Seems the CPU/ waitress issue is still with us.

[image]

```
WARNING:waitress.queue:Task queue depth is 55
WARNING:waitress.queue:Task queue depth is 56
WARNING:waitress.queue:Task queue depth is 57
[2022-05-22 04:50:17,455] INFO in optimization: Status: Optimal
[2022-05-22 04:50:17,456] INFO in optimization: Total value of the Cost function = -12.39
[2022-05-22 04:50:18,336] INFO in command_line: Setting up needed data
[2022-05-22 04:50:18,347] INFO in web_server:  >> Publishing data...
[2022-05-22 04:50:18,349] INFO in command_line: Publishing data to HASS instance
[2022-05-22 04:50:18,414] INFO in retrieve_hass: Successfully posted to sensor.p_pv_forecast = -5.0
[2022-05-22 04:50:18,464] INFO in retrieve_hass: Successfully posted to sensor.p_load_forecast = 306.0
[2022-05-22 04:50:18,502] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable0 = 0.0
[2022-05-22 04:50:18,539] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable1 = 0.0
[2022-05-22 04:50:18,599] INFO in retrieve_hass: Successfully posted to sensor.p_batt_forecast = 0.0
[2022-05-22 04:50:18,615] INFO in command_line: Setting up needed data
[2022-05-22 04:50:18,624] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2022-05-22 04:50:22,736] INFO in forecast: Retrieving data from hass for load forecast using method = naive
[2022-05-22 04:50:22,738] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-05-22 04:50:26,086] INFO in web_server:  >> Performing naive MPC optimization...
[2022-05-22 04:50:26,086] INFO in command_line: Performing naive MPC optimization
[2022-05-22 04:50:26,145] INFO in optimization: Perform an iteration of a naive MPC controller
[2022-05-22 04:50:26,364] WARNING in optimization: Failed LP solve with PULP_CBC_CMD solver, falling back to default Pulp
[2022-05-22 04:50:26,427] WARNING in optimization: Failed LP solve with default Pulp solver, falling back to GLPK_CMD
WARNING:waitress.queue:Task queue depth is 56
WARNING:waitress.queue:Task queue depth is 57
[2022-05-22 04:55:16,528] INFO in optimization: Status: Optimal
[2022-05-22 04:55:16,529] INFO in optimization: Total value of the Cost function = -12.53
[2022-05-22 04:55:17,164] INFO in command_line: Setting up needed data
[2022-05-22 04:55:17,175] INFO in web_server:  >> Publishing data...
[2022-05-22 04:55:17,177] INFO in command_line: Publishing data to HASS instance
[2022-05-22 04:55:17,245] INFO in retrieve_hass: Successfully posted to sensor.p_pv_forecast = -5.0
[2022-05-22 04:55:17,290] INFO in retrieve_hass: Successfully posted to sensor.p_load_forecast = 306.0
[2022-05-22 04:55:17,328] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable0 = 0.0
[2022-05-22 04:55:17,366] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable1 = 0.0
[2022-05-22 04:55:17,413] INFO in retrieve_hass: Successfully posted to sensor.p_batt_forecast = 0.0
[2022-05-22 04:55:17,436] INFO in command_line: Setting up needed data
[2022-05-22 04:55:17,446] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2022-05-22 04:55:21,479] INFO in forecast: Retrieving data from hass for load forecast using method = naive
[2022-05-22 04:55:21,481] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-05-22 04:55:25,688] INFO in web_server:  >> Performing naive MPC optimization...
[2022-05-22 04:55:25,689] INFO in command_line: Performing naive MPC optimization
[2022-05-22 04:55:25,742] INFO in optimization: Perform an iteration of a naive MPC controller
[2022-05-22 04:55:25,960] WARNING in optimization: Failed LP solve with PULP_CBC_CMD solver, falling back to default Pulp
[2022-05-22 04:55:26,027] WARNING in optimization: Failed LP solve with default Pulp solver, falling back to GLPK_CMD
WARNING:waitress.queue:Task queue depth is 56
WARNING:waitress.queue:Task queue depth is 57
WARNING:waitress.queue:Task queue depth is 58
WARNING:waitress.queue:Task queue depth is 59
WARNING:waitress.queue:Task queue depth is 60
WARNING:waitress.queue:Task queue depth is 61
WARNING:waitress.queue:Task queue depth is 62
WARNING:waitress.queue:Task queue depth is 63
WARNING:waitress.queue:Task queue depth is 64
WARNING:waitress.queue:Task queue depth is 65
WARNING:waitress.queue:Task queue depth is 66
WARNING:waitress.queue:Task queue depth is 67
WARNING:waitress.queue:Task queue depth is 68
WARNING:waitress.queue:Task queue depth is 69
WARNING:waitress.queue:Task queue depth is 70
WARNING:waitress.queue:Task queue depth is 71
WARNING:waitress.queue:Task queue depth is 72
WARNING:waitress.queue:Task queue depth is 73
WARNING:waitress.queue:Task queue depth is 74
WARNING:waitress.queue:Task queue depth is 75
WARNING:waitress.queue:Task queue depth is 76
WARNING:waitress.queue:Task queue depth is 77
[... 10 lines cut ...]
[2022-05-22 05:52:22,448] INFO in command_line: Setting up needed data
[2022-05-22 05:52:22,458] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2022-05-22 05:52:27,017] INFO in forecast: Retrieving data from hass for load forecast using method = naive
[2022-05-22 05:52:27,020] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-05-22 05:52:30,009] INFO in web_server:  >> Performing naive MPC optimization...
[2022-05-22 05:52:30,010] INFO in command_line: Performing naive MPC optimization
[2022-05-22 05:52:30,060] INFO in optimization: Perform an iteration of a naive MPC controller
[... cut ...]
WARNING:waitress.queue:Task queue depth is 81
WARNING:waitress.queue:Task queue depth is 82
WARNING:waitress.queue:Task queue depth is 83
WARNING:waitress.queue:Task queue depth is 84
WARNING:waitress.queue:Task queue depth is 85
WARNING:waitress.queue:Task queue depth is 86
WARNING:waitress.queue:Task queue depth is 87
WARNING:waitress.queue:Task queue depth is 88
WARNING:waitress.queue:Task queue depth is 89
WARNING:waitress.queue:Task queue depth is 90
WARNING:waitress.queue:Task queue depth is 91

davidusb-geek, 2022-05-22:
It is possible to tune the numbers of threads used by waitress. I could add this option to see if its better for you. But I'm not experiencing this issue at all so it is maybe related to the size of your LP problem. Two possible solutions, try reducing the prediction horizon to hence reduce the size of the LP formulation or try calling the optimization at a lower frequency, say 10min to seeof the problem persists.

davidusb-geek, 2022-05-23:
Another option in your specific case is the falling back to the GLPK solver. This is also maybe the root of the issue. You can go ahead and try to solve this.
While the add-on is running you can search for the container name using `docker ps -a` then you can go inside the docker container using:
```
docker exec -it <container-name> /bin/bash
```
Once inside the container open Python with `python3` and import pulp with `import pulp as plp`, then launch the pulp tests to see what messages it is throwing and what solvers are available for you:
```
plp.pulpTestAll()
```
From we can try to find a solution to why pulp is not loading the default solver in your machine.

davidusb-geek, 2022-05-23:
Otherwise, I've just released a new version v0.2.14 where I added a new threads parameter to change defaults value of threads used by waitress. It seems that this could help improve these kind of issues.

purcell-lab, 2022-05-23:
Walking back to 24 time slots has made things more stable, but the 12 hours misses to full solar production cycle so it tries to schedule loads overnight, when it would be better to wait for solar the next day.

I also found this report: https://github.com/coin-or/pulp/issues/433, so it looks like the cbc is not compiled for arm within the docker ;-(

pulpTestAll() inside my Ubuntu OS completes using pulp.solvers.COIN_CMD, so it is possible to get this working:

```
mark@odroid:~$ python3
Python 3.8.10 (default, Mar 15 2022, 12:22:08) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pulp as plp
>>> plp.pulpTestAll()
Solver <class 'pulp.solvers.PULP_CBC_CMD'> unavailable
Solver <class 'pulp.solvers.CPLEX_DLL'> unavailable
Solver <class 'pulp.solvers.CPLEX_CMD'> unavailable
Solver <class 'pulp.solvers.CPLEX_PY'> unavailable
         Testing zero subtraction
         Testing inconsistant lp solution
         Testing continuous LP solution
         Testing maximize continuous LP solution
         Testing unbounded continuous LP solution
         Testing Long Names
         Testing repeated Names
         Testing zero constraint
         Testing zero objective
         Testing LpVariable (not LpAffineExpression) objective
         Testing Long lines in LP
         Testing LpAffineExpression divide
         Testing MIP solution
         Testing MIP solution with floats in objective
         Testing MIP relaxation
         Testing feasibility problem (no objective)
         Testing an infeasible problem
         Testing an integer infeasible problem
         Testing column based modelling
         Testing fractional constraints
         Testing elastic constraints (no change)
         Testing elastic constraints (freebound)
         Testing elastic constraints (penalty unchanged)
         Testing elastic constraints (penalty unbounded)
* Solver <class 'pulp.solvers.COIN_CMD'> passed.
Solver <class 'pulp.solvers.COINMP_DLL'> unavailable
Solver <class 'pulp.solvers.GLPK_CMD'> unavailable
Solver <class 'pulp.solvers.XPRESS'> unavailable
Solver <class 'pulp.solvers.GUROBI'> unavailable
Solver <class 'pulp.solvers.GUROBI_CMD'> unavailable
Solver <class 'pulp.solvers.PYGLPK'> unavailable
Solver <class 'pulp.solvers.YAPOSIB'> unavailable
>>> 
```

Inside the docker image it fails:

```
root@odroid:/home/mark# docker exec -it addon_5b918bf2_emhass /bin/bash
root@5b918bf2-emhass:/# python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> E        Testing elastic constraints (penalty unbounded)
KeyboardInterrupt
>>> import pulp as plp
>>> plp.pulpTestAll()
ssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssss         Test that logic put in place for deprecation handling of indexs works
.        Testing 'indexs' param continues to work for LpVariable.dicts
         Testing 'indexs' param continues to work for LpVariable.matrix
.        Testing 'indices' argument works in LpVariable.dicts
         Testing 'indices' param continues to work for LpVariable.matrix
.        Testing invalid status
[... 956 lines cut ...]
  File "/usr/local/lib/python3.9/dist-packages/pulp/tests/test_pulp.py", line 1358, in pulpTestCheck
    status = prob.solve(solver, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/pulp/pulp.py", line 1913, in solve
    status = solver.actualSolve(self, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/pulp/apis/coin_api.py", line 137, in actualSolve
    return self.solve_CBC(lp, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/pulp/apis/coin_api.py", line 194, in solve_CBC
    cbc = subprocess.Popen(args, stdout=pipe, stderr=pipe, stdin=devnull)
  File "/usr/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.9/subprocess.py", line 1823, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 8] Exec format error: '/usr/local/lib/python3.9/dist-packages/pulp/apis/../solverdir/cbc/linux/64/cbc'

----------------------------------------------------------------------
Ran 840 tests in 16.920s

FAILED (errors=41, skipped=672)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/pulp/tests/run_tests.py", line 12, in pulpTestAll
    raise pulp.PulpError("Tests Failed")
pulp.constants.PulpError: Tests Failed
>>> 
```

davidusb-geek, 2022-05-23:
We also had a discussion about this here: [coin-or/pulp#102](https://github.com/coin-or/pulp/issues/102)

davidusb-geek, 2022-05-23:
> Walking back to 24 time slots has made things more stable, but the 12 hours misses to full solar production cycle so it tries to schedule loads overnight, when it would be better to wait for solar the next day.

This exactly why I implemented the MPC with the option to provide the remaining hours to meet for your deferrable loads. This is why I was saying somewhere else that these values should be variables and not fixed. If you provide zero remaining hours then no load will be scheduled after the optimization.

purcell-lab, 2022-05-24:
> Maybe you can try to see if the CBC solver can be installed. It should as it is already on the Dockerfile. Do:
> 
> ```
> apt-get install -y --no-install-recommends coinor-cbc
> ```

Looks like the coinor-cbc package is installed in the docker and provides an ARM aarch64 binary, /usr/bin/cbc, which does run.

Looks like the /usr/local/lib python packages are providing x68=64 & 80386 binaries which do not run on my arm architecture.

Thus this issue is the pulp python library is pulling in the wrong binaries, so needs to be fixed upstream.

As a workaround are we able to specify that on arm emhass should use the valid /usr/bin/cbc executable?

```
root@5b918bf2-emhass:/# dpkg -L coinor-cbc
/.
/usr
/usr/bin
/usr/bin/cbc
/usr/share
/usr/share/doc
/usr/share/doc/coinor-cbc
/usr/share/doc/coinor-cbc/changelog.Debian.gz
/usr/share/doc/coinor-cbc/copyright
/usr/share/man
/usr/share/man/man1
/usr/share/man/man1/cbc.1.gz
root@5b918bf2-emhass:/# file /usr/bin/cbc 
/usr/bin/cbc: ELF 64-bit LSB pie executable, ARM aarch64, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-aarch64.so.1, BuildID[sha1]=fce9754155613dbc2aac30e52ff5c333b04cf3e7, for GNU/Linux 3.7.0, stripped
root@5b918bf2-emhass:/# 
root@5b918bf2-emhass:/# file /usr/local/lib/python3.9/dist-packages/pulp/solverdir/cbc/linux/64/cbc 
/usr/local/lib/python3.9/dist-packages/pulp/solverdir/cbc/linux/64/cbc: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.9, BuildID[sha1]=9e8eafbf26da3f7a1cecac1a24a976274fb04516, not stripped
root@5b918bf2-emhass:/# file /usr/local/lib/python3.9/dist-packages/pulp/solverdir/cbc/linux/32/cbc 
/usr/local/lib/python3.9/dist-packages/pulp/solverdir/cbc/linux/32/cbc: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.2, for GNU/Linux 2.6.9, BuildID[sha1]=dc5438ae3ebec28b55aa2053ef943cda426fd15e, not stripped
root@5b918bf2-emhass:/# /usr/bin/cbc 
Welcome to the CBC MILP Solver 
Version: 2.10.5 
Build Date: Oct 15 2020 

CoinSolver takes input from arguments ( - switches to stdin)
Enter ? for list of commands or help
Coin:Total time (CPU seconds):       0.00   (Wallclock seconds):       1.83

root@5b918bf2-emhass:/# /usr/local/lib/python3.9/dist-packages/pulp/solverdir/cbc/linux/64/cbc
bash: /usr/local/lib/python3.9/dist-packages/pulp/solverdir/cbc/linux/64/cbc: cannot execute binary file: Exec format error
root@5b918bf2-emhass:/#

davidusb-geek, 2022-05-24:
Ok, could you please check inside the emhass docker and in a python3 console what are your available solvers ?
```
import pulp as plp
solver_list = plp.listSolvers()
```

Also in this same console test if the following works:
```
from pulp import PULP_CBC_CMD
model = plp.LpProblem("Example", pl.LpMinimize)
path_to_solver = '/usr/bin/cbc'
solver = PULP_CBC_CMD(path=path_to_solver)
_var = pl.LpVariable('a')
_var2 = pl.LpVariable('a2')
model += _var + _var2 == 1
result = model.solve(solver)
```

Depending on these test we will add either the option to provide the solver name or just the solver path as an input parameter.

purcell-lab, 2022-05-25:
> Depending on these test we will add either the option to provide the solver name or just the solver path as an input parameter.

Not sure how helpful this output is:

```
root@odroid:/home/mark# docker exec -it addon_5b918bf2_emhass /bin/bash
root@5b918bf2-emhass:/# python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pulp as plp
>>> solver_list = plp.listSolvers()
>>> 
>>> from pulp import PULP_CBC_CMD
>>> model = plp.LpProblem("Example", pl.LpMinimize)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'pl' is not defined
>>> path_to_solver = '/usr/bin/cbc'
>>> solver = PULP_CBC_CMD(path=path_to_solver)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/pulp/apis/coin_api.py", line 385, in __init__
    raise PulpSolverError("Use COIN_CMD if you want to set a path")
pulp.apis.core.PulpSolverError: Use COIN_CMD if you want to set a path
>>> _var = pl.LpVariable('a')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'pl' is not defined
>>> _var2 = pl.LpVariable('a2')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'pl' is not defined
>>> model += _var + _var2 == 1
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'model' is not defined
>>> result = model.solve(solver)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'model' is not defined
>>>

purcell-lab, 2022-05-25:
I was able to get success using the following:

```
>>> path_to_cbc = r'/usr/bin/cbc'
>>> import pulp as pl
>>> model = pl.LpProblem("Example", pl.LpMinimize)
>>> solver = pl.COIN_CMD(path=path_to_cbc)
>>> _var = pl.LpVariable('a')
>>> _var2 = pl.LpVariable('a2')
>>> model += _var + _var2 == 1
>>> result = model.solve(solver)
Welcome to the CBC MILP Solver 
Version: 2.10.5 
Build Date: Oct 15 2020 

command line - /usr/bin/cbc /tmp/ade0097a6851468f83f20b596a093a0e-pulp.mps timeMode elapsed branch printingOptions all solution /tmp/ade0097a6851468f83f20b596a093a0e-pulp.sol (default strategy 1)
At line 2 NAME          MODEL
At line 3 ROWS
At line 6 COLUMNS
At line 10 RHS
At line 12 BOUNDS
At line 16 ENDATA
Problem MODEL has 1 rows, 3 columns and 2 elements
Coin0008I MODEL read with 0 errors
Option for timeMode changed from cpu to elapsed
Presolve 0 (-1) rows, 0 (-3) columns and 0 (-2) elements
Empty problem - 0 rows, 0 columns and 0 elements
Optimal - objective value 0
After Postsolve, objective 0, infeasibilities - dual 0 (0), primal 0 (0)
Optimal objective 0 - 0 iterations time 0.002, Presolve 0.00
Option for printingOptions changed from normal to all
Total time (CPU seconds):       0.00   (Wallclock seconds):       0.00

davidusb-geek, 2022-05-25:
Alright then, in the next version I'll add the option to pass the solver name and the path. From what I could read the coin cbc solver is more robust and efficient than glpk.

davidusb-geek, 2022-05-31:
Hi, in the new v0.2.15 you can now provide the solver name and eventually the path. The path is mandatory when using 'COIN_CMD'. You can try that to see if this solve this issue. The default value for the solver name is 'PULP_CBC_CMD' so you will need to change that to 'COIN_CMD' along with the correct path '/usr/bin/cbc'

purcell-lab, 2022-06-01:
Thanks for your flexibility in making this change.

Passing the solver name and path seems to be quite solid.

I have been using my local /usr/bin/cbc for past 48 hours and seems solid.

```
[2022-06-01 15:10:00,082] INFO in command_line: Setting up needed data
[2022-06-01 15:10:00,087] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2022-06-01 15:10:04,152] INFO in forecast: Retrieving data from hass for load forecast using method = naive
[2022-06-01 15:10:04,153] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-06-01 15:10:06,902] INFO in web_server:  >> Performing naive MPC optimization...
[2022-06-01 15:10:06,902] INFO in command_line: Performing naive MPC optimization
[2022-06-01 15:10:06,924] INFO in optimization: Perform an iteration of a naive MPC controller
[2022-06-01 15:10:07,322] INFO in optimization: Status: Optimal
[2022-06-01 15:10:07,323] INFO in optimization: Total value of the Cost function = 75.76
[2022-06-01 15:10:07,984] INFO in command_line: Setting up needed data
[2022-06-01 15:10:07,988] INFO in web_server:  >> Publishing data...
[2022-06-01 15:10:07,989] INFO in command_line: Publishing data to HASS instance
[2022-06-01 15:10:08,036] INFO in retrieve_hass: Successfully posted to sensor.p_pv_forecast = 2162.66
[2022-06-01 15:10:08,082] INFO in retrieve_hass: Successfully posted to sensor.p_load_forecast = 1425.9
[2022-06-01 15:10:08,122] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable0 = 1500.0
[2022-06-01 15:10:08,158] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable1 = 5500.0
[2022-06-01 15:10:08,215] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable2 = 11500.0
[2022-06-01 15:10:08,263] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable3 = 2400.0
[2022-06-01 15:10:08,309] INFO in retrieve_hass: Successfully posted to sensor.p_batt_forecast = 0.0
[2022-06-01 15:10:08,346] INFO in retrieve_hass: Successfully posted to sensor.soc_batt_forecast = 100.0
[2022-06-01 15:11:00,079] INFO in command_line: Setting up needed data
[2022-06-01 15:11:00,084] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2022-06-01 15:11:03,670] INFO in forecast: Retrieving data from hass for load forecast using method = naive
[2022-06-01 15:11:03,672] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-06-01 15:11:06,245] INFO in web_server:  >> Performing naive MPC optimization...
[2022-06-01 15:11:06,245] INFO in command_line: Performing naive MPC optimization
[2022-06-01 15:11:06,267] INFO in optimization: Perform an iteration of a naive MPC controller
[2022-06-01 15:11:06,629] INFO in optimization: Status: Optimal
[2022-06-01 15:11:06,629] INFO in optimization: Total value of the Cost function = 75.76
[2022-06-01 15:11:07,264] INFO in command_line: Setting up needed data
[2022-06-01 15:11:07,269] INFO in web_server:  >> Publishing data...
[2022-06-01 15:11:07,269] INFO in command_line: Publishing data to HASS instance
[2022-06-01 15:11:07,304] INFO in retrieve_hass: Successfully posted to sensor.p_pv_forecast = 2162.66
[2022-06-01 15:11:07,338] INFO in retrieve_hass: Successfully posted to sensor.p_load_forecast = 1425.9
[2022-06-01 15:11:07,374] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable0 = 1500.0
[2022-06-01 15:11:07,424] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable1 = 5500.0
[2022-06-01 15:11:07,472] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable2 = 11500.0
[2022-06-01 15:11:07,526] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable3 = 2400.0
[2022-06-01 15:11:07,575] INFO in retrieve_hass: Successfully posted to sensor.p_batt_forecast = 0.0
[2022-06-01 15:11:07,623] INFO in retrieve_hass: Successfully posted to sensor.soc_batt_forecast = 100.0
[2022-06-01 15:12:00,080] INFO in command_line: Setting up needed data
[2022-06-01 15:12:00,086] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2022-06-01 15:12:03,720] INFO in forecast: Retrieving data from hass for load forecast using method = naive
[2022-06-01 15:12:03,721] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-06-01 15:12:05,749] INFO in web_server:  >> Performing naive MPC optimization...
[2022-06-01 15:12:05,749] INFO in command_line: Performing naive MPC optimization
[2022-06-01 15:12:05,771] INFO in optimization: Perform an iteration of a naive MPC controller
[2022-06-01 15:12:06,133] INFO in optimization: Status: Optimal
[2022-06-01 15:12:06,134] INFO in optimization: Total value of the Cost function = 75.76
[2022-06-01 15:12:06,932] INFO in command_line: Setting up needed data
[2022-06-01 15:12:06,945] INFO in web_server:  >> Publishing data...
[2022-06-01 15:12:06,946] INFO in command_line: Publishing data to HASS instance
[2022-06-01 15:12:07,006] INFO in retrieve_hass: Successfully posted to sensor.p_pv_forecast = 2162.66
[2022-06-01 15:12:07,067] INFO in retrieve_hass: Successfully posted to sensor.p_load_forecast = 1425.9
[2022-06-01 15:12:07,105] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable0 = 1500.0
[2022-06-01 15:12:07,152] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable1 = 5500.0
[2022-06-01 15:12:07,199] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable2 = 11500.0
[2022-06-01 15:12:07,247] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable3 = 2400.0
[... cut ...]
[2022-06-01 15:14:06,975] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable1 = 5500.0
[2022-06-01 15:14:07,017] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable2 = 11500.0
[2022-06-01 15:14:07,056] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable3 = 2400.0
[2022-06-01 15:14:07,107] INFO in retrieve_hass: Successfully posted to sensor.p_batt_forecast = 0.0
[2022-06-01 15:14:07,147] INFO in retrieve_hass: Successfully posted to sensor.soc_batt_forecast = 100.0

closed 2022-06-01 by purcell-lab