davidusb-geek/emhass-add-on #12: Can't update since 0.1.42

prinzkafka, 2022-05-16:
`Can't install davidusb/image-armv7-emhass:0.2.6: 404 Client Error for http+docker://localhost/v1.41/images/create?tag=0.2.6&fromImage=davidusb%2Fimage-armv7-emhass&platform=linux%2Farm%2Fv7: Not Found ("manifest for davidusb/image-armv7-emhass:0.2.6 not found: manifest unknown: manifest unknown")`

Home Assistant OS on Raspberry Pi 4 ( core-2022.5.4)
Home Assistant Supervisor ( Home Assistant OS 7.6)

davidusb-geek, 2022-05-16:
Hi. Yes the latest stable version for arm arch was 0.1.42. The add-on has been unstable since v0.2.0. 
The latest v0.2.7 will hopefully solve the latest issues. I will then build the corresponding images for arm and you should be able to successfully update. These arm images are built manually as they take up to 6h to build...

davidusb-geek, 2022-05-16:
Ok, just tested v0.2.7 and the add-on is functional now. I've just launched the workflow to build the arm images. They should be available in 5 to 6h.

prinzkafka, 2022-05-16:
just wanted to install the version 0.2.7, but now there is already the 0.2.8, so the error comes again :-) so i will wait for the new arm image 

`Can't install davidusb/image-armv7-emhass:0.2.8: 404 Client Error for http+docker://localhost/v1.41/images/create?tag=0.2.8&fromImage=davidusb%2Fimage-armv7-emhass&platform=linux%2Farm%2Fv7: Not Found ("manifest for davidusb/image-armv7-emhass:0.2.8 not found: manifest unknown: manifest unknown")
`

davidusb-geek, 2022-05-16:
Hi there. Yes another error was raised and fixed. So a new patched version was released. But the images are not yet built. They will be tonight.

davidusb-geek, 2022-05-16:
Ok, just launched workflow. You will be able to install the update in 5 to 6h

closed 2022-05-17 by prinzkafka