davidusb-geek/emhass-add-on #13: mpc-optim - TypeError: string indices must be integers

purcell-lab, 2022-05-16:
dayahead optim works, naive-mpc-optim fails:

```
mark@odroid:~$ curl -i -H "Content-Type: application/json" -X POST -d '{"prod_price_forecast":[0.35, 0.37, 0.37, 0.58, 0.58, 0.37, 0.28, 0.29, 0.26, 0.21, 0.25, 0.2, 0.19, 0.19, 0.19, 0.27, 0.27, 0.26, 0.29, 0.33, 0.58
, 0.57, 0.32, 0.3, 0.2, 0.23, 0.21, 0.15, 0.06, 0.06, 0.06, 0.08, 0.06, 0.1, 0.1, 0.19, 0.21, 0.3, 0.57, 0.37, 0.47, 0.47, 1.05, 0.57, 0.47, 0.47, 0.47, 0.47],"load_cost_forecast":[0.48, 0.51, 0.51, 0.73, 0.73, 0.5, 0.
41, 0.42, 0.39, 0.32, 0.37, 0.31, 0.31, 0.3, 0.31, 0.39, 0.4, 0.39, 0.42, 0.46, 0.73, 0.72, 0.45, 0.42, 0.32, 0.34, 0.33, 0.26, 0.16, 0.16, 0.16, 0.18, 0.16, 0.21, 0.21, 0.3, 0.33, 0.42, 0.72, 0.51, 0.61, 0.61, 1.25, 0.72, 0.61, 0.61, 0.61, 0.61]}' http://localhost:5000/action/dayahead-optim
HTTP/1.1 201 CREATED
Content-Length: 45
Content-Type: text/html; charset=utf-8
Date: Mon, 16 May 2022 10:28:16 GMT
Server: waitress

EMHASS >> Action dayahead-optim executed... 
mark@odroid:~$ curl -i -H "Content-Type: application/json" -X POST -d '{"prod_price_forecast":[0.35, 0.37, 0.37, 0.58, 0.58, 0.37, 0.28, 0.29, 0.26, 0.21, 0.25, 0.2, 0.19, 0.19, 0.19, 0.27, 0.27, 0.26, 0.29, 0.33, 0.58, 0.57, 0.32, 0.3, 0.2, 0.23, 0.21, 0.15, 0.06, 0.06, 0.06, 0.08, 0.06, 0.1, 0.1, 0.19, 0.21, 0.3, 0.57, 0.37, 0.47, 0.47, 1.05, 0.57, 0.47, 0.47, 0.47, 0.47],"load_cost_forecast":[0.48, 0.51, 0.51, 0.73, 0.73, 0.5, 0.41, 0.42, 0.39, 0.32, 0.37, 0.31, 0.31, 0.3, 0.31, 0.39, 0.4, 0.39, 0.42, 0.46, 0.73, 0.72, 0.45, 0.42, 0.32, 0.34, 0.33, 0.26, 0.16, 0.16, 0.16, 0.18, 0.16, 0.21, 0.21, 0.3, 0.33, 0.42, 0.72, 0.51, 0.61, 0.61, 1.25, 0.72, 0.61, 0.61, 0.61, 0.61],"prediction_horizon":48,"soc_init":0.17,"soc_final":0.17,"def_total_hours":[4,5,3,1]}' http://localhost:5000/action/naive-mpc-optim
HTTP/1.1 500 INTERNAL SERVER ERROR
Content-Length: 265
Content-Type: text/html; charset=utf-8
Date: Mon, 16 May 2022 10:29:16 GMT
Server: waitress

<!doctype html>
<html lang=en>
<title>500 Internal Server Error</title>
<h1>Internal Server Error</h1>
<p>The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.</p>
```

```
[2022-05-16 20:29:05,631] INFO in optimization: Perform optimization for the day-ahead
INFO:web_server:Perform optimization for the day-ahead
[2022-05-16 20:29:05,827] WARNING in optimization: Failed LP solve with PULP_CBC_CMD solver, falling back to default Pulp
WARNING:web_server:Failed LP solve with PULP_CBC_CMD solver, falling back to default Pulp
[2022-05-16 20:29:05,882] WARNING in optimization: Failed LP solve with default Pulp solver, falling back to GLPK_CMD
WARNING:web_server:Failed LP solve with default Pulp solver, falling back to GLPK_CMD
[2022-05-16 20:29:07,268] INFO in optimization: Status: Optimal
INFO:web_server:Status: Optimal
[2022-05-16 20:29:07,269] INFO in optimization: Total value of the Cost function = -15.61
INFO:web_server:Total value of the Cost function = -15.61
[2022-05-16 20:29:07,866] INFO in command_line: Setting up needed data
INFO:web_server:Setting up needed data
[2022-05-16 20:29:07,870] INFO in web_server:  >> Publishing data...
INFO:web_server: >> Publishing data...
[2022-05-16 20:29:07,871] INFO in command_line: Publishing data to HASS instance
INFO:web_server:Publishing data to HASS instance
[2022-05-16 20:29:07,919] INFO in retrieve_hass: Successfully posted value in existing entity_id
INFO:web_server:Successfully posted value in existing entity_id
[2022-05-16 20:29:07,952] INFO in retrieve_hass: Successfully posted value in existing entity_id
INFO:web_server:Successfully posted value in existing entity_id
[2022-05-16 20:29:07,984] INFO in retrieve_hass: Successfully posted value in existing entity_id
INFO:web_server:Successfully posted value in existing entity_id
[2022-05-16 20:29:08,020] INFO in retrieve_hass: Successfully posted value in existing entity_id
INFO:web_server:Successfully posted value in existing entity_id
[2022-05-16 20:29:08,058] INFO in retrieve_hass: Successfully posted value in existing entity_id
INFO:web_server:Successfully posted value in existing entity_id
[2022-05-16 20:29:08,092] INFO in retrieve_hass: Successfully posted value in existing entity_id
INFO:web_server:Successfully posted value in existing entity_id
[2022-05-16 20:29:08,116] INFO in retrieve_hass: Successfully posted value in existing entity_id
INFO:web_server:Successfully posted value in existing entity_id
[2022-05-16 20:29:16,077] INFO in command_line: Setting up needed data
INFO:web_server:Setting up needed data
[2022-05-16 20:29:16,089] INFO in forecast: Retrieving weather forecast data using method = scrapper
INFO:web_server:Retrieving weather forecast data using method = scrapper
[2022-05-16 20:29:19,304] INFO in forecast: Retrieving data from hass for load forecast using method = naive
INFO:web_server:Retrieving data from hass for load forecast using method = naive
[2022-05-16 20:29:19,305] INFO in retrieve_hass: Retrieve hass get data method initiated...
INFO:web_server:Retrieve hass get data method initiated...
[2022-05-16 20:29:22,205] INFO in web_server:  >> Performing naive MPC optimization...
INFO:web_server: >> Performing naive MPC optimization...
[2022-05-16 20:29:22,205] INFO in command_line: Performing naive MPC optimization
INFO:web_server:Performing naive MPC optimization
[2022-05-16 20:29:22,228] ERROR in app: Exception on /action/naive-mpc-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 136, in action_call
    opt_res = naive_mpc_optim(input_data_dict, app.logger)
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 185, in naive_mpc_optim
    prediction_horizon = input_data_dict['params']['passed_data']['prediction_horizon']
TypeError: string indices must be integers
ERROR:web_server:Exception on /action/naive-mpc-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 136, in action_call
    opt_res = naive_mpc_optim(input_data_dict, app.logger)
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 185, in naive_mpc_optim
    prediction_horizon = input_data_dict['params']['passed_data']['prediction_horizon']
TypeError: string indices must be integers
```

davidusb-geek, 2022-05-17:
Hi, this specific issue was solved. Other issues arised in the mean time. They are now normally solved. I've tested the MPC on the add-on and is working fine on my side.
Also the `prediction_horizon` parameter is an integer with the number of time steps to use for the MPC iteration. I've set the default to 10 for this.
The following curl command is working fine for me:
```
curl -i -H "Content-Type: application/json" -X POST -d '{"pv_power_forecast":[0, 70, 141.22, 246.18, 513.5, 753.27, 1049.89, 1797.93, 1697.3, 3078.93], "prediction_horizon":10, "soc_init":0.5,"soc_final":0.6,"def_total_hours":[1,3]}' http://localhost:5000/action/naive-mpc-optim
```

purcell-lab, 2022-05-18:
Thanks, the prediction_horizon is working very well for me as I don't always have forecasts for the next 48 time steps.

[image]

The below scripts work for minutes 26-29 & minutes 46-59, due to the way emhass is rounding the forecasts to the nearest time block.  I'll raise a separate issue. Ideally emhass would always receive currentvalue & forecasts for each.

```
"soc_init":{{states('sensor.powerwall_charge')|float(0)/100}}
"prediction_horizon":{{(state_attr('sensor.amber_feed_in_forecast', 'forecasts')|map(attribute='per_kwh')|list|length)}}
"load_cost_forecast":{{(state_attr('sensor.amber_general_forecast', 'forecasts') |map(attribute='per_kwh')|list)}}
"prod_price_forecast":{{(state_attr('sensor.amber_feed_in_forecast', 'forecasts')|map(attribute='per_kwh')|list)}}

curl -i -H "Content-Type: application/json" -X POST -d '{"prod_price_forecast":{{(
             (state_attr('sensor.amber_feed_in_forecast', 'forecasts')|map(attribute='per_kwh')|list) )
          }},"load_cost_forecast":{{(
             (state_attr('sensor.amber_general_forecast', 'forecasts') |map(attribute='per_kwh')|list) )
          }},"prediction_horizon":{{(state_attr('sensor.amber_feed_in_forecast', 'forecasts')|map(attribute='per_kwh')|list|length)
          }},"soc_init":{{states('sensor.powerwall_charge')|float(0)/100
          }},"soc_final":0.01,"def_total_hours":[4,5,3,1]}' http://localhost:5000/action/naive-mpc-optim
```

```
"soc_init":0.49
"prediction_horizon":32
"load_cost_forecast":[0.25, 0.21, 0.22, 0.28, 0.28, 0.33, 0.42, 0.33, 0.33, 0.61, 1.49, 15.36, 1.49, 0.79, 0.72, 0.61, 0.47, 0.45, 0.45, 0.45, 0.45, 0.45, 0.44, 0.43, 0.44, 0.38, 0.37, 0.42, 0.38, 0.38, 0.37, 0.38]
"prod_price_forecast":[0.14, 0.1, 0.11, 0.16, 0.16, 0.21, 0.3, 0.21, 0.21, 0.47, 1.26, 13.88, 1.26, 0.63, 0.57, 0.47, 0.34, 0.33, 0.32, 0.32, 0.32, 0.32, 0.31, 0.31, 0.31, 0.26, 0.25, 0.29, 0.25, 0.25, 0.25, 0.26]

curl -i -H "Content-Type: application/json" -X POST -d '{"prod_price_forecast":[0.14, 0.1, 0.11, 0.16, 0.16, 0.21, 0.3, 0.21, 0.21, 0.47, 1.26, 13.88, 1.26, 0.63, 0.57, 0.47, 0.34, 0.33, 0.32, 0.32, 0.32, 0.32, 0.31, 0.31, 0.31, 0.26, 0.25, 0.29, 0.25, 0.25, 0.25, 0.26],"load_cost_forecast":[0.25, 0.21, 0.22, 0.28, 0.28, 0.33, 0.42, 0.33, 0.33, 0.61, 1.49, 15.36, 1.49, 0.79, 0.72, 0.61, 0.47, 0.45, 0.45, 0.45, 0.45, 0.45, 0.44, 0.43, 0.44, 0.38, 0.37, 0.42, 0.38, 0.38, 0.37, 0.38],"prediction_horizon":32,"soc_init":0.49,"soc_final":0.01,"def_total_hours":[4,5,3,1]}' http://localhost:5000/action/naive-mpc-optim
```

closed 2022-05-18 by purcell-lab