davidusb-geek/emhass-add-on #14: Duplicate entries in log window

purcell-lab, 2022-05-18:
I am seeing duplicate lines, one with time stamp and one without for the INFO logging entries.

```
[2022-05-19 07:31:13,737] INFO in optimization: Total value of the Cost function = -10.1
INFO:web_server:Total value of the Cost function = -10.1
```

Is it possible to only display each entry once?

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
[2022-05-19 07:30:45,287] INFO in web_server: Launching the emhass webserver at: http://0.0.0.0:5000
[2022-05-19 07:30:45,287] INFO in web_server: Home Assistant data fetch will be performed using url: http://supervisor/core/api
[2022-05-19 07:30:45,288] INFO in web_server: The base path is: /usr/src
[2022-05-19 07:30:45,292] INFO in web_server: Using core emhass version: 0.3.8
[2022-05-19 07:31:06,375] INFO in command_line: Setting up needed data
INFO:web_server:Setting up needed data
[2022-05-19 07:31:06,431] INFO in forecast: Retrieving weather forecast data using method = scrapper
INFO:web_server:Retrieving weather forecast data using method = scrapper
[2022-05-19 07:31:09,662] INFO in forecast: Retrieving data from hass for load forecast using method = naive
INFO:web_server:Retrieving data from hass for load forecast using method = naive
[2022-05-19 07:31:09,663] INFO in retrieve_hass: Retrieve hass get data method initiated...
INFO:web_server:Retrieve hass get data method initiated...
[2022-05-19 07:31:13,392] INFO in web_server:  >> Performing naive MPC optimization...
INFO:web_server: >> Performing naive MPC optimization...
[2022-05-19 07:31:13,393] INFO in command_line: Performing naive MPC optimization
INFO:web_server:Performing naive MPC optimization
[2022-05-19 07:31:13,416] INFO in optimization: Perform an iteration of a naive MPC controller
INFO:web_server:Perform an iteration of a naive MPC controller
[2022-05-19 07:31:13,563] WARNING in optimization: Failed LP solve with PULP_CBC_CMD solver, falling back to default Pulp
WARNING:web_server:Failed LP solve with PULP_CBC_CMD solver, falling back to default Pulp
[2022-05-19 07:31:13,616] WARNING in optimization: Failed LP solve with default Pulp solver, falling back to GLPK_CMD
WARNING:web_server:Failed LP solve with default Pulp solver, falling back to GLPK_CMD
[2022-05-19 07:31:13,737] INFO in optimization: Status: Optimal
INFO:web_server:Status: Optimal
[2022-05-19 07:31:13,737] INFO in optimization: Total value of the Cost function = -10.1
INFO:web_server:Total value of the Cost function = -10.1

davidusb-geek, 2022-05-19:
Hi. Just published a new version. This should be solved.

purcell-lab, 2022-05-19:
Thanks All good now.

A lot of work for you in the last day.

```
[2022-05-20 08:20:09,040] INFO in retrieve_hass: Successfully posted to sensor.p_pv_forecast = 2235.0
[2022-05-20 08:20:09,086] INFO in retrieve_hass: Successfully posted to sensor.p_load_forecast = 2289.0
[2022-05-20 08:20:09,111] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable0 = 1500.0
[2022-05-20 08:20:09,130] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable1 = 5500.0
[2022-05-20 08:20:09,152] INFO in retrieve_hass: Successfully posted to sensor.p_batt_forecast = -4547.0
[2022-05-20 08:25:00,090] INFO in command_line: Setting up needed data
[2022-05-20 08:25:00,095] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2022-05-20 08:25:03,381] INFO in forecast: Retrieving data from hass for load forecast using method = naive
[2022-05-20 08:25:03,383] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-05-20 08:25:07,172] INFO in web_server:  >> Performing naive MPC optimization...
[2022-05-20 08:25:07,172] INFO in command_line: Performing naive MPC optimization
[2022-05-20 08:25:07,194] INFO in optimization: Perform an iteration of a naive MPC controller
[2022-05-20 08:25:07,324] WARNING in optimization: Failed LP solve with PULP_CBC_CMD solver, falling back to default Pulp
[2022-05-20 08:25:07,370] WARNING in optimization: Failed LP solve with default Pulp solver, falling back to GLPK_CMD
[2022-05-20 08:25:07,741] INFO in optimization: Status: Optimal
[2022-05-20 08:25:07,742] INFO in optimization: Total value of the Cost function = -13.53
[2022-05-20 08:25:08,147] INFO in command_line: Setting up needed data
[2022-05-20 08:25:08,158] INFO in web_server:  >> Publishing data...
[2022-05-20 08:25:08,158] INFO in command_line: Publishing data to HASS instance
[2022-05-20 08:25:08,216] INFO in retrieve_hass: Successfully posted to sensor.p_pv_forecast = 2235.0
[2022-05-20 08:25:08,254] INFO in retrieve_hass: Successfully posted to sensor.p_load_forecast = 2289.0
[2022-05-20 08:25:08,286] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable0 = 1500.0
[2022-05-20 08:25:08,323] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable1 = 5500.0
[2022-05-20 08:25:08,370] INFO in retrieve_hass: Successfully posted to sensor.p_batt_forecast = -4832.0
[2022-05-20 08:26:00,080] INFO in command_line: Setting up needed data
[2022-05-20 08:26:00,085] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2022-05-20 08:26:03,310] INFO in forecast: Retrieving data from hass for load forecast using method = naive
[2022-05-20 08:26:03,312] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-05-20 08:26:06,417] INFO in web_server:  >> Performing naive MPC optimization...
[2022-05-20 08:26:06,417] INFO in command_line: Performing naive MPC optimization
[2022-05-20 08:26:06,439] INFO in optimization: Perform an iteration of a naive MPC controller
[2022-05-20 08:26:06,578] WARNING in optimization: Failed LP solve with PULP_CBC_CMD solver, falling back to default Pulp
[2022-05-20 08:26:06,631] WARNING in optimization: Failed LP solve with default Pulp solver, falling back to GLPK_CMD
[2022-05-20 08:26:07,003] INFO in optimization: Status: Optimal
[2022-05-20 08:26:07,004] INFO in optimization: Total value of the Cost function = -13.53
[2022-05-20 08:26:07,403] INFO in command_line: Setting up needed data
[2022-05-20 08:26:07,414] INFO in web_server:  >> Publishing data...
[2022-05-20 08:26:07,414] INFO in command_line: Publishing data to HASS instance
[2022-05-20 08:26:07,482] INFO in retrieve_hass: Successfully posted to sensor.p_pv_forecast = 2235.0
[2022-05-20 08:26:07,516] INFO in retrieve_hass: Successfully posted to sensor.p_load_forecast = 2289.0
[2022-05-20 08:26:07,548] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable0 = 1500.0
[2022-05-20 08:26:07,580] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable1 = 5500.0
[2022-05-20 08:26:07,625] INFO in retrieve_hass: Successfully posted to sensor.p_batt_forecast = -4832.0
[2022-05-20 08:27:00,077] INFO in command_line: Setting up needed data
[2022-05-20 08:27:00,081] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2022-05-20 08:27:03,297] INFO in forecast: Retrieving data from hass for load forecast using method = naive
[2022-05-20 08:27:03,298] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-05-20 08:27:07,253] INFO in web_server:  >> Performing naive MPC optimization...
[2022-05-20 08:27:07,253] INFO in command_line: Performing naive MPC optimization
[2022-05-20 08:27:07,275] INFO in optimization: Perform an iteration of a naive MPC controller
[2022-05-20 08:27:07,404] WARNING in optimization: Failed LP solve with PULP_CBC_CMD solver, falling back to default Pulp
[2022-05-20 08:27:07,466] WARNING in optimization: Failed LP solve with default Pulp solver, falling back to GLPK_CMD
[2022-05-20 08:27:07,830] INFO in optimization: Status: Optimal
[2022-05-20 08:27:07,831] INFO in optimization: Total value of the Cost function = -13.47
[2022-05-20 08:27:08,212] INFO in command_line: Setting up needed data
[2022-05-20 08:27:08,216] INFO in web_server:  >> Publishing data...
[2022-05-20 08:27:08,217] INFO in command_line: Publishing data to HASS instance
[2022-05-20 08:27:08,267] INFO in retrieve_hass: Successfully posted to sensor.p_pv_forecast = 2235.0
[... cut ...]
[2022-05-20 08:29:06,612] INFO in retrieve_hass: Successfully posted to sensor.p_pv_forecast = 2235.0
[2022-05-20 08:29:06,642] INFO in retrieve_hass: Successfully posted to sensor.p_load_forecast = 2289.0
[2022-05-20 08:29:06,670] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable0 = 1500.0
[2022-05-20 08:29:06,700] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable1 = 5500.0
[2022-05-20 08:29:06,732] INFO in retrieve_hass: Successfully posted to sensor.p_batt_forecast = -4547.0

closed 2022-05-19 by purcell-lab