davidusb-geek/emhass-add-on #15: publish-data - not loading forecast data into HA

kcoffau, 2022-06-02:
Disclaimer: This could be a user issue!

EMHASS Add-on installed via Addon Store on HA OS running on a VM

Added this info config YAML
shell_command:
  dayahead_optim: "curl -i -H \"Content-Type: application/json\" -X POST -d '{}' http://localhost:5000/action/dayahead-optim"
  publish_data: "curl -i -H \"Content-Type: application/json\" -X POST -d '{}' http://localhost:5000/action/publish-data "

Automations in place to call these.

Run the automation, I can observe via web http://HA:5000 that the optimisation process has run.

**unsure on how to check if publish worked**

Entities show sensor.p_......  show historical values.     When I run a simple APEX card for 24 hours in future, there is no data.

I use Apex cards for other data, future dated, works OK (non EMHASS related).

I have not created the /home/user/emhass/..... (I cant get to shell of the HA).

questions:
1. as I did the Add on install, was I supposed to create the Config, Secrets, Data, scripts files and folders?  If so... how!
2. How can I check if the publish process actually worked? Is there a better way?  Seems to be working as the history seems to load.
3. have I missed something, done something silly?

davidusb-geek, 2022-06-02:
> questions:
>
> 1. as I did the Add on install, was I supposed to create the Config, Secrets, Data, scripts files and folders? If so... how!
> 2. How can I check if the publish process actually worked? Is [...]

Hi, answering to these questions

1. No absolutely not, if you're using the add-on you have nothing to do with all those instructions. Just follow the add-on install instructions and that's it. Config, Secrets, Data, scripts files and folders are for installing EMHASS using classic Python virtual environment. 
2. If publish data as worked you can see the correct logs on the add-on Log tab. You should see something like this:

> [2022-06-02 12:45:00,043] INFO in web_server:  >> Publishing data...
> [2022-06-02 12:45:00,043] INFO in command_line: Publishing data to HASS instance
> [2022-06-02 12:45:00,072] INFO in retrieve_hass: Successfully posted to sensor.p_pv_forecast = 1537.4
> [2022-06-02 12:45:00,089] INFO in retrieve_hass: Successfully posted to sensor.p_load_forecast = 233.49
> [2022-06-02 12:45:00,106] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable0 = 0.0
> [2022-06-02 12:45:00,121] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable1 = 750.0

In this case if you go to Home Assistant > Developer Tools > States, then you can search for say `sensor.p_pv_forecast` and see the future dates and values as attributes to this sensor.

If the publish data is not working then we can know why by looking at the logged messages. Can you share those?

3. I see that you may have messed with different installation methods. In any case the add-on is the simplest, you don't need to open any console to make it work.

kcoffau, 2022-06-02:
Hi David,

*facepalm*. Completely forgot about states in Dev.  I was doing it the hard way.

Data below, thank you for your confirmation and patience. 

Clearly from below the publish worked and the state has the future data....  Looks like the issue was PICNIC and I need to check what Ive done wrong elsewhere!

This is an amazing application that really takes energy automation to a new level in HA!

Thank you!

Cheers,
KC

Run Publish:
[2022-06-02 21:55:00,038] INFO in command_line: Setting up needed data
[2022-06-02 21:55:00,044] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2022-06-02 21:55:02,323] INFO in forecast: Retrieving data from hass for load forecast using method = naive
[2022-06-02 21:55:02,325] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-06-02 21:55:06,150] INFO in web_server:  >> Performing naive MPC optimization...
[2022-06-02 21:55:06,150] INFO in command_line: Performing naive MPC optimization
[2022-06-02 21:55:06,168] INFO in optimization: Perform an iteration of a naive MPC controller
[2022-06-02 21:55:06,340] INFO in optimization: Status: Infeasible
[2022-06-02 21:55:06,341] INFO in optimization: Total value of the Cost function = 15.69
[2022-06-02 21:55:06,873] INFO in command_line: Setting up needed data
[2022-06-02 21:55:06,879] INFO in web_server:  >> Publishing data...
[2022-06-02 21:55:06,879] INFO in command_line: Publishing data to HASS instance
[2022-06-02 21:55:06,917] INFO in retrieve_hass: Successfully posted to sensor.p_pv_forecast = -1.5
[2022-06-02 21:55:06,939] INFO in retrieve_hass: Successfully posted to sensor.p_load_forecast = 3274.99
[2022-06-02 21:55:06,960] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable0 = 0.0
[2022-06-02 21:55:06,980] INFO in retrieve_hass: Successfully posted to sensor.p_batt_forecast = -19393.94
[2022-06-02 21:55:07,004] INFO in retrieve_hass: Successfully posted to sensor.soc_batt_forecast = 30.0
[2022-06-02 21:58:50,128] INFO in web_server: EMHASS server online, serving index.html...
[2022-06-02 21:58:57,770] INFO in command_line: Setting up needed data
[2022-06-02 21:58:57,774] INFO in web_server:  >> Publishing data...
[2022-06-02 21:58:57,774] INFO in command_line: Publishing data to HASS instance
[2022-06-02 21:58:57,805] INFO in retrieve_hass: Successfully posted to sensor.p_pv_forecast = -1.5
[2022-06-02 21:58:57,827] INFO in retrieve_hass: Successfully posted to sensor.p_load_forecast = 3274.99
[2022-06-02 21:58:57,848] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable0 = 0.0
[2022-06-02 21:58:57,869] INFO in retrieve_hass: Successfully posted to sensor.p_batt_forecast = -19393.94
[2022-06-02 21:58:57,891] INFO in retrieve_hass: Successfully posted to sensor.soc_batt_forecast = 30.0

States:

sensor.p_pv_forecast
unit_of_measurement: W
friendly_name: PV Power Forecast
forecasts: 
- date: '2022-06-02 22:00:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-02 22:30:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-02 23:00:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-02 23:30:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 00:00:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 00:30:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 01:00:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 01:30:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 02:00:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 02:30:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 03:00:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 03:30:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 04:00:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 04:30:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 05:00:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 05:30:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 06:00:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 06:30:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 07:00:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 07:30:00+10:00'
  p_pv_forecast: 70
- date: '2022-06-03 08:00:00+10:00'
  p_pv_forecast: 269
- date: '2022-06-03 08:30:00+10:00'
  p_pv_forecast: 471
- date: '2022-06-03 09:00:00+10:00'
  p_pv_forecast: 656
- date: '2022-06-03 09:30:00+10:00'
  p_pv_forecast: 816
- date: '2022-06-03 10:00:00+10:00'
  p_pv_forecast: 954
- date: '2022-06-03 10:30:00+10:00'
  p_pv_forecast: 1055
- date: '2022-06-03 11:00:00+10:00'
  p_pv_forecast: 1124
- date: '2022-06-03 11:30:00+10:00'
  p_pv_forecast: 1161
- date: '2022-06-03 12:00:00+10:00'
  p_pv_forecast: 1165
- date: '2022-06-03 12:30:00+10:00'
  p_pv_forecast: 1157
- date: '2022-06-03 13:00:00+10:00'
  p_pv_forecast: 1116
- date: '2022-06-03 13:30:00+10:00'
  p_pv_forecast: 1013
- date: '2022-06-03 14:00:00+10:00'
  p_pv_forecast: 881
- date: '2022-06-03 14:30:00+10:00'
  p_pv_forecast: 1245
- date: '2022-06-03 15:00:00+10:00'
  p_pv_forecast: 1485
- date: '2022-06-03 15:30:00+10:00'
  p_pv_forecast: 653
- date: '2022-06-03 16:00:00+10:00'
  p_pv_forecast: 177
- date: '2022-06-03 16:30:00+10:00'
  p_pv_forecast: 3
- date: '2022-06-03 17:00:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 17:30:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 18:00:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 18:30:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 19:00:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 19:30:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 20:00:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 20:30:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 21:00:00+10:00'
  p_pv_forecast: -2
- date: '2022-06-03 21:30:00+10:00'
  p_pv_forecast: -2