davidusb-geek/emhass-add-on #16: TypeError: Object of type int64 is not JSON serializable input_data_dict['rh'].post_data(opt_res_latest['P_PV']

purcell-lab, 2022-06-05:
Evening,

When posting my own pv_forecast with mpc_opt.

I get a valid optimisation, which is unable to be published??

```
mark@odroid:~$ curl -i -H "Content-Type: application/json" -X POST -d '{"load_cost_forecast":[0.5, 0.52, 0.55, 0.51, 0.44, 0.43, 0.43, 0.43, 0.43, 0.43, 0.42, 0.39, 0.24, 0.21, 0.17, 0.24, 0.24, 0.42, 0.44, 0.72, 0.77, 0.7, 0.61, 0.42, 0.52, 0.49, 0.49, 0.43, 0.43, 0.31, 0.21, 0.2, 0.17, 0.36, 0.43, 0.43, 0.43, 0.44, 0.53, 0.64, 0.72, 1.66, 1.3, 0.9, 0.72, 0.65, 0.64, 0.63], "prod_price_forecast":[0.37, 0.39, 0.42, 0.38, 0.31, 0.31, 0.31, 0.31, 0.3, 0.31, 0.29, 0.27, 0.13, 0.11, 0.06, 0.13, 0.13, 0.3, 0.31, 0.57, 0.61, 0.55, 0.46, 0.3, 0.38, 0.35, 0.36, 0.31, 0.31, 0.19, 0.1, 0.1, 0.06, 0.24, 0.31, 0.31, 0.31, 0.31, 0.39, 0.5, 0.57, 1.42, 1.1, 0.73, 0.57, 0.5, 0.5, 0.48], "pv_power_forecast":[-9, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 36, 204, 361, 571, 1008, 1497, 2130, 2532, 2947, 2823, 2603, 2246, 1909, 1835, 1730, 1458, 1184, 824, 473, 309, 74, 0, 0, 0, 0, 0, 0], "prediction_horizon":48,"soc_init":0.28,"soc_final":0.05,"def_total_hours":[4,1,0,0]}' http://localhost:5000/action/naive-mpc-optim
HTTP/1.1 201 CREATED
Content-Length: 46
Content-Type: text/html; charset=utf-8
Date: Sun, 05 Jun 2022 11:01:10 GMT
Server: waitress

EMHASS >> Action naive-mpc-optim executed... 
mark@odroid:~$ 
```

```

index | P_PV | P_Load | P_deferrable0 | P_deferrable1 | P_deferrable2 | P_deferrable3 | P_grid_pos | P_grid_neg | P_grid | P_batt | SOC_opt | unit_load_cost | unit_prod_price | cost_profit | cost_fun_profit
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
2022-06-05 21:00:00+10:00 | -9 | 1946.845714 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | 1955.845700 | 0.203749 | 0.50 | 0.37 | -0.000000 | -0.000000
2022-06-05 21:30:00+10:00 | 0 | 708.769697 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | 708.769700 | 0.176116 | 0.52 | 0.39 | -0.000000 | -0.000000
2022-06-05 22:00:00+10:00 | 0 | 333.208633 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | 333.208630 | 0.163126 | 0.55 | 0.42 | -0.000000 | -0.000000
2022-06-05 22:30:00+10:00 | 0 | 314.302158 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | 314.302160 | 0.150872 | 0.51 | 0.38 | -0.000000 | -0.000000
2022-06-05 23:00:00+10:00 | 0 | 317.472441 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | 317.472440 | 0.138495 | 0.44 | 0.31 | -0.000000 | -0.000000
2022-06-05 23:30:00+10:00 | 0 | 249.879433 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | 249.879430 | 0.128753 | 0.43 | 0.31 | -0.000000 | -0.000000
2022-06-06 00:00:00+10:00 | 0 | 350.214286 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | 350.214290 | 0.115100 | 0.43 | 0.31 | -0.000000 | -0.000000
2022-06-06 00:30:00+10:00 | 0 | 273.406015 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | 273.406020 | 0.104441 | 0.43 | 0.31 | -0.000000 | -0.000000
2022-06-06 01:00:00+10:00 | 0 | 304.934307 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | 304.934310 | 0.092552 | 0.43 | 0.30 | -0.000000 | -0.000000
2022-06-06 01:30:00+10:00 | 0 | 251.008264 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | 251.008260 | 0.082766 | 0.43 | 0.31 | -0.000000 | -0.000000
2022-06-06 02:00:00+10:00 | 0 | 310.043478 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | 310.043480 | 0.070679 | 0.42 | 0.29 | -0.000000 | -0.000000
2022-06-06 02:30:00+10:00 | 0 | 302.710280 | 0.0 | 0.0 | 0.0 | 0.0 | 302.71028 | 0.00000 | 302.71028 | 0.000000 | 0.070679 | 0.39 | 0.27 | -0.059029 | -0.059029
2022-06-06 03:00:00+10:00 | 0 | 256.774194 | 1500.0 | 0.0 | 0.0 | 0.0 | 6756.77420 | 0.00000 | 6756.77420 | -5000.000000 | 0.246605 | 0.24 | 0.13 | -0.810813 | -0.810813
2022-06-06 03:30:00+10:00 | 0 | 315.753731 | 1500.0 | 0.0 | 0.0 | 0.0 | 6815.75370 | 0.00000 | 6815.75370 | -5000.000000 | 0.422531 | 0.21 | 0.11 | -0.715654 | -0.715654
2022-06-06 04:00:00+10:00 | 0 | 316.470588 | 1500.0 | 5500.0 | 0.0 | 0.0 | 12316.47100 | 0.00000 | 12316.47100 | -5000.000000 | 0.598457 | 0.17 | 0.06 | -1.046900 | -1.046900
2022-06-06 04:30:00+10:00 | 0 | 263.228814 | 1500.0 | 0.0 | 0.0 | 0.0 | 6763.22880 | 0.00000 | 6763.22880 | -5000.000000 | 0.774383 | 0.24 | 0.13 | -0.811587 | -0.811587
2022-06-06 05:00:00+10:00 | 0 | 1068.586207 | 1500.0 | 0.0 | 0.0 | 0.0 | 7568.58620 | 0.00000 | 7568.58620 | -5000.000000 | 0.950309 | 0.24 | 0.13 | -0.908230 | -0.908230
2022-06-06 05:30:00+10:00 | 0 | 3223.010471 | 0.0 | 0.0 | 0.0 | 0.0 | 1863.24090 | 0.00000 | 1863.24090 | 1359.769500 | 0.897296 | 0.42 | 0.30 | -0.391281 | -0.391281
2022-06-06 06:00:00+10:00 | 0 | 1674.293103 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | 1674.293100 | 0.832022 | 0.44 | 0.31 | -0.000000 | -0.000000
2022-06-06 06:30:00+10:00 | 0 | 1493.184713 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | -3507.61530 | -3507.61530 | 5000.800000 | 0.637059 | 0.72 | 0.57 | 0.999670 | 0.999670
2022-06-06 07:00:00+10:00 | 0 | 1622.557895 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | -3378.24210 | -3378.24210 | 5000.800000 | 0.442096 | 0.77 | 0.61 | 1.030364 | 1.030364
2022-06-06 07:30:00+10:00 | 36 | 958.800000 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | -4078.00000 | -4078.00000 | 5000.800000 | 0.247133 | 0.70 | 0.55 | 1.121450 | 1.121450
2022-06-06 08:00:00+10:00 | 204 | 1324.479798 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | -3880.32020 | -3880.32020 | 5000.800000 | 0.052170 | 0.61 | 0.46 | 0.892474 | 0.892474
2022-06-06 08:30:00+10:00 | 361 | 1340.185430 | 0.0 | 0.0 | 0.0 | 0.0 | 979.18543 | 0.00000 | 979.18543 | 0.000000 | 0.052170 | 0.42 | 0.30 | -0.205629 | -0.205629
2022-06-06 09:00:00+10:00 | 571 | 626.652941 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | 55.652941 | 0.050000 | 0.52 | 0.38 | -0.000000 | -0.000000
2022-06-06 09:30:00+10:00 | 1008 | 340.121622 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | -667.87838 | -667.87838 | 0.000000 | 0.050000 | 0.49 | 0.35 | 0.116879 | 0.116879
2022-06-06 10:00:00+10:00 | 1497 | 550.753425 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | -946.24658 | -946.24658 | 0.000000 | 0.050000 | 0.49 | 0.36 | 0.170324 | 0.170324
2022-06-06 10:30:00+10:00 | 2130 | 593.021277 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | -1536.978700 | 0.104079 | 0.43 | 0.31 | -0.000000 | -0.000000
2022-06-06 11:00:00+10:00 | 2532 | 452.708861 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | -2079.291100 | 0.177239 | 0.43 | 0.31 | -0.000000 | -0.000000
2022-06-06 11:30:00+10:00 | 2947 | 458.083333 | 0.0 | 0.0 | 0.0 | 0.0 | 1064.88880 | 0.00000 | 1064.88880 | -3553.805500 | 0.302280 | 0.31 | 0.19 | -0.165058 | -0.165058
2022-06-06 12:00:00+10:00 | 2823 | 1121.208333 | 1500.0 | 0.0 | 0.0 | 0.0 | 4798.20830 | 0.00000 | 4798.20830 | -5000.000000 | 0.478206 | 0.21 | 0.10 | -0.503812 | -0.503812
2022-06-06 12:30:00+10:00 | 2603 | 5027.729730 | 1500.0 | 0.0 | 0.0 | 0.0 | 8924.72970 | 0.00000 | 8924.72970 | -5000.000000 | 0.654132 | 0.20 | 0.10 | -0.892473 | -0.892473
2022-06-06 13:00:00+10:00 | 2246 | 529.109091 | 1500.0 | 5500.0 | 0.0 | 0.0 | 10283.10900 | 0.00000 | 10283.10900 | -5000.000000 | 0.830058 | 0.17 | 0.06 | -0.874064 | -0.874064
2022-06-06 13:30:00+10:00 | 1909 | 288.527197 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | -1620.472800 | 0.887075 | 0.36 | 0.24 | -0.000000 | -0.000000
2022-06-06 14:00:00+10:00 | 1835 | 411.375000 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | -1423.625000 | 0.937165 | 0.43 | 0.31 | -0.000000 | -0.000000
2022-06-06 14:30:00+10:00 | 1730 | 1390.769841 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | -339.230160 | 0.949101 | 0.43 | 0.31 | -0.000000 | -0.000000
2022-06-06 15:00:00+10:00 | 1458 | 498.605405 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | -959.394590 | 0.982858 | 0.43 | 0.31 | -0.000000 | -0.000000
2022-06-06 15:30:00+10:00 | 1184 | 696.797927 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | -487.202070 | 1.000000 | 0.44 | 0.31 | -0.000000 | -0.000000
2022-06-06 16:00:00+10:00 | 824 | 718.871921 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | -105.12808 | -105.12808 | 0.000000 | 1.000000 | 0.53 | 0.39 | 0.020500 | 0.020500
2022-06-06 16:30:00+10:00 | 473 | 582.425806 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | 0.00000 | 0.00000 | 109.425810 | 0.995734 | 0.64 | 0.50 | -0.000000 | -0.000000
2022-06-06 17:00:00+10:00 | 309 | 1116.168919 | 0.0 | 0.0 | 0.0 | 0.0 | 0.00000 | -4193.63110 | -4193.63110 | 5000.800000 | 0.800771 | 0.72 | 0.57 | 1.195185 | 1.195185
[... cut ...]
  File "/usr/lib/python3.9/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/usr/lib/python3.9/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/lib/python3.9/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/usr/lib/python3.9/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
TypeError: Object of type int64 is not JSON serializable
```

davidusb-geek, 2022-06-05:
Very strange. I'm not able to reproduce the error. The unittests are passing correctly. Those serialization errors with json and np types are a bit obscure to me. But there is a simple fix to avoid this type of error in any case. I'll apply that as a fix in the next version.

davidusb-geek, 2022-06-07:
Hi, did you managed to test this? This should have been solved with the latest versions.

davidusb-geek, 2022-06-07:
Closing as it seems to be solved.

closed 2022-06-07 by davidusb-geek

purcell-lab, 2022-06-07:
Thanks.  All good.