davidusb-geek/emhass-add-on #17: KeyError: 'sensor.power_load_no_var_loads'

purcell-lab, 2022-06-10:
I don't seem to be paassing the now variables correctly.

```
2022-06-11 06:53:04,172] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-06-11 06:53:06,065] ERROR in app: Exception on /action/naive-mpc-optim [POST]
Traceback (most recent call last):
  File "/root/.local/lib/python3.9/site-packages/pandas/core/indexes/base.py", line 3621, in get_loc
    return self._engine.get_loc(casted_key)
  File "pandas/_libs/index.pyx", line 136, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/index.pyx", line 163, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 5198, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 5206, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'sensor.power_load_no_var_loads'

davidusb-geek, 2022-06-10:
What curl command did you use when you obtained this error? What runtime data are you passing when executing the command?

purcell-lab, 2022-06-10:
I hadn't changed my curl command.  

Do I need to include the now values on the curl commend line as well?

Which runtime parameter label should I use?

I have tried:

```
mark@odroid:~$ curl -i -H "Content-Type: application/json" -X POST -d '{"load_cost_forecast":[0.48, 0.47, 0.61, 0.61, 0.52, 0.42, 0.42, 0.27, 0.27, 0.38, 0.42, 0.32, 0.46, 0.7, 0.72, 0.78, 11.66, 15.36, 15.36, 15.36, 14.46, 1.65, 1.58, 0.84, 0.82, 0.78, 0.84, 0.82, 0.78, 0.77, 0.78, 0.74, 0.48, 0.43, 0.46, 0.43, 0.42, 0.4], "prod_price_forecast":[0.35, 0.34, 0.46, 0.46, 0.39, 0.29, 0.29, 0.16, 0.16, 0.26, 0.29, 0.21, 0.33, 0.55, 0.57, 0.62, 10.51, 13.88, 13.88, 13.88, 13.06, 1.41, 1.35, 0.67, 0.66, 0.62, 0.67, 0.65, 0.62, 0.61, 0.62, 0.58, 0.35, 0.31, 0.33, 0.31, 0.3, 0.28], "power_load_no_var_loads":738, "power_photovoltaics":6610, "prediction_horizon":38, "soc_init":0.12,"soc_final":0.05, "def_total_hours":[4,1,0,0]}' http://localhost:5000/action/naive-mpc-optim
HTTP/1.1 500 INTERNAL SERVER ERROR
Content-Length: 265
Content-Type: text/html; charset=utf-8
Date: Fri, 10 Jun 2022 23:12:07 GMT
Server: waitress

<!doctype html>
<html lang=en>
<title>500 Internal Server Error</title>
<h1>Internal Server Error</h1>
<p>The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.</p>
mark@odroid:~$

davidusb-geek, 2022-06-11:
No, you should not pass those parameters. At runtime the parameters are the same, I just added two new parameters that you can pass: alpha and beta.
The thing is that the original error message is not explicitly telling where the error is in the code. I will see if I'm able to reproduce it.

The now values PV and Load forecast are retrieved automatically from Home Assistant, no need to pass them at runtime.

purcell-lab, 2022-06-11:
Thanks,

When I pass no values I get that error with mpc-optim, dayahead optim is still functional for me.

Let me know if I can provide any further debug info.

purcell-lab, 2022-06-12:
Have rolled back to 0.2.15 and working fine.

When I upgrade to 0.2.18, I get the errors.

Using the same mpc-optim call in curl.

```
curl -i -H "Content-Type: application/json" -X POST -d '{"load_cost_forecast":[0.46, 0.4, 0.42, 0.42, 0.42, 0.42, 0.42, 0.42, 0.42, 0.42, 0.69, 0.71, 0.62, 0.73, 0.93, 2.1, 11.08, 2.06, 1.25, 1.21, 1.27, 0.8, 0.84, 0.78, 0.78, 0.78, 0.76, 0.73, 0.74, 0.71, 0.7, 0.69, 0.61, 0.56, 0.56, 0.57], "prod_price_forecast":[0.33, 0.28, 0.3, 0.3, 0.3, 0.3, 0.3, 0.3, 0.3, 0.3, 0.54, 0.56, 0.48, 0.58, 0.76, 1.82, 9.99, 1.78, 1.05, 1.01, 1.06, 0.64, 0.67, 0.62, 0.62, 0.62, 0.61, 0.58, 0.58, 0.56, 0.55, 0.54, 0.47, 0.42, 0.42, 0.43], "prediction_horizon":36,"soc_init":0.19,"soc_final":0.05,"def_total_hours":[4,1,0,0]}' http://localhost:5000/action/naive-mpc-optim
```

* enhass-addon 0.2.15
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
[2022-06-12 10:16:06,873] INFO in web_server: Launching the emhass webserver at: http://0.0.0.0:5000
[2022-06-12 10:16:06,873] INFO in web_server: Home Assistant data fetch will be performed using url: http://supervisor/core/api
[2022-06-12 10:16:06,874] INFO in web_server: The base path is: /usr/src
[2022-06-12 10:16:06,878] INFO in web_server: Using core emhass version: 0.3.13
[2022-06-12 10:16:22,724] INFO in web_server: EMHASS server online, serving index.html...
[2022-06-12 10:17:00,078] INFO in command_line: Setting up needed data
[2022-06-12 10:17:00,131] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2022-06-12 10:17:03,357] INFO in forecast: Retrieving data from hass for load forecast using method = naive
[2022-06-12 10:17:03,359] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-06-12 10:17:05,333] INFO in web_server:  >> Performing naive MPC optimization...
[2022-06-12 10:17:05,333] INFO in command_line: Performing naive MPC optimization
[2022-06-12 10:17:05,357] INFO in optimization: Perform an iteration of a naive MPC controller
[2022-06-12 10:17:05,567] INFO in optimization: Status: Optimal
[2022-06-12 10:17:05,568] INFO in optimization: Total value of the Cost function = -10.24
[2022-06-12 10:17:06,191] INFO in command_line: Setting up needed data
[2022-06-12 10:17:06,204] INFO in web_server:  >> Publishing data...
[2022-06-12 10:17:06,205] INFO in command_line: Publishing data to HASS instance
[2022-06-12 10:17:06,255] INFO in retrieve_hass: Successfully posted to sensor.p_pv_forecast = 7863.19
[2022-06-12 10:17:06,292] INFO in retrieve_hass: Successfully posted to sensor.p_load_forecast = 465.16
[2022-06-12 10:17:06,328] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable0 = 1500.0
[2022-06-12 10:17:06,360] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable1 = 5500.0
[2022-06-12 10:17:06,398] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable2 = 0.0
[2022-06-12 10:17:06,435] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable3 = 0.0
[2022-06-12 10:17:06,484] INFO in retrieve_hass: Successfully posted to sensor.p_batt_forecast = -398.03
[2022-06-12 10:17:06,529] INFO in retrieve_hass: Successfully posted to sensor.soc_batt_forecast = 19.4
```

* emhass-addon 0.2.18
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
[2022-06-12 10:18:52,556] INFO in web_server: Launching the emhass webserver at: http://0.0.0.0:5000
[2022-06-12 10:18:52,556] INFO in web_server: Home Assistant data fetch will be performed using url: http://supervisor/core/api
[2022-06-12 10:18:52,556] INFO in web_server: The base path is: /usr/src
[2022-06-12 10:18:52,561] INFO in web_server: Using core emhass version: 0.3.16
[2022-06-12 10:19:00,081] INFO in command_line: Setting up needed data
[2022-06-12 10:19:00,145] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-06-12 10:19:01,039] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2022-06-12 10:19:04,335] ERROR in app: Exception on /action/naive-mpc-optim [POST]
Traceback (most recent call last):
  File "/root/.local/lib/python3.9/site-packages/pandas/core/indexes/base.py", line 3621, in get_loc
    return self._engine.get_loc(casted_key)
[... 16 lines cut ...]
    input_data_dict = set_input_data_dict(config_path, str(config_path.parent), costfun,
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 110, in set_input_data_dict
    P_PV_forecast = fcst.get_power_from_weather(df_weather, set_mix_forecast=True, df_now=df_input_data)
  File "/usr/local/lib/python3.9/dist-packages/emhass/forecast.py", line 351, in get_power_from_weather
[... cut ...]
[2022-06-12 10:19:04,617] ERROR in command_line: P_deferrable3 was not found in results DataFrame. Optimization task may need to be relaunched or it did not converged to a solution.
[2022-06-12 10:19:04,617] ERROR in command_line: P_batt was not found in results DataFrame. Optimization task may need to be relaunched or it did not converged to a solution.
[2022-06-12 10:19:04,649] INFO in retrieve_hass: Successfully posted value in a newly created entity_id
[2022-06-12 10:19:04,682] INFO in retrieve_hass: Successfully posted value in a newly created entity_id
```

purcell-lab, 2022-06-12:
emhass-addon 0.2.17, which uses core emhass 0.3.16, also has the error:

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
[2022-06-12 10:27:41,817] INFO in web_server: Launching the emhass webserver at: http://0.0.0.0:5000
[2022-06-12 10:27:41,817] INFO in web_server: Home Assistant data fetch will be performed using url: http://supervisor/core/api
[2022-06-12 10:27:41,817] INFO in web_server: The base path is: /usr/src
[2022-06-12 10:27:41,821] INFO in web_server: Using core emhass version: 0.3.16
[2022-06-12 10:28:00,091] INFO in command_line: Setting up needed data
[2022-06-12 10:28:00,148] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-06-12 10:28:00,976] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2022-06-12 10:28:04,264] ERROR in app: Exception on /action/naive-mpc-optim [POST]
Traceback (most recent call last):
  File "/root/.local/lib/python3.9/site-packages/pandas/core/indexes/base.py", line 3621, in get_loc
    return self._engine.get_loc(casted_key)
  File "pandas/_libs/index.pyx", line 136, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/index.pyx", line 163, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 5198, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 5206, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'sensor.power_photovoltaics'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 134, in action_call
    input_data_dict = set_input_data_dict(config_path, str(config_path.parent), costfun,
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 110, in set_input_data_dict
    P_PV_forecast = fcst.get_power_from_weather(df_weather, set_mix_forecast=True, df_now=df_input_data)
  File "/usr/local/lib/python3.9/dist-packages/emhass/forecast.py", line 351, in get_power_from_weather
    P_PV_forecast = forecast.get_mix_forecast(
  File "/usr/local/lib/python3.9/dist-packages/emhass/forecast.py", line 280, in get_mix_forecast
    first_fcst = alpha*df_forecast.iloc[0] + beta*df_now[col].iloc[-1]
  File "/root/.local/lib/python3.9/site-packages/pandas/core/frame.py", line 3505, in __getitem__
    indexer = self.columns.get_loc(key)
  File "/root/.local/lib/python3.9/site-packages/pandas/core/indexes/base.py", line 3623, in get_loc
    raise KeyError(key) from err
KeyError: 'sensor.power_photovoltaics'
[2022-06-12 10:28:04,352] INFO in command_line: Setting up needed data
[2022-06-12 10:28:04,355] INFO in web_server:  >> Publishing data...
[2022-06-12 10:28:04,355] INFO in command_line: Publishing data to HASS instance
[2022-06-12 10:28:04,401] INFO in retrieve_hass: Successfully posted to sensor.p_pv_forecast = 1391.13
[2022-06-12 10:28:04,437] INFO in retrieve_hass: Successfully posted to sensor.p_load_forecast = 169.94
[2022-06-12 10:28:04,481] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable0 = 0.0
[2022-06-12 10:28:04,526] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable1 = 0.0
[2022-06-12 10:28:04,527] ERROR in command_line: P_deferrable2 was not found in results DataFrame. Optimization task may need to be relaunched or it did not converged to a solution.
[2022-06-12 10:28:04,529] ERROR in command_line: P_deferrable3 was not found in results DataFrame. Optimization task may need to be relaunched or it did not converged to a solution.
[2022-06-12 10:28:04,531] ERROR in command_line: P_batt was not found in results DataFrame. Optimization task may need to be relaunched or it did not converged to a solution.
[2022-06-12 10:28:04,573] INFO in retrieve_hass: Successfully posted to sensor.p_grid_forecast = -1221.19
[2022-06-12 10:28:04,617] INFO in retrieve_hass: Successfully posted to sensor.total_cost_fun_value = -0.61
```

purcell-lab, 2022-06-12:
I suspect these two lines should have `sensor_` instead of `sensor.`

But, when I try to modify to test it downloads  the original again.

https://github.com/davidusb-geek/emhass/blob/f83d9067a1674fcb5afa6dbe6592a969daf54a4f/src/emhass/forecast.py#L354

https://github.com/davidusb-geek/emhass/blob/f83d9067a1674fcb5afa6dbe6592a969daf54a4f/src/emhass/forecast.py#L541

davidusb-geek, 2022-06-12:
Yes that should be it, I'll fix that as soon as I can. There are generic variables names for those sensors names, I just forgot to use them and my unittests are not catching this case.

davidusb-geek, 2022-06-12:
Ok, reproduced the error, updated the unittests, normally solved in the new release v0.2.19

purcell-lab, 2022-06-13:
Thanks,

Working well.

```
/usr/local/lib/python3.9/dist-packages/emhass/web_server.py:46: FutureWarning:
Dropping of nuisance columns in DataFrame reductions (with 'numeric_only=None') is deprecated; in a future version this will raise TypeError.  Select only valid columns before calling the reduction.
[2022-06-13 11:14:06,857] INFO in command_line: Setting up needed data
[2022-06-13 11:14:06,869] INFO in web_server:  >> Publishing data...
[2022-06-13 11:14:06,870] INFO in command_line: Publishing data to HASS instance
[2022-06-13 11:14:06,948] INFO in retrieve_hass: Successfully posted to sensor.p_pv_forecast = 9644.92
[2022-06-13 11:14:06,994] INFO in retrieve_hass: Successfully posted to sensor.p_load_forecast = 464.07
[2022-06-13 11:14:07,036] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable0 = 1500.0
[2022-06-13 11:14:07,075] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable1 = 5500.0
[2022-06-13 11:14:07,113] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable2 = 0.0
[2022-06-13 11:14:07,157] INFO in retrieve_hass: Successfully posted to sensor.p_deferrable3 = 0.0
[2022-06-13 11:14:07,199] INFO in retrieve_hass: Successfully posted to sensor.p_batt_forecast = -2180.85
[2022-06-13 11:14:07,245] INFO in retrieve_hass: Successfully posted to sensor.soc_batt_forecast = 77.67
[2022-06-13 11:14:07,309] INFO in retrieve_hass: Successfully posted to sensor.p_grid_forecast = 0.0
[2022-06-13 11:14:07,364] INFO in retrieve_hass: Successfully posted to sensor.total_cost_fun_value = -4.54
[2022-06-13 11:15:00,081] INFO in command_line: Setting up needed data
[2022-06-13 11:15:00,085] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-06-13 11:15:00,953] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2022-06-13 11:15:04,654] INFO in forecast: Retrieving data from hass for load forecast using method = naive
[2022-06-13 11:15:04,656] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-06-13 11:15:06,970] INFO in web_server:  >> Performing naive MPC optimization...
[2022-06-13 11:15:06,970] INFO in command_line: Performing naive MPC optimization
[2022-06-13 11:15:06,978] INFO in optimization: Perform an iteration of a naive MPC controller
[2022-06-13 11:15:07,270] INFO in optimization: Status: Optimal
[2022-06-13 11:15:07,270] INFO in optimization: Total value of the Cost function = -4.54
```

closed 2022-06-13 by purcell-lab