davidusb-geek/emhass-add-on #18: Failure when installing Add-Onn

vdiogo, 2022-06-13:
I get the following message when trying to install this add-on. Tried to refresh the repository (deleted and added), but same behaviour

`Can't install davidusb/image-armv7-emhass:0.2.19: 404 Client Error for http+docker://localhost/v1.40/images/create?tag=0.2.19&fromImage=davidusb%2Fimage-armv7-emhass&platform=linux%2Farm%2Fv7: Not Found ("manifest for davidusb/image-armv7-emhass:0.2.19 not found: manifest unknown: manifest unknown")`

davidusb-geek, 2022-06-13:
Hi, yes this is normal. The images are being built. The ARM images take very long time to compile. I unsuccessfully tried different thing to reduced compiling times. I'm still looking for solutions to this. 

As you can see here:
[image]
The last built timeout after 6h of compiling. I relaunched it 2h ago to see if it compiles correctly this time...

davidusb-geek, 2022-06-13:
Ok, it's done you should be able to install with no problems.
Found a solution for faster compiles on arm archs, now it compiled on just 1h:
[image]

closed 2022-06-13 by davidusb-geek