davidusb-geek/emhass-add-on #19: ARMv7 v0.2.19 library issue

mirecekd, 2022-06-13:
Hi David,

it seems there is something wrong with ARMv7 image.

``Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 12, in <module>
    import pandas as pd
  File "/usr/local/lib/python3.9/dist-packages/pandas/__init__.py", line 16, in <module>
    raise ImportError(
ImportError: Unable to import required dependencies:
numpy:
IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!
Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.
We have compiled some common reasons and troubleshooting tips at:
    https://numpy.org/devdocs/user/troubleshooting-importerror.html
Please note and check the following:
  * The Python version is: Python3.9 from "/usr/bin/python3"
  * The NumPy version is: "1.22.4"
and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.
Original error was: libcblas.so.3: cannot open shared object file: No such file or directory

davidusb-geek, 2022-06-13:
Hi, thank for reporting this...
I think I found a solution here: https://numpy.org/devdocs/user/troubleshooting-importerror.html
I just applied this and the images are compiling. They will be available in 1h.

davidusb-geek, 2022-06-13:
Hi, please check when you can if the problem persist. The images are now published. You will need to uninstall an reinstall the add-on, save your configuration before doing this.

mirecekd, 2022-06-14:
Hi David, 
works great now :-)

closed 2022-06-14 by mirecekd