davidusb-geek/emhass-add-on #20: Solax X3 4K Inverter

vdiogo, 2022-06-15:
Hi team
Looking forward to use this add-on, but can't manage to find my inverter (Solax X3 4K) in the PVLIB DB.
Any guidance you can provide? Maybe an alternative (though equivalent) Inverter I can use?
Many thanks

davidusb-geek, 2022-06-15:
Hi, I understand that it is not ideal. The PVlib and CEC/Sandia db are updated regularly, so let's hope that in the future they include your model. In the meantime you can just find a model with similar characteristics. Mainly nominal power in kW and startup DC voltage. In our case this should be enough as we're only interested in the inverter efficiency curve and this will not drastically change from one model to another in the same nominal power range.

davidusb-geek, 2022-06-16:
If you install PVLib and do:

```
from pvlib import pvsystem
invdb = pvsystem.retrieve_sam('CECInverter')
cols = invdb.columns[(invdb.loc['Paco',:]==4000)].tolist()
invdb.loc[['Paco','Vdco','Vdcmax'],cols].transpose()
```

You obtain this list of possible candidates to use:

[image]

This is supposing that your model is 4 kW AC. 
The voltage are not matching very well.
What you can do is to set one of these. Then in Home Assistant store the forecasts obtained with EMHASS and compare that to your real PV production. This way you can check the validity of the chosen model. It is a good idea to do this even if your exact model was in the database.