davidusb-geek/emhass-add-on #23: Error for pv_module and pv_inverter

geekysh, 2022-07-27:
I'm trying to configure emhass on my home assistant installation running on a Raspberry Pi 4 rpi4-64.
I'll get errors when using my selected pv_module and pv_inverter. Maybe I don't apply the right number of _? or is there a bug?

I figured out which pv-module is most similar to my panels (as my panels from Bauer are not listed in the csv file), so I choose:
Vietnam Sunergy Joint Stock Company VSUN380-120BMH

Same for the inverter, but luckily the vendor exist, only diff is that my inverter is a 4.6KTL:
Huawei Technologies Co Ltd : SUN2000-5KTL-USL0 [240V]

When applying config, starting emhass and launching day-ahead optimization I get following log entries:
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
[2022-07-27 23:23:23,463] INFO in web_server: Launching the emhass webserver at: http://0.0.0.0:5000
[2022-07-27 23:23:23,464] INFO in web_server: Home Assistant data fetch will be performed using url: http://supervisor/core/api
[2022-07-27 23:23:23,466] INFO in web_server: The base path is: /usr/src
[2022-07-27 23:23:23,474] INFO in web_server: Using core emhass version: 0.3.17
[2022-07-27 23:23:32,087] INFO in web_server: EMHASS server online, serving index.html...
[2022-07-27 23:24:21,390] INFO in command_line: Setting up needed data
[2022-07-27 23:24:21,479] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2022-07-27 23:24:24,611] ERROR in app: Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/root/.local/lib/python3.9/site-packages/pandas/core/indexes/base.py", line 3621, in get_loc
    return self._engine.get_loc(casted_key)
  File "pandas/_libs/index.pyx", line 136, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/index.pyx", line 163, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 5198, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 5206, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'Vietnam_Sunergy_Joint_Stock_Company_VSUN380_120BMH'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 134, in action_call
    input_data_dict = set_input_data_dict(config_path, str(config_path.parent), costfun,
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 76, in set_input_data_dict
    P_PV_forecast = fcst.get_power_from_weather(df_weather)
  File "/usr/local/lib/python3.9/dist-packages/emhass/forecast.py", line 317, in get_power_from_weather
    module = cec_modules[self.plant_conf['module_model'][i]]
  File "/root/.local/lib/python3.9/site-packages/pandas/core/frame.py", line 3505, in __getitem__
    indexer = self.columns.get_loc(key)
  File "/root/.local/lib/python3.9/site-packages/pandas/core/indexes/base.py", line 3623, in get_loc
    raise KeyError(key) from err
KeyError: 'Vietnam_Sunergy_Joint_Stock_Company_VSUN380_120BMH'`
 
When replacing the pv-module name in the config with the documentation example:
CSUN_Eurasia_Energy_Systems_Industry_and_Trade_CSUN295_60M
I'll get this:
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
[2022-07-27 23:34:08,710] INFO in web_server: Launching the emhass webserver at: http://0.0.0.0:5000
[2022-07-27 23:34:08,710] INFO in web_server: Home Assistant data fetch will be performed using url: http://supervisor/core/api
[2022-07-27 23:34:08,711] INFO in web_server: The base path is: /usr/src
[2022-07-27 23:34:08,718] INFO in web_server: Using core emhass version: 0.3.17
[2022-07-27 23:34:21,858] INFO in web_server: EMHASS server online, serving index.html...
[2022-07-27 23:34:33,264] INFO in command_line: Setting up needed data
[2022-07-27 23:34:33,353] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2022-07-27 23:34:35,533] ERROR in app: Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/root/.local/lib/python3.9/site-packages/pandas/core/indexes/base.py", line 3621, in get_loc
    return self._engine.get_loc(casted_key)
  File "pandas/_libs/index.pyx", line 136, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/index.pyx", line 163, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 5198, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 5206, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'Huawei_Technologies_Co_Ltd___SUN2000_5KTL_USL0__240V_'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 134, in action_call
    input_data_dict = set_input_data_dict(config_path, str(config_path.parent), costfun,
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 76, in set_input_data_dict
    P_PV_forecast = fcst.get_power_from_weather(df_weather)
  File "/usr/local/lib/python3.9/dist-packages/emhass/forecast.py", line 318, in get_power_from_weather
    inverter = cec_inverters[self.plant_conf['inverter_model'][i]]
  File "/root/.local/lib/python3.9/site-packages/pandas/core/frame.py", line 3505, in __getitem__
    indexer = self.columns.get_loc(key)
  File "/root/.local/lib/python3.9/site-packages/pandas/core/indexes/base.py", line 3623, in get_loc
    raise KeyError(key) from err
KeyError: 'Huawei_Technologies_Co_Ltd___SUN2000_5KTL_USL0__240V_'`

davidusb-geek, 2022-07-29:
Hi,

For your inverter I've found that you should put it this way:
'Huawei_Technologies_Co___Ltd___SUN2000_5KTL_USL0__240V_'

However I wasn't able to find that module name in the database.

If you have Python installed somewhere, you can use PVLib to find the available names in the database using this code snippet:

```
from pvlib import pvsystem
moddb = pvsystem.retrieve_sam('CECMod')
print(moddb.columns)
```

geekysh, 2022-08-01:
Hi I've no idea how to install python on home assistant and/or remotely connect to retrieve the database contents. Isn't it based on the SAM CEC database?
When I look up the panels in the CEC database, it exists:
[image]

Maybe you can update the addon with all latest contents of the online CEC database?
The Inverter name is solved as you mentioned. Thanks for that!

geekysh, 2022-08-01:
Yes correct, in that file I also looked up the model that most matches my panels config. 
If you look in the csv you'll find: Vietnam Sunergy Joint Stock Company VSUN380-120BMH
So that's weird? Or is that file not 100% the same as what can be found in the emhass database?

davidusb-geek, 2022-08-01:
I just can't find any "Vietnam Sunergy".

We can just look at that CSV file columnbs directly on github, there is nothing on letter V...

[image]

davidusb-geek, 2022-08-01:
Ok from what I can get the model you're looking for is on the latest version of the database. For some weird reason it is not shown on github but it showed fine when I downloaded and opened the CSV file.
The thing is that as I said before the PVLib module needs to be updated to read that latest database. 
This is not the case on the current PVLib code as seen here: https://github.com/pvlib/pvlib-python/blob/f8921bd3879c4bcd9c0242005319729ab04ce7f6/pvlib/pvsystem.py#L1529-L1547
As I said there is nothing I can do as this is a PVLib issue.

davidusb-geek, 2022-08-01:
So you'll find the available modules on this file:

[sam-library-cec-modules-2019-03-05.csv](https://github.com/davidusb-geek/emhass-add-on/files/9234460/sam-library-cec-modules-2019-03-05.csv)

davidusb-geek, 2022-08-03:
Hi, just closing this as the solution is to pick a module from the csv file provided. This correspond to the file used in the latest PVlib version. I'll keep a notification to always update to the latest version of PVLib. Hopefully including more modules and inverter models in the future.

closed 2022-08-03 by davidusb-geek

purcell-lab, 2022-08-03:
Pv_module selection is very error prone, from my own experience.

Would a drop down of available options work?

Yours Aye,
Mark Purcell

***@***.***
04 2822 7764

davidusb-geek, 2022-09-09:
A drop menu would be great, just don't have the time to develop that.
This is the file for the inverters:
[sam-library-cec-inverters-2019-03-05.csv](https://github.com/davidusb-geek/emhass-add-on/files/9532724/sam-library-cec-inverters-2019-03-05.csv)

copadewe, 2023-10-22:
Within this list you mentioned, I could not found my modules and even not my inverter :-(
Any idea?
Solon Blue 230/7 235Wp
Sungrow Hybrid Inverter SH10rt
Sungrow SBR battery
cheers Oliver

rwunsch, 2023-11-16:
hey @copadewe ,

did you find a solution?
I have a very similar (and likely popular) set-up:
Sungrow Hybrid Inverter SH10rt
Sungrow SBR battery 128
BSM440PMB7-46SC

How would we be able to update the list, or add a configuration for Inverter(s) and Module(s)?