davidusb-geek/emhass-add-on #24: Passing forecast data do not work

overas, 2022-08-28:
I have posted most of the description of the problem here: 
https://community.home-assistant.io/t/emhass-an-energy-management-for-home-assistant/338126/111?u=haraldov

What is the Home Assistant log saying when you execute that shell command?

2022-08-28 20:38:03.701 DEBUG (MainThread) [homeassistant.components.shell_command] Stdout of command: `curl -i -H "Content-Type: application/json" -X POST -d '{"load_cost_forecast": {{ ((( state_attr('sensor.nordpool', 'raw_today') + state_attr('sensor.nordpool', 'raw_tomorrow')) |map(attribute='value')|list)[:48]) }}' http://localhost:5000/action/dayahead-optim`, return code: 0:
2022-08-28 20:38:03.701 DEBUG (MainThread) [homeassistant.components.shell_command] Stderr of command: `curl -i -H "Content-Type: application/json" -X POST -d '{"load_cost_forecast": {{ ((( state_attr('sensor.nordpool', 'raw_today') + state_attr('sensor.nordpool', 'raw_tomorrow')) |map(attribute='value')|list)[:48]) }}' http://localhost:5000/action/dayahead-optim`, return code: 0:
➜  config

However it doesn't get updated in the model.

Home Assistant 2022.8.7
Supervisor 2022.08.3
Operating System 8.5
Frontend 20220802.0 - latest
Raspberry Pi 4

EMHASS
Current version: 0.2.20

[2022-08-28 00:14:04,984] INFO in web_server: Launching the emhass webserver at: http://0.0.0.0:5000
[2022-08-28 00:14:04,985] INFO in web_server: Home Assistant data fetch will be performed using url: http://supervisor/core/api
[2022-08-28 00:14:04,985] INFO in web_server: The base path is: /usr/src
[2022-08-28 00:14:04,989] INFO in web_server: Using core emhass version: 0.3.18
[2022-08-28 00:15:24,545] INFO in web_server: EMHASS server online, serving index.html...
[2022-08-28 00:20:29,186] INFO in web_server: EMHASS server online, serving index.html...
[2022-08-28 00:22:31,750] INFO in web_server: EMHASS server online, serving index.html...
[2022-08-28 00:22:33,664] INFO in command_line: Setting up needed data
[2022-08-28 00:22:33,759] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2022-08-28 00:22:39,605] INFO in forecast: Retrieving data from hass for load forecast using method = naive
[2022-08-28 00:22:39,607] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2022-08-28 00:22:53,887] INFO in web_server: EMHASS server online, serving index.html...
[2022-08-28 00:22:56,188] INFO in web_server:  >> Performing dayahead optimization...
[2022-08-28 00:22:56,189] INFO in command_line: Performing day-ahead forecast optimization
[2022-08-28 00:22:56,198] INFO in optimization: Perform optimization for the day-ahead
[2022-08-28 00:22:56,755] INFO in optimization: Status: Optimal
[2022-08-28 00:22:56,756] INFO in optimization: Total value of the Cost function = -2.02
[2022-08-28 20:37:38,794] INFO in web_server: EMHASS server online, serving index.html...
[2022-08-28 20:38:27,230] INFO in web_server: EMHASS server online, serving index.html...

## System Information

version | core-2022.8.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.5
os_name | Linux
os_version | 5.15.32-v8
arch | aarch64
timezone | Europe/Oslo
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.26.2
Stage | running
Available Repositories | 1164
Downloaded Repositories | 12

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | September 7, 2022 at 02:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-west-2-3.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Easee EV Charger</summary>

component_version | 0.9.44
-- | --
reach_easee_cloud | ok
connected2stream | true

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.5
-- | --
update_channel | stable
supervisor_version | supervisor-2022.08.3
agent_version | 1.2.1
docker_version | 20.10.14
disk_total | 118.3 GB
disk_used | 14.8 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | ESPHome (2022.3.1), SSH & Web Terminal (12.0.2), Mosquitto broker (6.1.2), File editor (5.3.3), Frigate NVR Proxy (1.3), Home Assistant Google Drive Backup (0.108.4), Z-Wave JS to MQTT (0.45.0), EMHASS (0.2.20)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 2
views | 10
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | August 25, 2022 at 08:33
-- | --
current_recorder_run | August 28, 2022 at 09:14
estimated_db_size | 575.15 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

davidusb-geek, 2022-08-28:
Could you please test if the following expression on your shell command throws a list and not a string?

```
{{ ((( state_attr('sensor.nordpool', 'raw_today') + state_attr('sensor.nordpool', 'raw_tomorrow')) |map(attribute='value')|list)[:48]) }}
```

overas, 2022-08-28:
The template editor says it is a list. Is this the correct way to test the expression?

[image]

```
Result type: list
[
  7.009,
  6.175,
  5.877,
  5.372,
  5.342,
  5.141,
  5.358,
  5.542,
  5.862,
  5.329,
  4.124,
  6.088,
  6.125,
  6.125,
  6.088,
  5.724,
  4.651,
  5.998,
  6.202,
  7.305,
  8.261,
  7.63,
  7.173,
  6.976,
  6.986,
  6.434,
  6.269,
  6.209,
  6.266,
  7.179,
  7.914,
  8.637,
  8.875,
  8.307,
  7.787,
  7.305,
  6.896,
  6.94,
  6.82,
  7.351,
  7.914,
  8.392,
  9.79,
  10.277,
  10.158,
  9.439,
  8.523,
  7.794
]
This template listens for the following state changed events:

Entity: sensor.nordpool
```

davidusb-geek, 2022-08-29:
Ok, no that list looks fine, this should be working correctly. A quick check: are you refreshing the GUI web server after launching the optimization? You will need to refresh (F5) to update the results table.