davidusb-geek/emhass-add-on #26: Error message during startup of add-on

RalfPfitzner, 2022-09-12:
Hi David,

I get a repeating error message during the startup of the add-on.

After reading (and re-reading) the documentation I think I did a correct setup of the add-on parameters in the config page.
Unfortunately I get this error message and have no idea what to do now?

Can you help with this?

Thanks in advance,
Ralf

`[2022-09-12 14:57:21,488] INFO in web_server: Launching the emhass webserver at: http://my_local_ip:5000
[2022-09-12 14:57:21,489] INFO in web_server: Home Assistant data fetch will be performed using url: https://my_public_ha
[2022-09-12 14:57:21,489] INFO in web_server: The base path is: /usr/src
[2022-09-12 14:57:21,491] INFO in web_server: Using core emhass version: 0.3.18
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 268, in <module>
    serve(app, host=web_ui_url, port=port, threads=8)
  File "/usr/local/lib/python3.9/dist-packages/waitress/__init__.py", line 13, in serve
    server = _server(app, **kw)
  File "/usr/local/lib/python3.9/dist-packages/waitress/server.py", line 78, in create_server
    last_serv = TcpWSGIServer(
  File "/usr/local/lib/python3.9/dist-packages/waitress/server.py", line 244, in __init__
    self.bind_server_socket()
  File "/usr/local/lib/python3.9/dist-packages/waitress/server.py", line 361, in bind_server_socket
    self.bind(sockaddr)
  File "/usr/local/lib/python3.9/dist-packages/waitress/wasyncore.py", line 396, in bind
    return self.socket.bind(addr)
OSError: [Errno 99] Cannot assign requested address`

davidusb-geek, 2022-09-12:
Hi, yes this is just failing miserably with the actual WSGIServer implementation. I am sorry.
So actually for the UI to work you just have to use the default parameters, ie:
```
  web_ui_url: 0.0.0.0
  hass_url: empty
```
So yes the UI with only be available at the localhost and you will only be able to access it if you are in the same network (or by a VPN if you are outside your network).
I will try to update this in the add-on config and in the documentation to avoid this kind of errors in the future.

davidusb-geek, 2022-09-12:
Ok. Please share your setup. Are you using HA OS or supervized?

davidusb-geek, 2022-09-12:
You're sure you're using the default parameters? Like this?
```
  web_ui_url: 0.0.0.0
  hass_url: empty
  long_lived_token: empty
```
Did you save your config with the button at the bottom of the add-on?

davidusb-geek, 2022-09-12:
The hass_url parameter should be the string "empty". I mean you have to write the word "empty"

RalfPfitzner, 2022-09-12:
That did the Trick!
Now I have to play with the solar modules and the inverter and gather some more data on my new sensor (total consumption minus the deferrable ones).
By the way, I would like to use forecast.solar values as I have these already. How would I do that?

davidusb-geek, 2022-09-12:
Great!
For solar.forecast I opened a thread asking this same exact question. The way to go will be to use a rest sensor. Take a look here: https://community.home-assistant.io/t/emhass-an-energy-management-for-home-assistant/338126?page=7
Marking this as solved 👍

closed 2022-09-12 by davidusb-geek