davidusb-geek/emhass-add-on #29: FR: Access EMHASS web interface via ingress

purcell-lab, 2022-09-15:
Ingress provides a means for an Home Assistant addon to expose its web service via the HA UI:

https://www.home-assistant.io/blog/2019/04/15/hassio-ingress/
https://developers.home-assistant.io/docs/add-ons/presentation/#ingress

Currently the EMHASS Web User Interface is accessed via http://homeassistant-ip:5000.

It would greatly simplify the user experience if EMHASS was configured to utilise the ingress interface.

My other add-on's are access via the Home Assistant UI and ingress interface:

ESPHome: http://odroid.local:8123/hassio/ingress/5c53de3b_esphome
Google Drive Backup: http://odroid.local:8123/hassio/ingress/cebe7a76_hassio_google_drive_backup
Terminal & SSH: http://odroid.local:8123/hassio/ingress/core_ssh

[image]
[image]

davidusb-geek, 2022-09-15:
I will test this for the next version of the add-on. I've never been happy with the webui server implementation and I've dropped support for custom webui urls. So maybe ingress can ease the use of the webui and its configuration

davidusb-geek, 2022-09-25:
Hi, I advanced a bit on testing this using this branch on emhass: https://github.com/davidusb-geek/emhass/tree/dev/ingress
It shows the UI when pressing the webui button but without the css formatting and most importantly the optimization button is not doing anything, totally unresponsive.
I think I've reached my limits on my skills using ingress with a flask server and trying to configure this properly.
Any help for this will be most than welcomed