davidusb-geek/emhass-add-on #30: Using comma separated values in the configuration do not work

overas, 2022-10-15:
I am using emhass version 0.2.22 addon.

In the configuration I have this config:
[image]

Under the form the help text says:
_This is a list of comma separated values with a number of elements consistent with the number of deferrable loads defined before. For example 3000, 750_
and
_A list of comma separated values. For example 5, 8_

I try to change the config to use comma separated values:
[image]

I then get this error when trying to save:
[image]

davidusb-geek, 2022-10-15:
Oh yes this is misleading, thank you for reporting this. That information is actually accurate just when using the core emhass python module, not for the add-on. I'll fix this information. In the add-on you just have to provide a list as shown on the top of your post, and not comma separated values

davidusb-geek, 2022-11-21:
Done! 👍

closed 2022-11-21 by davidusb-geek