davidusb-geek/emhass-add-on #32: Not able to add the EMHASS add on

dutchronnie, 2023-01-15:
When i want to add the repository for emhass, i get an error message.

<Integration davidusb-geek/emhass> Repository structure for v0.3.21 is not compliant

I try to add the URL: https://github.com/davidusb-geek/emhass

davidusb-geek, 2023-01-15:
The correct URL is: https://github.com/davidusb-geek/emhass-add-on

dutchronnie, 2023-01-16:
Are you sure this is the correct url?

[image]

I am not able to add the repository to Homeassistant

davidusb-geek, 2023-01-16:
Hi, yes sure it is the correct URL. Very weird behavior. 
That URL has been working fine for one and a half year now and this is the first time that I see this problem. So my guess is that may be there is an issue with your system. Try taking a look at the logs to see the error message when you try to add the URL. Otherwise try to reboot your system.

davidusb-geek, 2023-01-16:
[image]

dutchronnie, 2023-01-16:
This is very strange, i just tried again, with the same url, and now the repository is added.

It was not 1 try , with the error message, yesterday tried several times, and also this morning.
I have not rebooted the system. 

But it works now, thanks

closed 2023-01-16 by davidusb-geek