davidusb-geek/emhass-add-on #33: Tried to install addon just now and get Internal Server Error

tinuva, 2023-01-31:
Installed version `0.2.28 ` on Home Assistant supervised.

Curl output
```
~ % curl -v http://home-assistant.heaven.za.net:5000
*   Trying 10.0.3.14:5000...
* Connected to home-assistant.heaven.za.net (10.0.3.14) port 5000 (#0)
> GET / HTTP/1.1
> Host: home-assistant.heaven.za.net:5000
> User-Agent: curl/7.86.0
> Accept: */*
>
* Mark bundle as not supporting multiuse
< HTTP/1.1 500 INTERNAL SERVER ERROR
< Content-Length: 265
< Content-Type: text/html; charset=utf-8
< Date: Tue, 31 Jan 2023 05:12:03 GMT
< Server: waitress
<
<!doctype html>
<html lang=en>
<title>500 Internal Server Error</title>
<h1>Internal Server Error</h1>
<p>The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.</p>
* Connection #0 to host home-assistant.heaven.za.net left intact
```

Docker logs:
```
root@home-assistant:/usr/share/hassio/homeassistant# docker logs addon_5b918bf2_emhass
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
[2023-01-31 07:09:05,580] INFO in web_server: Launching the emhass webserver at: http://0.0.0.0:5000
[2023-01-31 07:09:05,580] INFO in web_server: Home Assistant data fetch will be performed using url: http://supervisor/core/api
[2023-01-31 07:09:05,580] INFO in web_server: The data path is: /share
[2023-01-31 07:09:05,581] INFO in web_server: Using core emhass version: 0.3.34
[2023-01-31 07:09:25,026] INFO in web_server: EMHASS server online, serving index.html...
[2023-01-31 07:09:25,030] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2525, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1822, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1820, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1796, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 121, in index
    with open(str(data_path / 'injection_dict.pkl'), "rb") as fid:
FileNotFoundError: [Errno 2] No such file or directory: '/share/injection_dict.pkl'
[2023-01-31 07:09:35,987] INFO in web_server: EMHASS server online, serving index.html...
[2023-01-31 07:09:35,990] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2525, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1822, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1820, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1796, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 121, in index
    with open(str(data_path / 'injection_dict.pkl'), "rb") as fid:
FileNotFoundError: [Errno 2] No such file or directory: '/share/injection_dict.pkl'
[2023-01-31 07:10:06,070] INFO in web_server: EMHASS server online, serving index.html...
[2023-01-31 07:10:06,072] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2525, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1822, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1820, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1796, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 121, in index
    with open(str(data_path / 'injection_dict.pkl'), "rb") as fid:
FileNotFoundError: [Errno 2] No such file or directory: '/share/injection_dict.pkl'
[2023-01-31 07:12:01,351] INFO in web_server: EMHASS server online, serving index.html...
[2023-01-31 07:12:01,353] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2525, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1822, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1820, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1796, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 121, in index
    with open(str(data_path / 'injection_dict.pkl'), "rb") as fid:
FileNotFoundError: [Errno 2] No such file or directory: '/share/injection_dict.pkl'
[2023-01-31 07:12:02,679] INFO in web_server: EMHASS server online, serving index.html...
[2023-01-31 07:12:02,681] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2525, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1822, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1820, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1796, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 121, in index
    with open(str(data_path / 'injection_dict.pkl'), "rb") as fid:
FileNotFoundError: [Errno 2] No such file or directory: '/share/injection_dict.pkl'
[2023-01-31 07:12:03,682] INFO in web_server: EMHASS server online, serving index.html...
[2023-01-31 07:12:03,685] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2525, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1822, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1820, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1796, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 121, in index
    with open(str(data_path / 'injection_dict.pkl'), "rb") as fid:
FileNotFoundError: [Errno 2] No such file or directory: '/share/injection_dict.pkl'
```

davidusb-geek, 2023-01-31:
There has been a recurrent problem when using custom urls. So try with the default url at localhost or at http;//0.0.0.0:5000
Change this on the add-on configuration and on your curl command

tinuva, 2023-01-31:
Ok. Here is the curl

```
root@home-assistant:/usr/share/hassio/homeassistant# curl -v http://localhost:5000/
*   Trying ::1:5000...
* Connected to localhost (::1) port 5000 (#0)
> GET / HTTP/1.1
> Host: localhost:5000
> User-Agent: curl/7.74.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 500 INTERNAL SERVER ERROR
< Content-Length: 265
< Content-Type: text/html; charset=utf-8
< Date: Tue, 31 Jan 2023 06:24:44 GMT
< Server: waitress
< 
<!doctype html>
<html lang=en>
<title>500 Internal Server Error</title>
<h1>Internal Server Error</h1>
<p>The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.</p>
* Connection #0 to host localhost left intact
* 
```
log
```
[2023-01-31 08:24:44,058] INFO in web_server: EMHASS server online, serving index.html...
[2023-01-31 08:24:44,061] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2525, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1822, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1820, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1796, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 121, in index
    with open(str(data_path / 'injection_dict.pkl'), "rb") as fid:
FileNotFoundError: [Errno 2] No such file or directory: '/share/injection_dict.pkl'
```

and also. So this one looks like my config is the issue.
```
root@home-assistant:/usr/share/hassio/homeassistant# curl -v -i -H 'Content-Type:application/json' -X POST -d '{}' http://localhost:5000/action/dayahead-optim
Note: Unnecessary use of -X or --request, POST is already inferred.
*   Trying ::1:5000...
* Connected to localhost (::1) port 5000 (#0)
> POST /action/dayahead-optim HTTP/1.1
> Host: localhost:5000
> User-Agent: curl/7.74.0
> Accept: */*
> Content-Type:application/json
> Content-Length: 2
> 
* upload completely sent off: 2 out of 2 bytes
* Mark bundle as not supporting multiuse
< HTTP/1.1 500 INTERNAL SERVER ERROR
HTTP/1.1 500 INTERNAL SERVER ERROR
< Content-Length: 265
Content-Length: 265
< Content-Type: text/html; charset=utf-8
Content-Type: text/html; charset=utf-8
< Date: Tue, 31 Jan 2023 06:19:41 GMT
Date: Tue, 31 Jan 2023 06:19:41 GMT
< Server: waitress
Server: waitress

< 
<!doctype html>
<html lang=en>
<title>500 Internal Server Error</title>
<h1>Internal Server Error</h1>
<p>The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.</p>
* Connection #0 to host localhost left intact
```
log
```
2023-01-31 08:19:42,263] ERROR in app: Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2525, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1822, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1820, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1796, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 135, in action_call
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 83, in set_input_data_dict
    P_load_forecast = fcst.get_load_forecast(method=optim_conf['load_forecast_method'])
  File "/usr/local/lib/python3.9/dist-packages/emhass/forecast.py", line 539, in get_load_forecast
    rh.prepare_data(self.retrieve_hass_conf['var_load'], load_negative = self.retrieve_hass_conf['load_negative'],
  File "/usr/local/lib/python3.9/dist-packages/emhass/retrieve_hass.py", line 184, in prepare_data
    self.df_final.drop([var_load], inplace=True, axis=1)
  File "/usr/local/lib/python3.9/dist-packages/pandas/util/_decorators.py", line 311, in wrapper
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/frame.py", line 4948, in drop
    return super().drop(
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/generic.py", line 4279, in drop
    obj = obj._drop_axis(labels, axis, level=level, errors=errors)
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/generic.py", line 4323, in _drop_axis
    new_axis = axis.drop(labels, errors=errors)
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/indexes/base.py", line 6644, in drop
    raise KeyError(f"{list(labels[mask])} not found in axis")
KeyError: "['sensor.power_load_no_var_loads'] not found in axis"
```

Going by that, the generic web page is not working.
The post, I need to look at the config.

davidusb-geek, 2023-01-31:
Ok your first `curl` command is not valid, so the errors are normal.

The second `curl` command is correct. 

This error `FileNotFoundError: [Errno 2] No such file or directory: '/share/injection_dict.pkl'` is solved in the version of the add-on that I pushed today v0.2.29. So update to that and it should be good.

For your final error I think that you just have to wait for 48 h of recorded data for sensor `sensor.power_load_no_var_loads`. It is weird however that this error was not caught with a clearer error message. But for me it basically means that you don't have enough data.

tinuva, 2023-01-31:
Thank you, new version have resolved the initial error I had.

I will give it a few days to get some data on my new template sensor.

closed 2023-01-31 by tinuva