davidusb-geek/emhass-add-on #35: Lost in how the add-on works

vdiogo, 2023-02-11:
I have setup the configuration parameters in the add-on such as:
- sensor returning PV power production
- sensor with load power
- inverter brand/ model
- solar panels brand/ model
- setup the day-ahead + publish data automations
- update config.yaml with the shell command

And when accessing the user interface, none of the options returns a value

I get this in the logs, which I'm not able to understand:
```
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1822, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1820, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1796, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 135, in action_call
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 70, in set_input_data_dict
    rh.get_data(days_list, var_list,
  File "/usr/local/lib/python3.9/dist-packages/emhass/retrieve_hass.py", line 144, in get_data
    self.df_final = pd.concat([self.df_final, df_day], axis=0)
UnboundLocalError: local variable 'df_day' referenced before assignment
[2023-02-11 14:09:05,328] INFO in command_line: Setting up needed data
[2023-02-11 14:09:05,341] INFO in forecast: Retrieving weather forecast data using method = scrapper
[2023-02-11 14:09:06,368] INFO in command_line: Setting up needed data
[2023-02-11 14:09:06,381] INFO in web_server:  >> Publishing data...
[2023-02-11 14:09:06,381] INFO in command_line: Publishing data to HASS instance
[2023-02-11 14:09:06,383] ERROR in command_line: File not found error, run an optimization task first.
[2023-02-11 14:09:06,384] ERROR in app: Exception on /action/publish-data [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2525, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1822, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1820, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1796, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 139, in action_call
    _ = publish_data(input_data_dict, app.logger)
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 290, in publish_data
    idx_closest = opt_res_latest.index.get_indexer([now_precise], method='nearest')[0]
UnboundLocalError: local variable 'opt_res_latest' referenced before assignment
[2023-02-11 14:09:12,936] INFO in forecast: Retrieving data from hass for load forecast using method = naive
[2023-02-11 14:09:12,940] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2023-02-11 14:09:13,012] ERROR in retrieve_hass: The retrieved JSON is empty, check that correct day or variable names are passed
[2023-02-11 14:09:13,013] ERROR in retrieve_hass: Either the names of the passed variables are not correct or days_to_retrieve is larger than the recorded history of your sensor (check your recorder settings)
[2023-02-11 14:09:13,014] ERROR in app: Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2525, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1822, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1820, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1796, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 135, in action_call
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 83, in set_input_data_dict
    P_load_forecast = fcst.get_load_forecast(method=optim_conf['load_forecast_method'])
  File "/usr/local/lib/python3.9/dist-packages/emhass/forecast.py", line 538, in get_load_forecast
    rh.get_data(days_list, var_list)
  File "/usr/local/lib/python3.9/dist-packages/emhass/retrieve_hass.py", line 144, in get_data
    self.df_final = pd.concat([self.df_final, df_day], axis=0)
UnboundLocalError: local variable 'df_day' referenced before assignment
[2023-02-11 14:10:00,550] INFO in command_line: Setting up needed data
[2023-02-11 14:10:00,563] INFO in web_server:  >> Publishing data...
[2023-02-11 14:10:00,563] INFO in command_line: Publishing data to HASS instance
[2023-02-11 14:10:00,564] ERROR in command_line: File not found error, run an optimization task first.
[2023-02-11 14:10:00,566] ERROR in app: Exception on /action/publish-data [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2525, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1822, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1820, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1796, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 139, in action_call
    _ = publish_data(input_data_dict, app.logger)
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 290, in publish_data
    idx_closest = opt_res_latest.index.get_indexer([now_precise], method='nearest')[0]
UnboundLocalError: local variable 'opt_res_latest' referenced before assignment
[2023-02-11 14:13:53,649] INFO in web_server: EMHASS server online, serving index.html...
[2023-02-11 14:13:53,672] WARNING in web_server: The data container dictionary is empty... Please launch an optimization task
[2023-02-11 14:13:54,959] INFO in command_line: Setting up needed data
[2023-02-11 14:13:54,973] INFO in retrieve_hass: Retrieve hass get data method initiated...
[2023-02-11 14:13:55,045] ERROR in retrieve_hass: The retrieved JSON is empty, check that correct day or variable names are passed
[2023-02-11 14:13:55,046] ERROR in retrieve_hass: Either the names of the passed variables are not correct or days_to_retrieve is larger than the recorded history of your sensor (check your recorder settings)
[2023-02-11 14:13:55,047] ERROR in app: Exception on /action/perfect-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2525, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1822, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1820, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1796, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 135, in action_call
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 70, in set_input_data_dict
    rh.get_data(days_list, var_list,
  File "/usr/local/lib/python3.9/dist-packages/emhass/retrieve_hass.py", line 144, in get_data
    self.df_final = pd.concat([self.df_final, df_day], axis=0)
UnboundLocalError: local variable 'df_day' referenced before assignment
```

davidusb-geek, 2023-02-11:
Half of your error messages are from trying to publish data before a correct optimization call.

When you call the optimization the messages to understand are these:

ERROR in retrieve_hass: The retrieved JSON is empty, check that correct day or variable names are passed
[2023-02-11 14:13:55,046] ERROR in retrieve_hass: Either the names of the passed variables are not correct or days_to_retrieve is larger than the recorded history of your sensor (check your recorder settings)

So check that. The names of your sensors and that you have enough data.
Otherwise there is a problem with your configuration. Share it to debug it and state some info about your system: type of HA install, your machine arch, etc

vdiogo, 2023-02-12:
Thank you for the prompt reply. 
I installed EMHASS 3 days ago on a fresh RPI3B+ HA Home Assistant Supervised where I only enabled my PV production sensor and Energy consumption sensor (without discounting the deferrable load as I'm still looking into which appliance to "trigger" the optimization result (hope this makes sense).
In the meanwhile I see that my PV sensor stopped sending results (claiming too many API queries, after which I adjusted the poll from 60 to 120 seconds).
So it may be that I just don't have enough data to publish the optimization results so will give it a couple more days and check again :)

davidusb-geek, 2023-02-12:
Ok it may be that. Also as you are using the supervisor then leave the `hass_url` and `long_lived_token` parameters to `empty`.

davidusb-geek, 2023-02-28:
Feel free to reopen if this wasn't solved with the proposed solution

closed 2023-02-28 by davidusb-geek