davidusb-geek/emhass-add-on #39: 0.3.3 not starting

kcoffau, 2023-03-10:
EMHASS 0.3.3
Home Assistant 2023.3.3
Supervisor 2023.03.1
Operating System 9.5
Frontend 20230309.0 - latest

Error:
EMHASS not starting:
Log:
/usr/bin/python3: Error while finding module specification for 'emhass.web_server' (ModuleNotFoundError: No module named 'emhass')

Went from version 0.2.9 to 0.3.3, hard restarted Home Assistant device, and EMHASS not starting.

Unsure on next troubleshooting steps?

kcoffau, 2023-03-10:
Config as FYI
hass_url: empty
long_lived_token: empty
costfun: profit
optimization_time_step: 30
historic_days_to_retrieve: 3
method_ts_round: first
set_total_pv_sell: false
lp_solver: COIN_CMD
lp_solver_path: /usr/bin/cbc
set_nocharge_from_grid: false
load_forecast_method: naive
sensor_power_photovoltaics: sensor.power_ac_fronius_inverter_1_192_168_10_122
sensor_power_load_no_var_loads: sensor.victron_ac_l1
number_of_deferrable_loads: 1
list_nominal_power_of_deferrable_loads:
  - nominal_power_of_deferrable_loads: 2400
list_operating_hours_of_each_deferrable_load:
  - operating_hours_of_each_deferrable_load: 4
list_peak_hours_periods_start_hours:
  - peak_hours_periods_start_hours: "17:00"
list_peak_hours_periods_end_hours:
  - peak_hours_periods_end_hours: "21:00"
list_treat_deferrable_load_as_semi_cont:
  - treat_deferrable_load_as_semi_cont: true
load_peak_hours_cost: 0.6
load_offpeak_hours_cost: 0.6
photovoltaic_production_sell_price: 0.4
maximum_power_from_grid: 15000
list_pv_module_model:
  - pv_module_model: CSUN_Eurasia_Energy_Systems_Industry_and_Trade_CSUN295_60M
list_pv_inverter_model:
  - pv_inverter_model: Fronius_International_GmbH__Fronius_Primo_5_0_1_208_240__240V_
list_surface_tilt:
  - surface_tilt: 15
list_surface_azimuth:
  - surface_azimuth: 10
list_modules_per_string:
  - modules_per_string: 11
list_strings_per_inverter:
  - strings_per_inverter: 2
set_use_battery: true
battery_discharge_power_max: 8500
battery_charge_power_max: 8000
battery_discharge_efficiency: 0.9
battery_charge_efficiency: 0.8
battery_nominal_energy_capacity: 80000
battery_minimum_state_of_charge: 0.1
battery_maximum_state_of_charge: 0.99999
battery_target_state_of_charge: 0.66
web_ui_url: 192.168.10.81

purcell-lab, 2023-03-10:
Same issue here.  0.3.3 doesn't start.

0.3.2 was last good version.

davidusb-geek, 2023-03-10:
Fixed! sorry about this, a silly mistake.
v0.3.4 should work fine

purcell-lab, 2023-03-10:
This issue was resolved for me with 0.3.4, thanks for the quick update.

kcoffau, 2023-03-10:
Hi, thanks for jumping on this so quickly.  I updated from 3.3 to 3.5 and now:
[image]

davidusb-geek, 2023-03-10:
That's a hass and supervisor issue, not emhass. Just reboot your system and retry. Otherwise uninstall the add-on, reboot your system, reinstall the add-on. It should do the trick.

kcoffau, 2023-03-10:
Don't worry, did a uninstall and reinstall and that rectified that issue.

closed 2023-03-10 by kcoffau