davidusb-geek/emhass-add-on #40: Publish p_deferrables half an hour to early

prinzkafka, 2023-03-24:
Hi, I have a strange problem. As you can see in the following graphics, the deferrables are always published half an hour too early to their value. 
Time Step is set to 60 min, it is published every 3 minutes.

Strangely enough the values in the EMHASS UI and also in the attributes of for example sensor.p_deferrable0 are correct:

[image]

and still the deferrabeles always start exactly half an hour before the correct time

[image]

`Deferrables schedule
- date: '2023-03-24 11:00:00+01:00' p_deferrable0: '2000.0'
- date: '2023-03-24 12:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-24 13:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-24 14:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-24 15:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-24 16:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-24 17:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-24 18:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-24 19:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-24 20:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-24 21:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-24 22:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-24 23:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-25 00:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-25 01:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-25 02:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-25 03:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-25 04:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-25 05:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-25 06:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-25 07:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-25 08:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-25 09:00:00+01:00' p_deferrable0: '0.0' 
- date: '2023-03-25 10:00:00+01:00' p_deferrable0: '0.0'`

the first entry in this list (11:00) changes the state of deferrable0 at 10:30 
my automations only reacts to p_deferrables...

so either I have a complete knot in my head or something doesn't fit there

davidusb-geek, 2023-03-24:
Change the `method_ts_round` to what it best fits your needs. The options are: "first", "last" and "nearest".

prinzkafka, 2023-03-27:
all right, set it to "last" for now, seems to work reasonably so far - before that I had it set to "nearest" - thanks a lot for your outstanding work on this addon

closed 2023-03-27 by prinzkafka