davidusb-geek/emhass-add-on #41: Pandas logs NonExistentTimeError: 2023-03-26 02:00:00

overas, 2023-03-25:
Home Assistant 2023.3.6
Supervisor 2023.03.2
Operating System 9.5
Frontend 20230309.1 - latest
Emhass  0.3.7

When the automation starts dayahead-optim or publish-data I get errors about NonExistentTimeError. The time zone I use 
[image]

```
2023-03-26 00:45:01,848 - web_server - ERROR - Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2528, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1825, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1823, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1799, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 170, in action_call
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 64, in set_input_data_dict
    fcst = forecast(retrieve_hass_conf, optim_conf, plant_conf,
  File "/usr/local/lib/python3.9/dist-packages/emhass/forecast.py", line 160, in __init__
    self.forecast_dates = pd.date_range(start=self.start_forecast,
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/indexes/extension.py", line 95, in method
    result = attr(self._data, *args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/arrays/datetimelike.py", line 1775, in round
    return self._round(freq, RoundTo.NEAREST_HALF_EVEN, ambiguous, nonexistent)
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/arrays/datetimelike.py", line 1761, in _round
    return result.tz_localize(
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/arrays/_mixins.py", line 84, in method
    return meth(self, *args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/arrays/datetimes.py", line 1043, in tz_localize
    new_dates = tzconversion.tz_localize_to_utc(
  File "pandas/_libs/tslibs/tzconversion.pyx", line 328, in pandas._libs.tslibs.tzconversion.tz_localize_to_utc
pytz.exceptions.NonExistentTimeError: 2023-03-26 02:00:00
```

[5b918bf2_emhass_2023-03-25T23-48-57.328Z.log](https://github.com/davidusb-geek/emhass-add-on/files/11070446/5b918bf2_emhass_2023-03-25T23-48-57.328Z.log)

## System Information

version | core-2023.3.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.10
os_name | Linux
os_version | 5.15.90
arch | x86_64
timezone | Europe/Oslo
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4953
Installed Version | 1.31.0
Stage | running
Available Repositories | 1313
Downloaded Repositories | 6

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | October 7, 2023 at 02:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-central-1-6.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.03.2
agent_version | 1.4.1
docker_version | 20.10.22
disk_total | 116.7 GB
disk_used | 14.3 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.5.0), SSH & Web Terminal (13.0.3), Mosquitto broker (6.1.3), ESPHome (2023.3.1), Z-Wave JS UI (1.8.0), EMHASS (0.3.7), Frigate (0.11.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 2
views | 10
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | March 3, 2023 at 16:47
-- | --
current_recorder_run | March 26, 2023 at 00:43
estimated_db_size | 801.98 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

overas, 2023-03-26:
Could the error be due to daylight saving time changes? Today we changed the clock one hour ahead.
26 Mar 2023 - Daylight Saving Time Started
When local standard time was about to reach
Sunday, 26 March 2023, 02:00:00 clocks were turned forward 1 hour to
Sunday, 26 March 2023, 03:00:00 local daylight time instead.

overas, 2023-03-26:
In the morning I started a dayahead-optim at 09:58. I get error because Nordpool do not have 24 hour with power price, but I do not get the error now.
 
```
2023-03-26 09:58:58,554 - web_server - INFO - Setting up needed data
2023-03-26 09:58:58,557 - web_server - ERROR - ERROR: The passed data is either not a list or the length is not correct, length should be 24
2023-03-26 09:58:58,557 - web_server - ERROR - Passed type is <class 'list'> and length is 24
2023-03-26 09:58:58,557 - web_server - ERROR - ERROR: The passed data is either not a list or the length is not correct, length should be 24
2023-03-26 09:58:58,557 - web_server - ERROR - Passed type is <class 'list'> and length is 24
2023-03-26 09:58:58,561 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2023-03-26 09:59:02,544 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2023-03-26 09:59:02,545 - web_server - INFO - Retrieve hass get data method initiated...
2023-03-26 09:59:03,351 - web_server - INFO -  >> Performing dayahead optimization...
2023-03-26 09:59:03,351 - web_server - INFO - Performing day-ahead forecast optimization
2023-03-26 09:59:03,353 - web_server - INFO - Perform optimization for the day-ahead
2023-03-26 09:59:03,396 - web_server - INFO - Status: Infeasible
2023-03-26 09:59:03,397 - web_server - INFO - Total value of the Cost function = -20.56
2023-03-26 10:00:00,126 - web_server - INFO - Setting up needed data
2023-03-26 10:00:00,132 - web_server - INFO -  >> Publishing data...
2023-03-26 10:00:00,132 - web_server - INFO - Publishing data to HASS instance
2023-03-26 10:00:00,182 - web_server - INFO - Successfully posted to sensor.p_pv_forecast = 432.4
2023-03-26 10:00:00,198 - web_server - INFO - Successfully posted to sensor.p_load_forecast = 823.98
2023-03-26 10:00:00,210 - web_server - INFO - Successfully posted to sensor.p_deferrable0 = 1950.0
2023-03-26 10:00:00,220 - web_server - INFO - Successfully posted to sensor.p_deferrable1 = 0.0
2023-03-26 10:00:00,231 - web_server - INFO - Successfully posted to sensor.p_deferrable2 = 0.0
2023-03-26 10:00:00,242 - web_server - INFO - Successfully posted to sensor.p_deferrable3 = 2968.42
2023-03-26 10:00:00,253 - web_server - INFO - Successfully posted to sensor.p_grid_forecast = 5310.0
2023-03-26 10:00:00,269 - web_server - INFO - Successfully posted to sensor.total_cost_fun_value = -20.56
2023-03-26 10:05:00,112 - web_server - INFO - Setting up needed data
```

overas, 2023-03-26:
Maybe it's related to problems I'm having with Nordpool today.
[image]

```
Logger: homeassistant.components.sensor
Source: custom_components/nordpool/sensor.py:344
Integration: Sensor (documentation, issues)
First occurred: 10:23:47 (2 occurrences)
Last logged: 10:23:47

Error adding entities for domain sensor with platform nordpool
Error while setting up nordpool platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 442, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 717, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 806, in add_to_platform_finish
    await self.async_added_to_hass()
  File "/config/custom_components/nordpool/sensor.py", line 521, in async_added_to_hass
    await self.handle_new_hr()
  File "/config/custom_components/nordpool/sensor.py", line 495, in handle_new_hr
    self._update()
  File "/config/custom_components/nordpool/sensor.py", line 344, in _update
    self._average = mean(today)
  File "/usr/local/lib/python3.10/statistics.py", line 329, in mean
    T, total, count = _sum(data)
  File "/usr/local/lib/python3.10/statistics.py", line 187, in _sum
    T = _coerce(T, typ)  # or raise TypeError
  File "/usr/local/lib/python3.10/statistics.py", line 237, in _coerce
    raise TypeError(msg % (T.__name__, S.__name__))
TypeError: don't know how to coerce float and NoneType
```

It is reported as a bug which are going to be fixed: https://github.com/custom-components/nordpool/issues/129

davidusb-geek, 2023-03-26:
Yes this was related to DST hour change. We were discussing this here: https://community.home-assistant.io/t/emhass-an-energy-management-for-home-assistant/338126/557?u=davidusb
I will eventually transfer this issue to add it to the to do list for EMHASS

overas, 2023-03-26:
Thanks for the reply again. I am closing the case as my Nordpool problem is not related to this case.

closed 2023-03-26 by overas