davidusb-geek/emhass-add-on #42: no valid repository

farshadnl, 2023-04-07:
Hello ,

i tried to add the emhass repository to my HA but it failed with error: No valid repository 
https://github.com/davidusb-geek/emhass.git
Please can you help with this issue?

Many thanks

farshadnl, 2023-04-07:
[image]

davidusb-geek, 2023-04-07:
You are not using the correct repository.

Use this: https://github.com/davidusb-geek/emhass-add-on.git

closed 2023-04-07 by davidusb-geek

farshadnl, 2023-04-09:
Many Thanks.

cloudbr34k84, 2023-07-23:
> You are not using the correct repository.
> 
> Use this: https://github.com/davidusb-geek/emhass-add-on.git

Im trying that link and it's not working as well
https://github.com/davidusb-geek/emhass-add-on.git
[image]

davidusb-geek, 2023-07-23:
Your error message links to emhass.git which like said is not the correct repository, the correct will be emhass-add-on.git. You can try restarting Home Assistant and retry adding the repository