davidusb-geek/emhass-add-on #43: Add-on not upgrading or downloading

purcell-lab, 2023-05-28:
Since about 0.3.9 the add-on hasn't be upgrading and if I remove and try and reinstall it doesn't succeed.

Also the EMHASS icon is broken on the add-on page.

[image]

I can't see any changes made in the repo, so am wondering if HA is processing add-ons differently?

[image]

davidusb-geek, 2023-05-29:
Yes I don't know what is going on here. There has been a change that we are not aware of. I will investigate this.

wesleydv, 2023-05-29:
Not entirely sure, but I believe this is also causing emhass not to show up in the add-on store for new installations after adding the repository. Repo was added successfully and no errors are showing up in the logs.

davidusb-geek, 2023-05-29:
Hi, I just found the issue. It was and easy but tricky fix on the `config.yaml` for add-ons.
I've just manually launched the docker images for rebuild. They should be available in 55min approx.

davidusb-geek, 2023-05-30:
I did had to make another last minute change but it is working now

kcoffau, 2023-05-30:
I updated from 0.3.9 to 0.3.11 (Add-on) and getting below error.  I have uninstalled and re-installed.

Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 331, in <module>
    params = build_params(params, options, args.addon)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 125, in build_params
    params['optim_conf'][19]['set_nodischarge_to_grid'] = options['set_nodischarge_to_grid']
IndexError: list index out of range

Full transparency: I happened to be fiddling with the CURL calls and updated EMHASS at the same time, so no I have NFI if the curl call stuffed it, or the update just has something in it my HA doesn't like.

kcoffau, 2023-05-30:
Yeah, fresh HA and fresh EMHASS install

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 331, in <module>
    params = build_params(params, options, args.addon)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 125, in build_params
    params['optim_conf'][19]['set_nodischarge_to_grid'] = options['set_nodischarge_to_grid']
IndexError: list index out of range

purcell-lab, 2023-05-30:
I can now auto download the new add-on version, but the new version 3.11 fails to load with a similar issue as @kcoffau .

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 331, in <module>
    params = build_params(params, options, args.addon)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 125, in build_params
    params['optim_conf'][19]['set_nodischarge_to_grid'] = options['set_nodischarge_to_grid']
IndexError: list index out of range
```

[image]

wesleydv, 2023-05-30:
Thx David, this solved the issue for me. I was able to install the add-on.

kcoffau, 2023-05-30:
thanks David, this is truely an amazing piece of work :-)

purcell-lab, 2023-05-30:
Thanks David, upgraded and running now.