davidusb-geek/emhass-add-on #45: No data published but logs display data published

vdiogo, 2023-07-02:
Since some days (27June) I don't have data published (and therefore triggering automations) to the EMHASS UI. When checking the logs I can see data published from recent timestamps.

I applied yesterday the latest EMHASS version, but still no updates.
These are the most recent logs:
```
2023-07-02 13:50:01,286 - web_server - INFO -  >> Publishing data...
2023-07-02 13:50:01,287 - web_server - INFO - Publishing data to HASS instance
2023-07-02 13:50:01,875 - web_server - INFO - Successfully posted to sensor.p_pv_forecast = -1.5
2023-07-02 13:50:01,935 - web_server - INFO - Successfully posted to sensor.p_load_forecast = 2427.52
2023-07-02 13:50:01,995 - web_server - INFO - Successfully posted to sensor.p_deferrable0 = 0.0
2023-07-02 13:50:02,042 - web_server - INFO - Successfully posted to sensor.p_grid_forecast = 2429.02
2023-07-02 13:50:02,110 - web_server - INFO - Successfully posted to sensor.total_cost_fun_value = -34.4
2023-07-02 13:50:02,163 - web_server - INFO - Successfully posted to sensor.unit_load_cost = 1.0
2023-07-02 13:50:02,307 - web_server - INFO - Successfully posted to sensor.unit_prod_price = 0.0
2023-07-02 13:55:00,998 - web_server - INFO - Setting up needed data
2023-07-02 13:55:01,422 - web_server - INFO -  >> Publishing data...
2023-07-02 13:55:01,423 - web_server - INFO - Publishing data to HASS instance
2023-07-02 13:55:01,948 - web_server - INFO - Successfully posted to sensor.p_pv_forecast = -1.5
2023-07-02 13:55:02,002 - web_server - INFO - Successfully posted to sensor.p_load_forecast = 2427.52
2023-07-02 13:55:02,056 - web_server - INFO - Successfully posted to sensor.p_deferrable0 = 0.0
2023-07-02 13:55:02,114 - web_server - INFO - Successfully posted to sensor.p_grid_forecast = 2429.02
2023-07-02 13:55:02,198 - web_server - INFO - Successfully posted to sensor.total_cost_fun_value = -34.4
2023-07-02 13:55:02,246 - web_server - INFO - Successfully posted to sensor.unit_load_cost = 1.0
2023-07-02 13:55:02,317 - web_server - INFO - Successfully posted to sensor.unit_prod_price = 0.0
2023-07-02 14:00:00,893 - web_server - INFO - Setting up needed data
2023-07-02 14:00:01,586 - web_server - INFO -  >> Publishing data...
```

This is latest plot:
[image]

davidusb-geek, 2023-07-05:
Have you tested restarting your HA instance and even restarting the machine where your HA instance is running?

vdiogo, 2023-07-05:
> Have you tested restarting your HA instance and even restarting the machine where your HA instance is running?

Yes, did both. Still can see data published in EMHASS logs but not published to the UI or triggering automations

vdiogo, 2023-07-10:
I have reinstalled the add-on and updated the configuration to have two (instead of one) deferrable loads.
Will be checking on the status and will report back. For now, still displays plots results for same date (26June) but logs don't show the previous error:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
2023-07-10 15:13:37,079 - web_server - INFO - Launching the emhass webserver at: http://0.0.0.0:5000
2023-07-10 15:13:37,080 - web_server - INFO - Home Assistant data fetch will be performed using url: http://supervisor/core/api
2023-07-10 15:13:37,080 - web_server - INFO - The data path is: /share
2023-07-10 15:13:37,116 - web_server - INFO - Using core emhass version: 0.4.13
waitress   INFO  Serving on http://0.0.0.0:5000
2023-07-10 15:15:00,681 - web_server - INFO - Setting up needed data
2023-07-10 15:15:01,317 - web_server - INFO -  >> Publishing data...
2023-07-10 15:15:01,318 - web_server - INFO - Publishing data to HASS instance
2023-07-10 15:15:01,757 - web_server - INFO - Successfully posted to sensor.p_pv_forecast = -1.5
2023-07-10 15:15:01,806 - web_server - INFO - Successfully posted to sensor.p_load_forecast = 2427.52
2023-07-10 15:15:01,855 - web_server - INFO - Successfully posted to sensor.p_deferrable0 = 0.0
2023-07-10 15:15:01,857 - web_server - ERROR - P_deferrable1 was not found in results DataFrame. Optimization task may need to be relaunched or it did not converged to a solution.
2023-07-10 15:15:01,902 - web_server - INFO - Successfully posted to sensor.p_grid_forecast = 2429.02
2023-07-10 15:15:01,965 - web_server - INFO - Successfully posted to sensor.total_cost_fun_value = -34.4
2023-07-10 15:15:02,014 - web_server - INFO - Successfully posted to sensor.unit_load_cost = 1.0
2023-07-10 15:15:02,061 - web_server - INFO - Successfully posted to sensor.unit_prod_price = 0.0
```
There is one error shown, which I assume is due to the new deferrable load that was added in the new configuration, which should be gone after some days of data published

vdiogo, 2023-07-12:
Still no data showing on the UI, however I'm getting the following errors on the automations running the shell_commands:

```
Logger: homeassistant.components.shell_command
Source: /usr/src/homeassistant/homeassistant/components/shell_command/__init__.py:117
Integration: Shell Command (documentation, issues)
First occurred: 10 de julho de 2023 às 14:25:03 (7 occurrences)
Last logged: 05:30:28

Error running command: `curl -i -H "Content-Type:application/json" -X POST -d '{}' http://localhost:5000/action/publish-data`, return code: 7
Error running command: `curl -i -H "Content-Type:application/json" -X POST -d '{}' http://localhost:5000/action/dayahead-optim`, return code: 52
NoneType: None

```

davidusb-geek, 2023-07-17:
Did you solved these issues?
I'm not able to understand what is going on with your previous messages.
Post the add-on logged message for better understanding. What is not working?

vdiogo, 2023-07-18:
I can now see the published data (based on day-ahead run I assume):
[image]
So will close this ticket now. Thanks

closed 2023-07-18 by vdiogo