davidusb-geek/emhass-add-on #46: Time out when running naive-mpc-optim

vdiogo, 2023-07-17:
I've setup the solcast integration, with which I have the following for sensor.solcast_24hrs_forecast:

`[3367, 3430, 3443, 3428, 3342, 3206, 3022, 2782, 2486, 2151, 1776, 1372, 959, 523, 214, 80, 31, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 2, 50, 167, 348, 606, 903, 1206, 1501, 1790, 2060, 2333, 2601, 2775]`

I've setup the following shell_command:
`  mpc_optim: "curl -i -H \"Content-Type: application/json\" -X POST -d '{\"pv_power_forecast\":{{states('sensor.solcast_24hrs_forecast')}}}' http://localhost:5000/action/naive-mpc-optim"`

Which returned the following error:
```
Logger: homeassistant.components.shell_command
Source: components/shell_command/__init__.py:86
Integration: Shell Command (documentation, issues)
First occurred: 12:53:05 (1 occurrences)
Last logged: 12:53:05

Timed out running command: `curl -i -H "Content-Type: application/json" -X POST -d '{"pv_power_forecast":{{states('sensor.solcast_24hrs_forecast')}}}' http://localhost:5000/action/naive-mpc-optim`, after: 60s
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/shell_command/__init__.py", line 87, in async_service_handler
    stdout_data, stderr_data = await process.communicate()
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/subprocess.py", line 198, in communicate
    stdin, stdout, stderr = await tasks.gather(stdin, stdout, stderr)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/subprocess.py", line 178, in _read_stream
    output = await stream.read()
             ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/streams.py", line 683, in read
    block = await self.read(self._limit)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/streams.py", line 690, in read
    await self._wait_for_data('read')
  File "/usr/local/lib/python3.11/asyncio/streams.py", line 522, in _wait_for_data
    await self._waiter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/shell_command/__init__.py", line 86, in async_service_handler
    async with async_timeout.timeout(COMMAND_TIMEOUT):
  File "/usr/local/lib/python3.11/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.11/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
TimeoutError

```

vdiogo, 2023-07-18:
> This seems like some kind of connection problem. The shell command is not able to connect with the add-on server at port 5000. Try just to ping it, are you able to do that?

Not really, sorry. Can you guide me here please?

>What's your configuration? Are you using HA OS?
Home Assistant 2023.7.2
Supervisor 2023.07.1
Operating System 10.3
Interface: 20230705.1 - latest

davidusb-geek, 2023-07-18:
So if you try right now EMHASS is working fine?

vdiogo, 2023-07-18:
> So if you try right now EMHASS is working fine?

Yes: all automations are triggering as expected and I can see the forecasts on the UI

vdiogo, 2023-07-18:
> Ok this means that from time to time you may have some connection issues but they have nothing to do with EMHASS

probably yes. Will close this ticket