davidusb-geek/emhass-add-on #49: Functionality

andzie82, 2023-09-04:
I have a problem determining if this is the solution I need. 

I pay for the electricity per hour and the prices get set everyday at 13.00 and from 00.00 to 23.59. 

Ac coupled PV system
Ac coupled battery system

everything is connected to HA. 

I would like to control the charge/discharge/idle using the following parameters from HA to determine which is best between sell/buy/consume

Hourly rate 
Forecast
SOC

for example I have 40% SOC, the forecast will give me 10%, should I charge the batteries with those 10% or is it most affordable to sell them and charge while low price? , I need 70% to ménage all night therefore I will need to charge 20%. When will it be the cheapest time to charge 20% and how long time do I need to charge? Is there other consumers or can I charge at maximum power?

davidusb-geek, 2023-09-15:
It seems like the perfect use case for EMHASS!
PV + battery + variable pricing.
This is why this add-on exists...

andzie82, 2023-09-15:
> It seems like the perfect use case for EMHASS!
> PV + battery + variable pricing.
> This is why this add-on exists...

Great, I’m having some issues setting it up though. Is there a YouTube video I can look at? Haven’t found anything. 

It seems that I have to change the device classes on my entities because they are not showing up.

davidusb-geek, 2023-09-15:
No, there is no youtube video.
You can ask for help directly in the community forum here: https://community.home-assistant.io/t/emhass-an-energy-management-for-home-assistant/

closed 2023-10-15 by davidusb-geek