davidusb-geek/emhass-add-on #50: emhass does not start after initial install

ehajer, 2023-10-13:
EMHASS 0.4
Home Assistant 2023.10.1
Supervisor 2023.10
System Raspberry Pi4
Raspbian 5.10.107-v71+
Frontend 20231005 - latest

After installing Emhass as addon, I tried to start the server with the default configuration from Supervisor panel.
[image]

Changing the configuration in the UI can be done, but saving the configuration after that is not possible.
[image]

In the Supervisor logs I find this errors:
`23-10-13 10:51:09 ERROR (MainThread) [supervisor.addons.addon] Add-on 5b918bf2_emhass has invalid options: not a valid value. Got {'hass_url': 'empty', 'long_lived_token': 'empty', 'costfun': 'profit', 'logging_level': 'INFO', 'optimization_time_step': 30, 'historic_days_to_retrieve': 2, 'method_ts_round': 'nearest', 'set_total_pv_sell': False, 'lp_solver': 'COIN_CMD', 'lp_solver_path': '/usr/bin/cbc', 'set_nocharge_from_grid': False, 'set_nodischarge_to_grid': False, 'set_battery_dynamic': False, 'battery_dynamic_max': 0.9, 'battery_dynamic_min': -0.9, 'load_forecast_method': 'naive', 'sensor_power_photovoltaics': 'sen...
23-10-13 10:51:09 WARNING (MainThread) [supervisor.addons] Can't start Add-on 5b918bf2_emhass`

Even when trying to start the addon in a CLI I get an error:
[image]

I am realy looking forward to start using this great project in my home installation, but at the moment I'm out of options to get it working.

davidusb-geek, 2023-10-13:
Hi, this is the first time that I see this kind of error.

What does the EMHASS logs says?

ehajer, 2023-10-13:
Hi David,

 

Thanks for your fast reaction.

The EMHASS Logs stay empty because EMHASS is not starting at all.

 

Just half hour ago I came a few steps further to a solution:

As mentioned  before, what I did after installing the addon, I made my first settings in the configuration, but when I tried to save these I got an error: “Cannot save configuration…”. 

After putting back the default/build-inn configuration again EMHASS refused to start, pointing me back to the configuration.

I now just copied a configuration file from one of the users on the EMHASS/HomeAssistant community page and I was able to save this as yaml-file and after that Emhass started just Fine.

 

This makes me very happy!! So now I’m working on my one configuration and will start learning how your fantastic project is going to run on my system. There’s a lot of reading to do I ‘ve seen, as well on the forum as in your extensive documentation.

So, many thanks again for your reaction and ofcours for this fantastic project.

 

Question still is, what is wrong with the build-inn configuration file? Maybe you can look in to that?

 

Greetings,

Erik Hajer

The Netherlands

jatty, 2023-10-14:
I think i have tracked down the problem:

The initial configuration file after install contains non-parsable timestamps:

```
list_peak_hours_periods_start_hours:
  - peak_hours_periods_start_hours: "02:54"
  - peak_hours_periods_start_hours: 1044
list_peak_hours_periods_end_hours:
  - peak_hours_periods_end_hours: 924
  - peak_hours_periods_end_hours: 1224
```

When setting right time stamps (like the first one "02:54", even without the quotes), the config file is parsable again.

jatty, 2023-10-14:
There seems to be foo which malforms the time stamps at roll-out for values > 12:00 midday without quotes.
Workaround could be using 12-hour clock values:
```
  - peak_hours_periods_start_hours: 5:24p
  - peak_hours_periods_end_hours: 3:24p
  - peak_hours_periods_end_hours: 8:24p
```

davidusb-geek, 2023-10-15:
This original issue seems to have been solved and a workaround was also provided by @jatty

europer, 2023-10-19:
no, i still have the error, even with the workaround, latest version (installed it yeterday)

europer, 2023-10-22:
Thanks for the last update 0.4.2. But I get the same error. "..Failed to save add-on configuration, not a valid value..."
With default or not. Is the add on version.
Happy sun ☀️☀️☀️🌞

jatty, 2023-10-22:
It seems i don't have tested the patch appropriately.

As a workaround, you should be able to get into the yaml configuration within homeassistant and modify the values in these lines like this:

```
list_peak_hours_periods_start_hours:
  - peak_hours_periods_start_hours: "06:00"
  - peak_hours_periods_start_hours: "17:00"
list_peak_hours_periods_end_hours:
  - peak_hours_periods_end_hours: "09:00"
  - peak_hours_periods_end_hours: "23:59"
```

pbix, 2023-12-16:
Also had this issue today with v0.4.2.  Below is a screenshot showing the invalid time values in the default configuration as they appear right after installation during the initial config.  Reformatting just these two time values was all I needed to do to get it to save properly.

Hope this helps.

[image]

davidusb-geek, 2024-01-15:
This is still an ongoing issue. Reopening it...

nicobresseleers, 2024-01-27:
also have the error:

hass_url: empty
long_lived_token: empty
costfun: profit
logging_level: INFO
optimization_time_step: 30
historic_days_to_retrieve: 2
method_ts_round: nearest
set_total_pv_sell: false
lp_solver: COIN_CMD
lp_solver_path: /usr/bin/cbc
set_nocharge_from_grid: false
set_nodischarge_to_grid: false
set_battery_dynamic: false
battery_dynamic_max: 0.9
battery_dynamic_min: -0.9
load_forecast_method: naive
sensor_power_photovoltaics: sensor.solarlog69_power_ac
sensor_power_load_no_var_loads: sensor.bresseleers_nico_69_load_net
number_of_deferrable_loads: 2
list_nominal_power_of_deferrable_loads:
  - nominal_power_of_deferrable_loads: 3000
  - nominal_power_of_deferrable_loads: 750
list_operating_hours_of_each_deferrable_load:
  - operating_hours_of_each_deferrable_load: 5
  - operating_hours_of_each_deferrable_load: 8
list_peak_hours_periods_start_hours:
  - peak_hours_periods_start_hours: "02:54"
  - peak_hours_periods_start_hours: "17:24"
list_peak_hours_periods_end_hours:
  - peak_hours_periods_end_hours: "15:24"
  - peak_hours_periods_end_hours: "20:24"
list_treat_deferrable_load_as_semi_cont:
  - treat_deferrable_load_as_semi_cont: true
  - treat_deferrable_load_as_semi_cont: true
load_peak_hours_cost: 0.1907
load_offpeak_hours_cost: 0.1419
photovoltaic_production_sell_price: 0.065
maximum_power_from_grid: 9000
list_pv_module_model:
  - pv_module_model: CSUN_Eurasia_Energy_Systems_Industry_and_Trade_CSUN295_60M
list_pv_inverter_model:
  - pv_inverter_model: Fronius_International_GmbH__Fronius_Primo_5_0_1_208_240__240V_
list_surface_tilt:
  - surface_tilt: 30
list_surface_azimuth:
  - surface_azimuth: 205
list_modules_per_string:
  - modules_per_string: 16
list_strings_per_inverter:
  - strings_per_inverter: 1
set_use_battery: false
battery_discharge_power_max: 1000
battery_charge_power_max: 1000
battery_discharge_efficiency: 0.95
battery_charge_efficiency: 0.95
battery_nominal_energy_capacity: 5000
battery_minimum_state_of_charge: 0.3
battery_maximum_state_of_charge: 0.9
battery_target_state_of_charge: 0.6

Logs:
24-01-27 17:34:39 ERROR (MainThread) [supervisor.addons.addon] Add-on 5b918bf2_emhass has invalid options: not a valid value. Got {'hass_url': 'empty', 'long_lived_token': 'empty', 'costfun': 'profit', 'logging_level': 'INFO', 'optimization_time_step': 30, 'historic_days_to_retrieve': 2, 'method_ts_round': 'nearest', 'set_total_pv_sell': False, 'lp_solver': 'COIN_CMD', 'lp_solver_path': '/usr/bin/cbc', 'set_nocharge_from_grid': False, 'set_nodischarge_to_grid': False, 'set_battery_dynamic': False, 'battery_dynamic_max': 0.9, 'battery_dynamic_min': -0.9, 'weight_battery_discharge': 1.0, 'weight_battery_charge': 1.0, 'loa...

24-01-27 17:38:07 ERROR (MainThread) [supervisor.addons.addon] Add-on 5b918bf2_emhass has invalid options: not a valid value. Got {'hass_url': 'empty', 'long_lived_token': 'empty', 'costfun': 'profit', 'logging_level': 'INFO', 'optimization_time_step': 30, 'historic_days_to_retrieve': 2, 'method_ts_round': 'nearest', 'set_total_pv_sell': False, 'lp_solver': 'COIN_CMD', 'lp_solver_path': '/usr/bin/cbc', 'set_nocharge_from_grid': False, 'set_nodischarge_to_grid': False, 'set_battery_dynamic': False, 'battery_dynamic_max': 0.9, 'battery_dynamic_min': -0.9, 'weight_battery_discharge': 1.0, 'weight_battery_charge': 1.0, 'loa...
24-01-27 17:42:16 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running

hours seems fine...

davidusb-geek, 2024-02-10:
Most probably solved with latests releases...

closed 2024-02-10 by davidusb-geek