davidusb-geek/emhass-add-on #52: Dynamic electricity prices

jatty, 2023-10-15:
I'd like to use emhass with dynamic electricity prices like they are provided with [Tibber](https://www.tibber.com/), [E.ON ](https://www.eon.de/de/pk/strom/dynamischer-tarif.html)(german) and others. From 2025 at the latest, electricity suppliers in Germany will be obliged to offer "dynamic electricity tariffs" to end consumers in conjunction with them using smart meters. So this will be an important feature of emhass in the future.

Up to now the config provides "peak hours" and "non-peak hours". With tibber you would have to provide 24 hours which can have an individual pricing for each hour. With other providers there may be prices per 15 minutes, when they correlate to the European spot price market in Germany, France et al. which also use 15 minute slots. These day-ahead-prices are published once a day in the afternoon.

How could it implemented into the software stack rather smooth?

davidusb-geek, 2023-10-15:
This is totally supported by passing your own list of data.
There are examples in the documentation here: https://emhass.readthedocs.io/en/latest/intro.html#passing-your-own-data
And here: https://emhass.readthedocs.io/en/latest/forecasts.html#passing-your-own-forecast-data

Read that carefully and you can easily achieve what you are looking for using Home Assistant templates.
There are also lots of people doing this with their configuration shared in this thread: https://community.home-assistant.io/t/emhass-an-energy-management-for-home-assistant

davidusb-geek, 2023-10-15:
When passing your own data as lists, those "peak hours" and "non-peak hours" for the original configuration will be ignored. Either way you can check that your passed data is correct by using the add-on web ui

davidusb-geek, 2023-10-15:
This is an example shell command from an user passing data from ENTSO-e:
```
trigger_entsoe: "curl -i -H \"Content-Type: application/json\" -X POST -d '{\"load_cost_forecast\":{{((state_attr('sensor.entsoe_average_electricity_price_today', 'prices_today') | map(attribute='price') | list + state_attr('sensor.entsoe_average_electricity_price_today', 'prices_tomorrow') | map(attribute='price') | list))[now().hour:][:24] }},\"prod_price_forecast\":{{((state_attr('sensor.entsoe_average_electricity_price_today', 'prices_today') | map(attribute='price') | list  + state_attr('sensor.entsoe_average_electricity_price_today', 'prices_tomorrow') | map(attribute='price') | list))[now().hour:][:24]}}}' http://localhost:5000/action/dayahead-optim"
```

jatty, 2023-10-15:
Didn't had a clue yet, thanks so much.

closed 2023-10-19 by davidusb-geek