davidusb-geek/emhass-add-on #54: Problems with saving the configuration

copadewe, 2023-10-19:
First, this looks like a gamechanger :-) so many people looking for it.
SO, i installed the Addon today, went to Configuration and set all my settings.
But after click on "save" i got error warnings.
This is good if there is something wrong, but there are in my eyes "all" fields noted.
[image]

Any idea how to solve it? Here the yaml:
hass_url: 192.168.178.55
long_lived_token: >-
  xxxxxxxxxxxxxxxxxxxxxxxxxxxxx
costfun: self-consumption
logging_level: DEBUG
optimization_time_step: 30
historic_days_to_retrieve: 2
method_ts_round: nearest
set_total_pv_sell: false
lp_solver: COIN_CMD
lp_solver_path: /usr/bin/cbc
set_nocharge_from_grid: false
set_nodischarge_to_grid: false
set_battery_dynamic: false
battery_dynamic_max: 0.9
battery_dynamic_min: -0.9
load_forecast_method: naive
sensor_power_photovoltaics: sensor.total_dc_power
sensor_power_load_no_var_loads: sensor.load_power
number_of_deferrable_loads: 2
list_nominal_power_of_deferrable_loads:
  - nominal_power_of_deferrable_loads: 3000
  - nominal_power_of_deferrable_loads: 750
list_operating_hours_of_each_deferrable_load:
  - operating_hours_of_each_deferrable_load: 5
  - operating_hours_of_each_deferrable_load: 8
list_peak_hours_periods_start_hours:
  - peak_hours_periods_start_hours: "05:54"
  - peak_hours_periods_start_hours: 624
list_peak_hours_periods_end_hours:
  - peak_hours_periods_end_hours: "09:24"
  - peak_hours_periods_end_hours: 714
list_treat_deferrable_load_as_semi_cont:
  - treat_deferrable_load_as_semi_cont: true
  - treat_deferrable_load_as_semi_cont: true
load_peak_hours_cost: 0.1907
load_offpeak_hours_cost: 0.1419
photovoltaic_production_sell_price: 0.065
maximum_power_from_grid: 9000
list_pv_module_model:
  - pv_module_model: CSUN_Eurasia_Energy_Systems_Industry_and_Trade_CSUN295_60M
list_pv_inverter_model:
  - pv_inverter_model: Fronius_International_GmbH__Fronius_Primo_5_0_1_208_240__240V_
list_surface_tilt:
  - surface_tilt: 30
list_surface_azimuth:
  - surface_azimuth: 205
list_modules_per_string:
  - modules_per_string: 16
list_strings_per_inverter:
  - strings_per_inverter: 1
set_use_battery: true
battery_discharge_power_max: 1000
battery_charge_power_max: 1000
battery_discharge_efficiency: 0.95
battery_charge_efficiency: 0.95
battery_nominal_energy_capacity: 5000
battery_minimum_state_of_charge: 0.3
battery_maximum_state_of_charge: 0.9
battery_target_state_of_charge: 0.6

Thx, Oliver

davidusb-geek, 2023-10-19:
Just released a new version. Try to update later. It should be solved.

copadewe, 2023-10-20:
> Just released a new version. Try to update later. It should be solved.

Thanks, but same again.
[image]

davidusb-geek, 2023-10-20:
This is related to #50
Any chance you can post the full log error? It is truncated. Go to the supervisor logger.
Also you have commas in your values: "0,3" should be "0.3"

copadewe, 2023-10-20:
This commas are there, i did not changed it.
Will see to get full error

copadewe, 2023-10-21:
I just took a copy of the yaml from this:
https://community.home-assistant.io/t/emhass-add-on-an-energy-management-optimization-add-on-for-home-assistant-os-and-supervised/405649/136?u=copadewe
and it work with this when I paste and copy it to my System.
I can also make changes, no idea why but it work with this now.
Have other problem - will first feed all my details and come back with another issue.
Oliver

copadewe, 2023-10-21:
THX @davidusb-geek

davidusb-geek, 2024-01-15:
This is still an ongoing issue. Reopening it...

nicobresseleers, 2024-01-27:
Same issue here..  default values, changing load and solar sensors: does not work.

davidusb-geek, 2024-01-27:
Just check `peak_hours_periods_start_hours` and `peak_hours_periods_end_hours` format and save configuration and it should work. But I agree there is an annoyinbg issue changing this automatically to the wrong format

nicobresseleers, 2024-01-27:
yes indeed .. working now..

it's not (yet) easy to understand how it works in detail .. need to study it a bit more. :-)

Would like to  ( based on consumption, weather/sun / battery Soc /dynamic energy pricing)

When to load battery and when to sell again and make profit:
High level:
charge between 03:00 and 06:00 6kW ( full battery)
discharge between 06 and 08 4kW ( or more if solar is expected)
Load again in the afternoon based on SoC and consumption planning make sure to have enough to cover peak and evening ( till 02 or 03 depending on when the low prices are planned)

Also trigger invertor when utilization is > 2500 W ( we have peak taxes in belgium if we consume > 2500 W avg for over 15minutes) -> now already done via automation

davidusb-geek, 2024-02-10:
This should be solved with latests versions... Feel free to reopen otherwise

closed 2024-02-10 by davidusb-geek