davidusb-geek/emhass-add-on #55: Error After Last Update

prinzkafka, 2023-10-20:
- Retrieving data from hass for load forecast using method = mlforecaster
2023-10-20 21:30:02,764 - web_server - INFO - Retrieve hass get data method initiated...
/usr/local/lib/python3.9/dist-packages/sklearn/base.py:348: InconsistentVersionWarning: Trying to unpickle estimator KNeighborsRegressor from version 1.3.0 when using version 1.3.1. This might lead to breaking code or invalid results. Use at your own risk. For more info please refer to:
https://scikit-learn.org/stable/model_persistence.html#security-maintainability-limitations
  warnings.warn(
2023-10-20 21:30:06,193 - web_server - ERROR - Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1455, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 869, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 867, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 852, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 179, in action_call
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 91, in set_input_data_dict
    P_load_forecast = fcst.get_load_forecast(method=optim_conf['load_forecast_method'])
  File "/usr/local/lib/python3.9/dist-packages/emhass/forecast.py", line 615, in get_load_forecast
    forecast_out = mlf.predict(data_last_window)
  File "/usr/local/lib/python3.9/dist-packages/emhass/machine_learning_forecaster.py", line 206, in predict
    predictions = self.forecaster.predict(steps=self.lags_opt,
  File "/usr/local/lib/python3.9/dist-packages/skforecast/ForecasterAutoreg/ForecasterAutoreg.py", line 696, in predict
    if self.differentiation is not None:
AttributeError: 'ForecasterAutoreg' object has no attribute 'differentiation'

copadewe, 2023-10-21:
@prinzkafka Is this comming up after last 10.4 update of HA?

prinzkafka, 2023-10-24:
did a new forecast-model-fit - now it seems to work again

closed 2023-10-24 by prinzkafka