davidusb-geek/emhass-add-on #58: Different timezones

tore1980, 2023-10-30:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
2023-10-30 09:40:51,746 - web_server - INFO - Launching the emhass webserver at: http://0.0.0.0:5000
2023-10-30 09:40:51,746 - web_server - INFO - Home Assistant data fetch will be performed using url: http://supervisor/core/api
2023-10-30 09:40:51,746 - web_server - INFO - The data path is: /share
2023-10-30 09:40:51,747 - web_server - INFO - Using core emhass version: 0.5.1
waitress   INFO  Serving on http://0.0.0.0:5000
2023-10-30 09:40:58,343 - web_server - INFO - EMHASS server online, serving index.html...
2023-10-30 09:41:00,691 - web_server - INFO - Setting up needed data
2023-10-30 09:41:00,711 - web_server - INFO - Retrieve hass get data method initiated...
2023-10-30 09:41:06,651 - web_server - INFO -  >> Performing perfect optimization...
2023-10-30 09:41:06,651 - web_server - INFO - Performing perfect forecast optimization
2023-10-30 09:41:06,653 - web_server - INFO - Perform optimization for perfect forecast scenario
2023-10-30 09:41:06,653 - web_server - INFO - Solving for day: 28-10-2023
2023-10-30 09:41:06,654 - web_server - ERROR - Exception on /action/perfect-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/arrays/datetimes.py", line 2446, in _infer_tz_from_endpoints
    inferred_tz = timezones.infer_tzinfo(start, end)
  File "pandas/_libs/tslibs/timezones.pyx", line 328, in pandas._libs.tslibs.timezones.infer_tzinfo
AssertionError: Inputs must both have the same timezone, pytz.FixedOffset(120) != pytz.FixedOffset(60)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1455, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 869, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 867, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 852, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 188, in action_call
    opt_res = perfect_forecast_optim(input_data_dict, app.logger)
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 199, in perfect_forecast_optim
    opt_res = input_data_dict['opt'].perform_perfect_forecast_optim(df_input_data, input_data_dict['days_list'])
  File "/usr/local/lib/python3.9/dist-packages/emhass/optimization.py", line 488, in perform_perfect_forecast_optim
    data_tp = df_input_data.copy().loc[pd.date_range(start=day_start, end=day_end, freq=self.freq)]
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/indexes/datetimes.py", line 1070, in date_range
    dtarr = DatetimeArray._generate_range(
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/arrays/datetimes.py", line 422, in _generate_range
    tz = _infer_tz_from_endpoints(start, end, tz)
  File "/usr/local/lib/python3.9/dist-packages/pandas/core/arrays/datetimes.py", line 2449, in _infer_tz_from_endpoints
    raise TypeError(
TypeError: Start and end cannot both be tz-aware with different timezones
2023-10-30 09:41:09,586 - web_server - INFO - Setting up needed data
2023-10-30 09:41:09,588 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2023-10-30 09:41:10,325 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2023-10-30 09:41:10,325 - web_server - INFO - Retrieve hass get data method initiated...
2023-10-30 09:41:19,416 - web_server - INFO -  >> Performing dayahead optimization...
2023-10-30 09:41:19,416 - web_server - INFO - Performing day-ahead forecast optimization
2023-10-30 09:41:19,418 - web_server - INFO - Perform optimization for the day-ahead
2023-10-30 09:41:19,461 - web_server - INFO - Status: Infeasible
2023-10-30 09:41:19,461 - web_server - INFO - Total value of the Cost function = -87.22

[image]

davidusb-geek, 2023-11-01:
This is related to DST switch. It should all be working correctly now.

tore1980, 2023-11-02:
Ok, thank you!

closed 2023-12-17 by davidusb-geek