davidusb-geek/emhass-add-on #59: Deferrable load as a semi-continuous variable

tore1980, 2023-11-02:
I have defined "treat_deferrable_load_as_semi_cont: true", On version: 0.4.2 this setting doesn't seem to work.

CONFIG:

hass_url: empty
long_lived_token: empty
costfun: profit
logging_level: DEBUG
optimization_time_step: 60
historic_days_to_retrieve: 2
method_ts_round: nearest
set_total_pv_sell: false
lp_solver: COIN_CMD
lp_solver_path: /usr/bin/cbc
set_nocharge_from_grid: false
set_nodischarge_to_grid: false
set_battery_dynamic: false
battery_dynamic_max: 0.9
battery_dynamic_min: -0.9
load_forecast_method: naive
sensor_power_photovoltaics: sensor.solis_ac_output_total_power
sensor_power_load_no_var_loads: sensor.emhass_power
number_of_deferrable_loads: 8
list_nominal_power_of_deferrable_loads:
  - nominal_power_of_deferrable_loads: 3750
  - nominal_power_of_deferrable_loads: 1500
  - nominal_power_of_deferrable_loads: 850
  - nominal_power_of_deferrable_loads: 2200
  - nominal_power_of_deferrable_loads: 700
  - nominal_power_of_deferrable_loads: 4000
  - nominal_power_of_deferrable_loads: 1000
  - nominal_power_of_deferrable_loads: 1500
list_operating_hours_of_each_deferrable_load:
  - operating_hours_of_each_deferrable_load: 2
  - operating_hours_of_each_deferrable_load: 4
  - operating_hours_of_each_deferrable_load: 7
  - operating_hours_of_each_deferrable_load: 2
  - operating_hours_of_each_deferrable_load: 3
  - operating_hours_of_each_deferrable_load: 3
  - operating_hours_of_each_deferrable_load: 0
  - operating_hours_of_each_deferrable_load: 1
list_peak_hours_periods_start_hours:
  - peak_hours_periods_start_hours: "07:00"
list_peak_hours_periods_end_hours:
  - peak_hours_periods_end_hours: "07:01"
list_treat_deferrable_load_as_semi_cont:
  - treat_deferrable_load_as_semi_cont: true
  - treat_deferrable_load_as_semi_cont: true
  - treat_deferrable_load_as_semi_cont: true
  - treat_deferrable_load_as_semi_cont: true
  - treat_deferrable_load_as_semi_cont: true
  - treat_deferrable_load_as_semi_cont: true
  - treat_deferrable_load_as_semi_cont: true
  - treat_deferrable_load_as_semi_cont: true
load_peak_hours_cost: 2
load_offpeak_hours_cost: 1
photovoltaic_production_sell_price: 1
maximum_power_from_grid: 5000
list_pv_module_model:
  - pv_module_model: Jinko_Solar_Co___Ltd_JKM405M_72HL
  - pv_module_model: Jinko_Solar_Co___Ltd_JKM405M_72HL
list_pv_inverter_model:
  - pv_inverter_model: Fronius_International_GmbH__Fronius_Primo_15_0_1_208_240__240V_
  - pv_inverter_model: Fronius_International_GmbH__Fronius_Primo_15_0_1_208_240__240V_
list_surface_tilt:
  - surface_tilt: 22
  - surface_tilt: 22
list_surface_azimuth:
  - surface_azimuth: 128
  - surface_azimuth: 255
list_modules_per_string:
  - modules_per_string: 20
  - modules_per_string: 10
list_strings_per_inverter:
  - strings_per_inverter: 1
  - strings_per_inverter: 1
set_use_battery: false
battery_discharge_power_max: 1
battery_charge_power_max: 10996
battery_discharge_efficiency: 0.95
battery_charge_efficiency: 0.95
battery_nominal_energy_capacity: 40000
battery_minimum_state_of_charge: 0.3
battery_maximum_state_of_charge: 0.9
battery_target_state_of_charge: 0.6

LOG:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
2023-11-02 18:54:01,759 - web_server - INFO - Launching the emhass webserver at: http://0.0.0.0:5000
2023-11-02 18:54:01,759 - web_server - INFO - Home Assistant data fetch will be performed using url: http://supervisor/core/api
2023-11-02 18:54:01,759 - web_server - INFO - The data path is: /share
2023-11-02 18:54:01,760 - web_server - INFO - Using core emhass version: 0.5.1
waitress   INFO  Serving on http://0.0.0.0:5000
2023-11-02 18:58:03,405 - web_server - INFO - Setting up needed data
2023-11-02 18:58:03,423 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2023-11-02 18:58:04,194 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2023-11-02 18:58:04,195 - web_server - INFO - Retrieve hass get data method initiated...
2023-11-02 18:58:13,244 - web_server - INFO -  >> Performing dayahead optimization...
2023-11-02 18:58:13,244 - web_server - INFO - Performing day-ahead forecast optimization
2023-11-02 18:58:13,252 - web_server - INFO - Perform optimization for the day-ahead
2023-11-02 18:58:13,294 - web_server - INFO - Status: Infeasible
2023-11-02 18:58:13,294 - web_server - INFO - Total value of the Cost function = -95.9

[image]

shell_command:
  publish_data: "curl -i -H 'Content-Type:application/json' -X POST -d '{}' http://localhost:5000/action/publish-data"
  
  post_nordpool_forecast: "curl -i -H 'Content-Type: application/json' -X POST -d '{\"load_cost_forecast\":{{(
        (state_attr('sensor.nordpool_kwh_oslo_nok_3_095_025', 'raw_tomorrow')|map(attribute='value')|list)[:24])
        }},\"prod_price_forecast\":{{(
        (state_attr('sensor.nordpool_kwh_oslo_sol_netleie', 'raw_tomorrow')|map(attribute='value')|list)[:24])
        }},\"def_total_hours\":{{([
        states('input_number.emhass_vvb_operasjonstid')|int(0),
        states('input_number.emhass_gang_varme_operasjonstid')|int(0),
        states('input_number.emhass_bad_varme_operasjonstid')|int(0),
        states('input_number.emhass_garasje_bod_operasjonstid')|int(0),
        states('input_number.emhass_bad_nede_varme_operasjonstid')|int(0),
        states('input_number.emhass_bil_lader_operasjonstid')|int(0),
        states('input_number.emhass_luftvp_operasjonstid')|int(0),
        states('input_number.emhass_vannvp_operasjonstid')|int(0)
        ] |list)[:8]}}}' http://localhost:5000/action/dayahead-optim"

shell_command:
  publish_data: "curl -i -H 'Content-Type:application/json' -X POST -d '{}' http://localhost:5000/action/publish-data"
  
  post_nordpool_forecast: "curl -i -H 'Content-Type: application/json' -X POST -d '{\"load_cost_forecast\":[1.305, 0.97, 0.795, 0.73, 0.791, 1.176, 1.515, 1.604, 1.611, 1.609, 1.608, 1.604, 1.598, 1.593, 1.597, 1.595, 1.601, 1.6, 1.602, 1.598, 1.576, 1.508, 1.318, 1.189],\"prod_price_forecast\":[0.574, 0.322, 0.191, 0.142, 0.188, 0.477, 0.673, 0.759, 0.778, 0.773, 0.771, 0.759, 0.742, 0.731, 0.741, 0.735, 0.752, 0.748, 0.754, 0.743, 0.719, 0.668, 0.583, 0.486],\"def_total_hours\":[3, 0, 0, 0, 0, 4, 0, 0]}' http://localhost:5000/action/dayahead-optim"

davidusb-geek, 2023-12-19:
Is not that "it doesn't work". It is just that you are adding too much constraints to your problem to the point that it is just infeasible. Try relaxing that, at least for some deferrable loads and see how it goes.

davidusb-geek, 2024-01-29:
Hi. Can you please try this with the latest version?
It should work better

davidusb-geek, 2024-02-10:
Any problem with this should have been solved with latest versions...

closed 2024-02-10 by davidusb-geek