davidusb-geek/emhass-add-on #61: EMHASS fails to start after 0.5.0 update

ragg987, 2023-12-17:
Hi, just updated to 0.5.0 and EMHASS is not starting correctly. Mine is a HAOS install which is all up-to-date (Core 2023.12.3).

Symptoms:
On starting add-on through the add-on menu of HAOS it shows started and consuming low CPU and memory. However if I switch to the EMHASS dashboard i get an error message "The add-on seems to not be ready, it might still be starting. Do you want to try again?" Clicking Yes does not work. Clicking No gives "502 Bad Gateway".

I have restarted HA and still getting the same. EMHASS logs
`Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.9/dist-packages/emhass/web_server.py", line 15, in <module>
    from emhass.command_line import set_input_data_dict
  File "/usr/local/lib/python3.9/dist-packages/emhass/command_line.py", line 20, in <module>
    from emhass.forecast import forecast
  File "/usr/local/lib/python3.9/dist-packages/emhass/forecast.py", line 15, in <module>
    import pvlib
  File "/usr/local/lib/python3.9/dist-packages/pvlib/__init__.py", line 3, in <module>
    from pvlib import (  # noqa: F401
  File "/usr/local/lib/python3.9/dist-packages/pvlib/clearsky.py", line 14, in <module>
    import h5py
  File "/usr/lib/python3/dist-packages/h5py/__init__.py", line 21, in <module>
    from . import _debian_h5py_serial as _h5py
  File "/usr/lib/python3/dist-packages/h5py/_debian_h5py_serial/__init__.py", line 46, in <module>
    from ._conv import register_converters as _register_converters
  File "h5py/_debian_h5py_serial/_conv.pyx", line 1, in init h5py._debian_h5py_serial._conv
  File "h5py/_debian_h5py_serial/h5t.pyx", line 293, in init h5py._debian_h5py_serial.h5t
  File "/usr/local/lib/python3.9/dist-packages/numpy/__init__.py", line 333, in __getattr__
    raise AttributeError("module {!r} has no attribute "
AttributeError: module 'numpy' has no attribute 'typeDict'
s6-supervise emhass: warning: unable to spawn ./finish: Permission denied`

ragg987, 2023-12-17:
To add it was working fine on the previous release. The HA logs do not show any error.

davidusb-geek, 2023-12-17:
it should be solved, upgrade to add-on v0.5.2 when it is available.
Please close the issue if everything goes back to normal

ragg987, 2023-12-18:
All good now, thanks for the fast fix. Closing issue.

Now I can get back to figuring the capability...

closed 2023-12-18 by ragg987