davidusb-geek/emhass-add-on #65: Status: Infeasible when using maximum_power_from_grid: 5310, but not when I use default value 9000

overas, 2024-01-06:
EMHASS add-on v0.5.4
Home Assistant 2024.1.2
Intel Nuc 5 with Hassos 

When I use the maximum_power_from_grid: 5310, I get Status: Infeasible. When I use default value for maximum_power_from_grid: 9000 I get Status: Optimal. 

[image]

Is this a bug or can I use maximum_power_from_grid: 9000 without any problem?

[image]

When running RESTful Command: trigger_forecast with maximum_power_from_grid: 5310:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
2024-01-06 22:56:07,855 - web_server - INFO - Launching the emhass webserver at: http://0.0.0.0:5000
2024-01-06 22:56:07,855 - web_server - INFO - Home Assistant data fetch will be performed using url: http://supervisor/core/api
2024-01-06 22:56:07,855 - web_server - INFO - The data path is: /share
2024-01-06 22:56:07,857 - web_server - INFO - Using core emhass version: 0.6.2
waitress   INFO  Serving on http://0.0.0.0:5000
2024-01-06 22:56:22,743 - web_server - INFO - Setting up needed data
2024-01-06 22:56:22,774 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2024-01-06 22:56:23,845 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2024-01-06 22:56:23,846 - web_server - INFO - Retrieve hass get data method initiated...
2024-01-06 22:56:26,409 - web_server - INFO -  >> Performing dayahead optimization...
2024-01-06 22:56:26,409 - web_server - INFO - Performing day-ahead forecast optimization
2024-01-06 22:56:26,417 - web_server - INFO - Perform optimization for the day-ahead
2024-01-06 22:56:26,435 - web_server - INFO - Status: Infeasible
2024-01-06 22:56:26,435 - web_server - INFO - Total value of the Cost function = -237.33
```

When running RESTful Command: trigger_forecast with maximum_power_from_grid: 9000:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
2024-01-06 22:58:37,305 - web_server - INFO - Launching the emhass webserver at: http://0.0.0.0:5000
2024-01-06 22:58:37,305 - web_server - INFO - Home Assistant data fetch will be performed using url: http://supervisor/core/api
2024-01-06 22:58:37,305 - web_server - INFO - The data path is: /share
2024-01-06 22:58:37,306 - web_server - INFO - Using core emhass version: 0.6.2
waitress   INFO  Serving on http://0.0.0.0:5000
2024-01-06 22:58:49,892 - web_server - INFO - Setting up needed data
2024-01-06 22:58:49,929 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2024-01-06 22:58:51,020 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2024-01-06 22:58:51,020 - web_server - INFO - Retrieve hass get data method initiated...
2024-01-06 22:58:53,518 - web_server - INFO -  >> Performing dayahead optimization...
2024-01-06 22:58:53,518 - web_server - INFO - Performing day-ahead forecast optimization
2024-01-06 22:58:53,529 - web_server - INFO - Perform optimization for the day-ahead
2024-01-06 22:58:53,554 - web_server - INFO - Status: Optimal
2024-01-06 22:58:53,554 - web_server - INFO - Total value of the Cost function = -228.02
```

RESTful Command: trigger_forecast:
```
  trigger_forecast:
    url: http://localhost:5000/action/dayahead-optim
    method: POST
    content_type: "application/json"
    timeout: 300
    payload: >-
      {
      "load_cost_forecast":{{((state_attr('sensor.nordpool_total_pris_med_stromstotte_og_energiledd', 'raw_today') | map(attribute='value') | list  + state_attr('sensor.nordpool_total_pris_med_stromstotte_og_energiledd','raw_tomorrow') | map(attribute='value') | list))[now().hour:][:24] }},
      "prod_price_forecast":{{((state_attr('sensor.nordpool_energipris', 'raw_today') | map(attribute='value') | list  + state_attr('sensor.nordpool_energipris', 'raw_tomorrow') | map(attribute='value') | list))[now().hour:][:24]}}
      }
```

Emhass addon config:

root@5b918bf2-emhass:/data# cat options.json
```
{
  "hass_url": "empty",
  "long_lived_token": "empty",
  "costfun": "profit",
  "logging_level": "DEBUG",
  "optimization_time_step": 60,
  "historic_days_to_retrieve": 2,
  "method_ts_round": "nearest",
  "set_total_pv_sell": false,
  "lp_solver": "COIN_CMD",
  "lp_solver_path": "/usr/bin/cbc",
  "set_nocharge_from_grid": false,
  "set_nodischarge_to_grid": false,
  "set_battery_dynamic": false,
  "battery_dynamic_max": 0.9,
  "battery_dynamic_min": -0.9,
  "weight_battery_discharge": 0.0,
  "weight_battery_charge": 0.0,
  "load_forecast_method": "naive",
  "sensor_power_photovoltaics": "sensor.ecu_current_power",
  "sensor_power_load_no_var_loads": "sensor.power_load_no_var_loads",
  "number_of_deferrable_loads": 7,
  "list_nominal_power_of_deferrable_loads": [
    {
      "nominal_power_of_deferrable_loads": 1964
    },
    {
      "nominal_power_of_deferrable_loads": 2622
    },
    {
      "nominal_power_of_deferrable_loads": 2193
    },
    {
      "nominal_power_of_deferrable_loads": 931
    },
    {
      "nominal_power_of_deferrable_loads": 1450
    },
    {
      "nominal_power_of_deferrable_loads": 652
    },
    {
      "nominal_power_of_deferrable_loads": 863
    }
  ],
  "list_operating_hours_of_each_deferrable_load": [
    {
      "operating_hours_of_each_deferrable_load": 8
    },
    {
      "operating_hours_of_each_deferrable_load": 8
    },
    {
      "operating_hours_of_each_deferrable_load": 8
    },
    {
      "operating_hours_of_each_deferrable_load": 8
    },
    {
      "operating_hours_of_each_deferrable_load": 8
    },
    {
      "operating_hours_of_each_deferrable_load": 8
    },
    {
      "operating_hours_of_each_deferrable_load": 8
    }
  ],
  "list_peak_hours_periods_start_hours": [
    {
      "peak_hours_periods_start_hours": "06:00"
    }
  ],
  "list_peak_hours_periods_end_hours": [
    {
      "peak_hours_periods_end_hours": "22:00"
    }
  ],
  "list_treat_deferrable_load_as_semi_cont": [
    {
      "treat_deferrable_load_as_semi_cont": false
    },
    {
      "treat_deferrable_load_as_semi_cont": false
    },
    {
      "treat_deferrable_load_as_semi_cont": false
    },
    {
      "treat_deferrable_load_as_semi_cont": false
    },
    {
      "treat_deferrable_load_as_semi_cont": false
    },
    {
      "treat_deferrable_load_as_semi_cont": false
    },
    {
      "treat_deferrable_load_as_semi_cont": false
    }
  ],
  "load_peak_hours_cost": 0.1907,
  "load_offpeak_hours_cost": 0.1419,
  "photovoltaic_production_sell_price": 0.065,
  "maximum_power_from_grid": 9000,
  "list_pv_module_model": [
    {
      "pv_module_model": "REC_Solar_REC295TP2"
    },
    {
      "pv_module_model": "REC_Solar_REC295TP2"
    },
    {
      "pv_module_model": "REC_Solar_REC295TP2"
    }
  ],
  "list_pv_inverter_model": [
    {
      "pv_inverter_model": "Altenergy_Power_System_Inc___QS1__240V_"
    },
    {
      "pv_inverter_model": "Altenergy_Power_System_Inc___QS1__240V_"
    },
    {
      "pv_inverter_model": "Altenergy_Power_System_Inc___YC600__240V_"
    }
  ],
  "list_surface_tilt": [
    {
      "surface_tilt": 38
    },
    {
      "surface_tilt": 41
    },
    {
      "surface_tilt": 41
    }
  ],
  "list_surface_azimuth": [
    {
      "surface_azimuth": 225
    },
    {
      "surface_azimuth": 225
    },
    {
      "surface_azimuth": 225
    }
  ],
  "list_modules_per_string": [
    {
      "modules_per_string": 12
    },
    {
      "modules_per_string": 4
    },
    {
      "modules_per_string": 2
    }
  ],
  "list_strings_per_inverter": [
    {
      "strings_per_inverter": 1
    },
    {
      "strings_per_inverter": 1
    },
    {
      "strings_per_inverter": 1
    }
  ],
  "set_use_battery": false,
  "battery_discharge_power_max": 1000,
  "battery_charge_power_max": 1000,
  "battery_discharge_efficiency": 0.95,
  "battery_charge_efficiency": 0.95,
  "battery_nominal_energy_capacity": 5000,
  "battery_minimum_state_of_charge": 0.3,
  "battery_maximum_state_of_charge": 0.9,
  "battery_target_state_of_charge": 0.6
```

davidusb-geek, 2024-01-06:
Ok but what is your power production and consumption. If you constraint the grid power do you have enough available power to supply your load.
Can you show the results table from the webui?

overas, 2024-01-07:
Here is the results table when I use default value for maximum_power_from_grid: 9000. If you want the results table for maximum_power_from_grid: 5310 I can arrange that too.

[image]
[image]

Here is the template for sensor.power_load_no_var_loads I use::
```
  - sensor:
      - name: "Power load no var loads"
        unique_id: fbfeef21-1aa3-4a54-b781-426f46fef597
        unit_of_measurement: W
        device_class: power
        state: >
          {% set powerload = states('sensor.total_load_consumption') | float(default=0) %}
          {% set vkbad1etg = states('sensor.varmekabel_bad1etg_electric_consumption_w') | float(default=0) %}
          {% set vkbad2etg = states('sensor.varmekabel_bad2etg_electric_consumption_w') | float(default=0) %}
          {% set vkgang = states('sensor.varmekabel_gang_electric_consumption_w') | float(default=0) %}
          {% set vkgmlstue = states('sensor.varmekabel_gmlstue_electric_consumption_w') | float(default=0) %}
          {% set vknystue = states('sensor.varmekabel_nystue_electric_consumption_w') | float(default=0) %}
          {% set vkkjokken = states('sensor.varmekabel_kjokken_electric_consumption_w') | float(default=0) %}
          {% set vvb = states('sensor.bryter_varmvannsbereder_electric_consumption_w') | float(default=0) %}
          {% set value = (powerload - vkbad1etg - vkbad2etg - vkgang - vkgmlstue - vknystue - vkkjokken - vvb) | round(1,default=0) %}
          {% if value < 0.0  %}
            {{ 0.0 }}
          {% else %}
            {{ value }}
          {% endif %}
```

For calculating the sensor.total_load_consumption for my house I use this template:
```
      - name: "Total load consumption"
        unique_id: c68d8562-abf0-42eb-8a72-e93e57b8b88a
        unit_of_measurement: "W"
        state: >
          {% set imported = states('sensor.power_toraldasen_32')|float(0) %}
          {% set pvproduced = states('sensor.ecu_current_power')|float(0) %}
          {% set export = states('sensor.power_production_toraldasen_32')|float(0) %}
          {{ (imported + pvproduced) - export}}
```

davidusb-geek, 2024-01-07:
Take a look at your P_grid most of the time it is higher than 5300. So if you constraint that unfeasible condition seems inevitable here. You will need to reduce either the number of deferrable loads that you have, their maximum power or the total operating time of each deferrable load.

overas, 2024-01-07:
Thank you for your help. I going to try to reduce the deferrable loads or time of each deferrable load.

closed 2024-01-17 by overas