davidusb-geek/emhass-add-on #72: Update add-on to (always) use latest EMHASS version

andreas-bulling, 2024-01-29:
The add-on is quite a bit behind in terms of version number. EMHASS is at 0.7.0 already while the add-on is at 0.5.4 since one month.

Any reason this is not kept in sync? Can one replace/update EMHASS to the latest version "inside" the add-on?

davidusb-geek, 2024-01-29:
Improvements and fixes on the main core code and the add-on code just don't follow the same pace. That's how it is.
I release a new add-on version when everything is well tested.

andreas-bulling, 2024-01-29:
I just wrote this and a few minutes later received the notification about the available add-on update. However, I see the following error when trying to update:

`Can't install davidusb/image-amd64-emhass:0.6.0: 404 Client Error for http+docker://localhost/v1.43/images/create?tag=0.6.0&fromImage=davidusb%2Fimage-amd64-emhass&platform=linux%2Famd64: Not Found ("manifest for davidusb/image-amd64-emhass:0.6.0 not found: manifest unknown: manifest unknown") `

andreas-bulling, 2024-01-29:
Nevermind, now it worked.

closed 2024-01-29 by andreas-bulling