davidusb-geek/emhass-add-on #74: emhass 0.6.2 IndexError: list index out of range

overas, 2024-02-04:
Intel Nuc 5i
Hassos 2024.1.6

With emhass 0.6.2 something changed. I get "IndexError: list index out of range" when I do a day-ahead optimization with this rest command:

```
  trigger_forecast:
    url: http://localhost:5000/action/dayahead-optim
    method: POST
    content_type: "application/json"
    timeout: 300
    payload: >-
      {
      "load_cost_forecast":{{((state_attr('sensor.nordpool_total_pris_med_energiledd', 'raw_today') | map(attribute='value') | list  + state_attr('sensor.nordpool_total_pris_med_energiledd','raw_tomorrow') | map(attribute='value') | list))[now().hour:][:24] }},
      "prod_price_forecast":{{((state_attr('sensor.nordpool_energipris', 'raw_today') | map(attribute='value') | list  + state_attr('sensor.nordpool_energipris', 'raw_tomorrow') | map(attribute='value') | list))[now().hour:][:24]}}
      }
```

The log gives:

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
2024-02-04 18:24:43,253 - web_server - INFO - Launching the emhass webserver at: http://0.0.0.0:5000
2024-02-04 18:24:43,253 - web_server - INFO - Home Assistant data fetch will be performed using url: http://supervisor/core/api
2024-02-04 18:24:43,253 - web_server - INFO - The data path is: /share
2024-02-04 18:24:43,255 - web_server - INFO - Using core emhass version: 0.7.3
waitress   INFO  Serving on http://0.0.0.0:5000
2024-02-04 18:24:54,031 - web_server - INFO - Setting up needed data
2024-02-04 18:24:54,072 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2024-02-04 18:24:55,189 - web_server - INFO - Retrieving data from hass for load forecast using method = mlforecaster
2024-02-04 18:24:55,190 - web_server - INFO - Retrieve hass get data method initiated...
2024-02-04 18:24:58,017 - web_server - INFO -  >> Performing dayahead optimization...
2024-02-04 18:24:58,017 - web_server - INFO - Performing day-ahead forecast optimization
2024-02-04 18:24:58,026 - web_server - INFO - Perform optimization for the day-ahead
2024-02-04 18:24:58,033 - web_server - ERROR - Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1463, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 872, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 870, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 855, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/web_server.py", line 67, in action_call
    opt_res = dayahead_forecast_optim(input_data_dict, app.logger)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/command_line.py", line 234, in dayahead_forecast_optim
    opt_res_dayahead = input_data_dict['opt'].perform_dayahead_forecast_optim(
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/optimization.py", line 574, in perform_dayahead_forecast_optim
    self.opt_res = self.perform_optimization(df_input_data, P_PV.values.ravel(),
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/optimization.py", line 288, in perform_optimization
    self.logger.debug("Deferrable load {}: Proposed optimization window: {} --> {}".format(k, def_start_timestep[k], def_end_timestep[k]))
                                                                                              ~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
2024-02-04 18:25:00,195 - web_server - INFO - Setting up needed data
[... 297 lines cut ...]
  ],
  "set_use_battery": false,
  "battery_nominal_energy_capacity": 5000,
  "hass_url": "empty",
  "long_lived_token": "empty",
  "optimization_time_step": 60,
  "historic_days_to_retrieve": 2,
  "method_ts_round": "nearest",
  "lp_solver": "COIN_CMD",
  "lp_solver_path": "/usr/bin/cbc",
  "set_battery_dynamic": false,
  "battery_dynamic_max": 0.9,
  "battery_dynamic_min": -0.9,
  "weight_battery_discharge": 0.0,
  "weight_battery_charge": 0.0,
  "load_forecast_method": "mlforecaster",
  "battery_discharge_power_max": 1000,
  "battery_charge_power_max": 1000,
  "battery_discharge_efficiency": 0.95,
  "battery_charge_efficiency": 0.95,
  "battery_minimum_state_of_charge": 0.3,
  "battery_maximum_state_of_charge": 0.9,
  "battery_target_state_of_charge": 0.6

```

overas, 2024-02-04:
When I run the day-head optimization rest command with emhass 0.6.2 I get in the hassos log:

```
Logger: homeassistant.components.rest_command
Source: components/rest_command/__init__.py:155
Integration: RESTful Command (documentation, issues)
First occurred: 18:53:16 (1 occurrences)
Last logged: 18:53:16

Error. Url: http://localhost:5000/action/dayahead-optim. Status code 500. Payload: b'{ "load_cost_forecast":[1.2218, 1.2162, 1.2132, 1.2069, 1.1557, 1.0437, 0.6484, 0.6323, 0.6183, 0.5514, 0.5802, 0.7232, 1.2777, 1.3927, 1.4696, 1.413, 1.3695, 1.3286, 1.327, 1.3418, 1.3718, 1.3977, 1.4682, 1.579], "prod_price_forecast":[0.74, 0.74, 0.73, 0.73, 0.68, 0.56, 0.17, 0.15, 0.14, 0.07, 0.1, 0.24, 0.68, 0.79, 0.87, 0.81, 0.77, 0.73, 0.73, 0.74, 0.77, 0.8, 0.87, 0.98] }'
```

but with emhass 0.6.1 I do not get this rest command failure. 

Maybe this is a issue after updating the web server?

davidusb-geek, 2024-02-04:
Solved!

overas, 2024-02-07:
Thanks for fast respons

closed 2024-02-07 by overas