davidusb-geek/emhass-add-on #78: ERROR - The ML forecaster file was not found, please run a model fit method before this predict method

overas, 2024-02-26:
Emhass 0.8.0
Hassos 2024.2.3

It seems the ML forecaster file is not stored after a restart of emhass addon. 
Reproduce the error:
1) Restart emhass addon. Everything is ok.
```
2024-02-26 19:13:17,761 - web_server - INFO - Launching the emhass webserver at: http://0.0.0.0:5000
2024-02-26 19:13:17,762 - web_server - INFO - Home Assistant data fetch will be performed using url: http://supervisor/core/api
2024-02-26 19:13:17,762 - web_server - INFO - The data path is: /app/data
2024-02-26 19:13:17,763 - web_server - INFO - Using core emhass version: 0.8.0
waitress   INFO  Serving on http://0.0.0.0:5000
```
2) Run RESTful Command: trigger_forecast

```
  trigger_forecast:
    url: http://localhost:5000/action/dayahead-optim
    method: POST
    content_type: "application/json"
    timeout: 300
    payload: >-
      {
      "load_cost_forecast":{{((state_attr('sensor.nordpool_med_nettleie', 'raw_today') | map(attribute='value') | list  + state_attr('sensor.nordpool_med_nettleie','raw_tomorrow') | map(attribute='value') | list))[now().hour:][:24] }},
      "prod_price_forecast":{{((state_attr('sensor.nordpool_uten_nettleie', 'raw_today') | map(attribute='value') | list  + state_attr('sensor.nordpool_uten_nettleie', 'raw_tomorrow') | map(attribute='value') | list))[now().hour:][:24]}}
      }
```

Here is the log after emhass addon restart and after running emhass forecast. 
```
2024-02-26 19:04:28,808 - web_server - INFO - Launching the emhass webserver at: http://0.0.0.0:5000
2024-02-26 19:04:28,808 - web_server - INFO - Home Assistant data fetch will be performed using url: http://supervisor/core/api
2024-02-26 19:04:28,808 - web_server - INFO - The data path is: /app/data
2024-02-26 19:04:28,809 - web_server - INFO - Using core emhass version: 0.8.0
waitress   INFO  Serving on http://0.0.0.0:5000
2024-02-26 19:04:34,583 - web_server - INFO -  >> Setting input data dict
2024-02-26 19:04:34,584 - web_server - INFO - Setting up needed data
2024-02-26 19:04:34,623 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2024-02-26 19:04:35,642 - web_server - INFO - Retrieving data from hass for load forecast using method = mlforecaster
2024-02-26 19:04:35,643 - web_server - INFO - Retrieve hass get data method initiated...
2024-02-26 19:04:38,879 - web_server - ERROR - The ML forecaster file was not found, please run a model fit method before this predict method
2024-02-26 19:04:38,880 - web_server - ERROR - Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1463, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 872, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 870, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 855, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/web_server.py", line 103, in action_call
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/command_line.py", line 93, in set_input_data_dict
    P_load_forecast = fcst.get_load_forecast(method=optim_conf['load_forecast_method'])
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/forecast.py", line 621, in get_load_forecast
    forecast_out = mlf.predict(data_last_window)
                   ^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'predict'
```

3) Update mlforecast with service: rest_command.ml_forecast_fit and service: rest_command.ml_forecast_tune

```
[... 65 lines cut ...]
2024-02-26 19:21:18,215 - web_server - ERROR - Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1463, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 872, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 870, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 855, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/web_server.py", line 103, in action_call
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/command_line.py", line 93, in set_input_data_dict
    P_load_forecast = fcst.get_load_forecast(method=optim_conf['load_forecast_method'])
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/forecast.py", line 621, in get_load_forecast
    forecast_out = mlf.predict(data_last_window)
                   ^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'predict'
```

overas, 2024-02-26:
Using Emhass 0.6.6 addon I do not get this error 

Here is the log after emhass addon restart and after running emhass forecast.
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun emhass (no readiness notification)
s6-rc: info: service legacy-services successfully started
2024-02-26 20:15:00,867 - web_server - INFO - Launching the emhass webserver at: http://0.0.0.0:5000
2024-02-26 20:15:00,867 - web_server - INFO - Home Assistant data fetch will be performed using url: http://supervisor/core/api
2024-02-26 20:15:00,867 - web_server - INFO - The data path is: /share
2024-02-26 20:15:00,868 - web_server - INFO - Using core emhass version: 0.7.7
waitress   INFO  Serving on http://0.0.0.0:5000
2024-02-26 20:15:05,885 - web_server - INFO - Setting up needed data
2024-02-26 20:15:05,922 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2024-02-26 20:15:06,940 - web_server - INFO - Retrieving data from hass for load forecast using method = mlforecaster
2024-02-26 20:15:06,941 - web_server - INFO - Retrieve hass get data method initiated...
2024-02-26 20:15:10,182 - web_server - INFO -  >> Performing dayahead optimization...
2024-02-26 20:15:10,182 - web_server - INFO - Performing day-ahead forecast optimization
2024-02-26 20:15:10,191 - web_server - INFO - Perform optimization for the day-ahead
2024-02-26 20:15:10,213 - web_server - INFO - Status: Optimal
2024-02-26 20:15:10,213 - web_server - INFO - Total value of the Cost function = -130.68
```

GeoDerp, 2024-02-26:
Mm, this may be because we need to set the data folder to `/data` to be persistent. The restart may be in some way resetting the files inside the docker 🤔

davidusb-geek, 2024-02-27:
We had some similar problems some time ago. The solution was to write temporary files to the share folder in HA, outside the add-on container.

GeoDerp, 2024-02-28:
@overas , this should be fixed in the next version. (0.8.1) 👍

overas, 2024-02-28:
Thanks for the fast fix. Closing case

closed 2024-02-28 by overas