davidusb-geek/emhass-add-on #91: Values published to Home Assistant do not have unique_id and cannot be managed through the UI.

HACS-bank, 2024-04-20:
Using core emhass version: 0.8.6

In the HA user interface, the settings dialog for e.g. sensor.p_pv_forecast can be accessed but produces the message:

_This entity ('sensor.p_pv_forecast') does not have a unique ID, therefore its settings cannot be managed from the UI. See the [documentation](https://www.home-assistant.io/faq/unique_id) for more detail._

This makes it harder for a user to e.g. put the EMHASS data into an "Area" in HA and put onto dashboards etc.

davidusb-geek, 2024-05-01:
How come it is not unique?
In any case, you can change your sensor name using the custom names functionalities (see the docs) and then you will surely have unique ID names.

purcell-lab, 2024-05-11:
UniqueIDs, is a feature of the Home assistant registry for individual entries, which is supposed to be set by the creating integration, in our case this should be EMHASS.

EMHASS doesn't correctly set these uniqueIDs so I would request this issue be reopened pending further investigation.

davidusb-geek, 2024-05-12:
Reopened. I just thought that this was easily solved by a user-provided unique custom ID.
I don't how this can be automatically set within EMHASS. But there is maybe an option that I don't know of.

purcell-lab, 2024-05-13:
unique_id can be registered at the same time as the entity creation:

https://developers.home-assistant.io/docs/entity_registry_index

_Any entity that is added to Home Assistant which specifies the [unique_id attribute](https://developers.home-assistant.io/docs/core/entity#generic-properties) will be registered in the registry._

_Unique ID of last resort
For entities that are setup by a config entry, the Config Entry ID can be used as a last resort if no other Unique ID is available._

davidusb-geek, 2024-05-15:
I don't know if the `POST /api/states/<entity_id>` endpoint accepts the `unique_id` key in the JSON?
It is not listed as an option on the documentation: [https://developers.home-assistant.io/docs/api/rest/](https://developers.home-assistant.io/docs/api/rest/)

GeoDerp, 2024-05-15:
> I don't know if the `POST /api/states/<entity_id>` endpoint accepts the `unique_id` key in the JSON? It is not listed as an option on the documentation: https://developers.home-assistant.io/docs/api/rest/

I agree, I hope this could be change in the future, or separate rest call option to generate a sensor with a `unique_id` first.

purcell-lab, 2024-11-23:
Something similar to https://github.com/bramstroker/homeassistant-powercalc/pull/2708 would be needed to generate the required UUID for EMHASS.

I'm not sure if only integrations can generate UUIDs in this fashion, or if Add-On's can also generate entities in this manner.

johanschelin, 2025-01-16:
I agree this would be nice to have