davidusb-geek/emhass-add-on #93: Error with MPC

RikBast, 2024-05-01:
I get this message quite often:

2024-05-01 19:30:00,116 - web_server - INFO -  >> Setting input data dict
2024-05-01 19:30:00,116 - web_server - INFO - Setting up needed data
2024-05-01 19:30:00,131 - web_server - INFO - Retrieve hass get data method initiated...
2024-05-01 19:30:00,415 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2024-05-01 19:30:01,606 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2024-05-01 19:30:01,606 - web_server - INFO - Retrieve hass get data method initiated...
2024-05-01 19:30:01,950 - web_server - ERROR - Exception on /action/naive-mpc-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1463, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 872, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 870, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 855, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/web_server.py", line 108, in action_call
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/command_line.py", line 129, in set_input_data_dict
    df_input_data_dayahead = utils.set_df_index_freq(df_input_data_dayahead)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/utils.py", line 687, in set_df_index_freq
    df = df.asfreq(sampling)
         ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/frame.py", line 10971, in asfreq
    return super().asfreq(
           ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/generic.py", line 8347, in asfreq
    return asfreq(
           ^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/resample.py", line 2232, in asfreq
    dti = date_range(obj.index.min(), obj.index.max(), freq=freq)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/indexes/datetimes.py", line 945, in date_range
    dtarr = DatetimeArray._generate_range(
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/arrays/datetimes.py", line 446, in _generate_range
    i8values = generate_regular_range(start, end, periods, freq, unit=unit)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/arrays/_ranges.py", line 68, in generate_regular_range
    e = b + (iend - b) // stride * stride + stride // 2 + 1
            ~~~~~~~~~^~~~~~~~~~~
ZeroDivisionError: integer division or modulo by zero

This is my call":

```
  post_mpc_optim: "curl -i -H \"Content-Type: application/json\" -X POST -d '{\"load_cost_forecast\":{{
          (states.sensor.electricity_consumption_price_24h_forecast.state)
          }}, \"prod_price_forecast\":{{
          (states.sensor.electricity_production_price_24h_forecast.state) 
          }}, \"prediction_horizon\":{{min(48,(states.sensor.electricity_consumption_price_24h_forecast.state.split(',')|count))
          }},\"soc_init\":{{states('sensor.battery_batterijpercentage')|float(0)/100}},\"soc_final\":0.9}' http://localhost:5000/action/naive-mpc-optim"
```

Resulting in this:

[image]

davidusb-geek, 2024-05-01:
It is hard to say without more information about your config.
But if your `\"soc_final\":0.9` is always set to that value it can be problematic?
Otherwise your real SOC seems to follow quite well the forecasted SOC, what is the issue in that graph?

davidusb-geek, 2024-05-01:
What is your MPC frequency. This problem may (probably) arise if two MPC calls are really close to each other (< 1 min). This needs testing to confirm.

RikBast, 2024-05-01:
> It is hard to say without more information about your config. But if your `\"soc_final\":0.9` is always set to that value it can be problematic? Otherwise your real SOC seems to follow quite well the forecasted SOC, what is the issue in that graph?

Maybe I understood the concept of soc_final wrong, isn't this the target SOC you want to reach?
Issue with the graph are the prediction spikes

RikBast, 2024-05-01:
> What is your MPC frequency. This problem may (probably) arise if two MPC calls are really close to each other (< 1 min). This needs testing to confirm.

Calls are done every 5 minutes

RikBast, 2024-05-02:
After changing some settings I  still get the error once in a while:

2024-05-02 11:30:00,455 - web_server - INFO - Passed runtime parameters: {'load_cost_forecast': [14.7156, 13.7255, 13.399, 13.4096, 13.9481, 17.3263, 18.9418, 20.129, 21.2759, 21.5261, 20.5063, 19.8385], 'prod_price_forecast': [0.336, -0.598, -0.906, -0.896, -0.388, 2.799, 4.323, 5.443, 6.525, 6.761, 5.799, 5.169], 'prediction_horizon': 12, 'alpha': 1, 'beta': 0, 'soc_init': 0.28, 'soc_final': 0.12}
2024-05-02 11:30:00,457 - web_server - INFO -  >> Setting input data dict
2024-05-02 11:30:00,457 - web_server - INFO - Setting up needed data
2024-05-02 11:30:00,476 - web_server - INFO - Retrieve hass get data method initiated...
2024-05-02 11:30:00,751 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2024-05-02 11:30:01,796 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2024-05-02 11:30:01,797 - web_server - INFO - Retrieve hass get data method initiated...
2024-05-02 11:30:02,310 - web_server - ERROR - Exception on /action/naive-mpc-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1463, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 872, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 870, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 855, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/web_server.py", line 108, in action_call
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/command_line.py", line 129, in set_input_data_dict
    df_input_data_dayahead = utils.set_df_index_freq(df_input_data_dayahead)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/utils.py", line 687, in set_df_index_freq
    df = df.asfreq(sampling)
         ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/frame.py", line 10971, in asfreq
    return super().asfreq(
           ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/generic.py", line 8347, in asfreq
    return asfreq(
           ^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/resample.py", line 2232, in asfreq
    dti = date_range(obj.index.min(), obj.index.max(), freq=freq)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/indexes/datetimes.py", line 945, in date_range
    dtarr = DatetimeArray._generate_range(
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/arrays/datetimes.py", line 446, in _generate_range
    i8values = generate_regular_range(start, end, periods, freq, unit=unit)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/arrays/_ranges.py", line 68, in generate_regular_range
    e = b + (iend - b) // stride * stride + stride // 2 + 1
            ~~~~~~~~~^~~~~~~~~~~
ZeroDivisionError: integer division or modulo by zero
2024-05-02 11:30:02,325 - web_server - INFO - Passed runtime parameters: {}
2024-05-02 11:30:02,325 - web_server - INFO -  >> Setting input data dict
2024-05-02 11:30:02,325 - web_server - INFO - Setting up needed data
2024-05-02 11:30:02,327 - web_server - INFO -  >> Publishing data...
2024-05-02 11:30:02,327 - web_server - INFO - Publishing data to HASS instance
2024-05-02 11:30:02,338 - web_server - INFO - Successfully posted to sensor.p_pv_forecast = 4033.28
2024-05-02 11:30:02,341 - web_server - INFO - Successfully posted to sensor.p_load_forecast = 228.65
2024-05-02 11:30:02,344 - web_server - INFO - Successfully posted to sensor.p_deferrable0 = 0.0
2024-05-02 11:30:02,347 - web_server - INFO - Successfully posted to sensor.p_batt_forecast = -3804.63
2024-05-02 11:30:02,351 - web_server - INFO - Successfully posted to sensor.soc_batt_forecast = 49.68
2024-05-02 11:30:02,354 - web_server - INFO - Successfully posted to sensor.p_grid_forecast = 0.0
2024-05-02 11:30:02,357 - web_server - INFO - Successfully posted to sensor.total_cost_fun_value = 94.26
2024-05-02 11:30:02,359 - web_server - INFO - Successfully posted to sensor.optim_status = Optimal
2024-05-02 11:30:02,362 - web_server - INFO - Successfully posted to sensor.unit_load_cost = 13.7255
2024-05-02 11:30:02,365 - web_server - INFO - Successfully posted to sensor.unit_prod_price = -0.598

davidusb-geek, 2024-05-02:
> isn't this the target SOC you want to reach?

That's what it is yes.

> Calls are done every 5 minutes

It should be fine to call MPC every 5 mins.

I have not seen this type of error before, we need to figure out a way to reproduce this.
Just a guess about databases. How are you managing your HA database/recorder?
Are you using the include entities option of the recorder to list the entities that you want to keep in the database? If yes, is the sensor `sensor.power_load_no_var_loads` included in that list?

RikBast, 2024-05-02:
Hi, all is on default for the HA database/recorder. I am runing HA almost for a month now,
sensor_power_load_no_var_loads has about 25 days of history only. But I see a lot of spikes (also < 0)

[image]

davidusb-geek, 2024-05-10:
Those spikes in the sensor are probably the source of the problems. Try to use some filtering before building the sensor. 
Up until Apr22 it was a nice smooth curve with no spikes, what changed there?

gieljnssns, 2024-05-11:
I think, looking at that picture, before 22-4 the data comes from the long term statistics db.

gieljnssns, 2024-05-11:
> But I see a lot of spikes (also < 0)

You should check your sensor from not going negative
```yaml
- trigger:
    - platform: time_pattern
      seconds: "/10"
  sensor:
    - name: Huidig verbruik zonder wp
      unique_id: 664ab9ff-3875-4a45-b53a-685b7a8f7f96
      unit_of_measurement: "W"
      device_class: power
      state: >
        {% if (states("sensor.huidig_verbruik") | float(default=0) -
            states("sensor.wasmachien_vermogen") | float(default=0) -
            states("sensor.droogkast_vermogen") | float(default=0) -
            states("sensor.warmtepompboiler_huidig_verbruik") | float(default=0) -
            states("sensor.warmtepomp_huidig_verbruik") | float(default=0) -
            states("sensor.afwasmachien_vermogen") | float(default=0)) | round(1) > 0 %}
            {{ (states("sensor.huidig_verbruik") | float(default=0) -
                states("sensor.wasmachien_vermogen") | float(default=0) -
                states("sensor.droogkast_vermogen") | float(default=0) -
                states("sensor.warmtepompboiler_huidig_verbruik") | float(default=0) -
                states("sensor.warmtepomp_huidig_verbruik") | float(default=0) -
                states("sensor.afwasmachien_vermogen") | float(default=0)) | round(1)}}
        {% else  %}
            {{ 0 | float(default=0) }}
        {% endif %}
      availability: >
        {{ is_number(states('sensor.huidig_verbruik')) }}
```

RikBast, 2024-05-11:
> I think, looking at that picture, before 22-4 the data comes from the long term statistics db.

Yes, correct, I just had a look

RikBast, 2024-05-11:
> You should check your sensor from not going negative
>
> ```yaml
> - trigger:
>     - platform: time_pattern
>       seconds: "/10"
>   sensor:
>     - name: Huidig verbruik zonder wp
>       unique_id: [...]

Thank you, I will try to implement this, but still wondering the cause for going negative. I don't see it in the Huawei data.

gieljnssns, 2024-05-11:
this could happen when the sensors you use in your template don't update on the same time.

RikBast, 2024-05-11:
> this could happen when the sensors you use in your template don't update on the same time.

I add this to configuration.yaml I presume?

gieljnssns, 2024-05-11:
What do you wan't to add?

RikBast, 2024-05-11:
> What do you wan't to add?
> 
> Paste this in your developer tools
> 
> ```yaml
> {{states('sensor.inverter_werkelijk_vermogen') | float(default=0) - 
>     states('sensor.power_meter_vermogen') | float(default=0) - 
>     states('sensor.waskeuken_wasdroger_waskeuken_618_power') | float(default=0) - 
>     states('sensor.waskeuken_wasmachine_waskeuken_611_power') | float(default=0)}}
> {% if (states('sensor.inverter_werkelijk_vermogen') | float(default=0) - 
>         states('sensor.power_meter_vermogen') | float(default=0) - 
>         states('sensor.waskeuken_wasdroger_waskeuken_618_power') | float(default=0) - 
>         states('sensor.waskeuken_wasmachine_waskeuken_611_power') | float(default=0)) > 0 %}
>         {{states('sensor.inverter_werkelijk_vermogen') | float(default=0) - 
>           states('sensor.power_meter_vermogen') | float(default=0) - 
>           states('sensor.waskeuken_wasdroger_waskeuken_618_power') | float(default=0) - 
>           states('sensor.waskeuken_wasmachine_waskeuken_611_power') | float(default=0)}}
> {% else %}
>   0
> {% endif %}
> {{states('sensor.inverter_werkelijk_vermogen')}}
> {{states('sensor.power_meter_vermogen')}}
> {{states('sensor.waskeuken_wasdroger_waskeuken_618_power')}}
> {{states('sensor.waskeuken_wasmachine_waskeuken_611_power')}}
> ```

In developers tools --> Templates it nicely calculates the power_no_var_loads:

247.0

        247.0

244
-5
0.7
1.3

RikBast, 2024-05-11:
> What do you wan't to add?
>
> Paste this in your developer tools
>
> ```yaml
> {{states('sensor.inverter_werkelijk_vermogen') | float(default=0) -
>     states('sensor.power_meter_vermogen') | [...]

Sorry for asking, but how do I implement this constraint? I am new to HA so trying to find my way around the system.
Thank you

gieljnssns, 2024-05-11:
Where have you defined your `sensor.power_no_var_loads`?

RikBast, 2024-05-11:
> Where have you defined your `sensor.power_no_var_loads`? And what is `sensor.inverter_werkelijk_vermogen`? Is this the sensor that holds your household power consumption? `sensor.power_meter_vermogen` is this one of your deferrable loads?
> 
> Your `sensor.power_load_no_var_loads` should be, the household power consumption sensor in Watts from Home Assistant. The deferrable loads that we will want to include in the optimization problem should be substracted from this sensor in HASS.

I defined sensor.power_no_var_loads as a template helper:

[image]

[image]

State Template:

```
{{(states('sensor.inverter_werkelijk_vermogen') | float(default=0) - states('sensor.power_meter_vermogen') | float(default=0)) - states('sensor.waskeuken_wasdroger_waskeuken_618_power') | float(default=0) - states('sensor.waskeuken_wasmachine_waskeuken_611_power') | float(default=0)}}
```

sensor.inverter_werkelijk_vermogen comes from the Huawei inverter, senso.power_meter_vermogen comes for the Huawei power meter. I use the Huawei integration available in HA.

gieljnssns, 2024-05-11:
But `sensor.inverter_werkelijk_vermogen` is your PV generation?

RikBast, 2024-05-11:
> But `sensor.inverter_werkelijk_vermogen` is your PV generation? And what is `sensor.power_meter_vermogen` exactly measuring?

The both measure the active power.

```
sensor.inverter_werkelijk_vermogen
```

[image]

```
sensor.power_meter_vermogen
```

[image]

The difference between the two = house hold power

gieljnssns, 2024-05-11:
Then I would try to change the state template to
```
{% if (states('sensor.inverter_werkelijk_vermogen') | float(default=0) - 
        states('sensor.power_meter_vermogen') | float(default=0) - 
        states('sensor.waskeuken_wasdroger_waskeuken_618_power') | float(default=0) - 
        states('sensor.waskeuken_wasmachine_waskeuken_611_power') | float(default=0)) > 0 %}
        {{states('sensor.inverter_werkelijk_vermogen') | float(default=0) - 
          states('sensor.power_meter_vermogen') | float(default=0) - 
          states('sensor.waskeuken_wasdroger_waskeuken_618_power') | float(default=0) - 
          states('sensor.waskeuken_wasmachine_waskeuken_611_power') | float(default=0)}}
{% else %}
  0
{% endif %}
```

RikBast, 2024-05-11:
> {% if (states('sensor.inverter_werkelijk_vermogen') | float(default=0) -
>         states('sensor.power_meter_vermogen') | float(default=0) - [...]

OK, I changed it, works it seems. Thank you for the support

RikBast, 2024-05-12:
I sometime still get an error:

```
2024-05-12 09:30:00,354 - web_server - INFO - Passed runtime parameters: {'load_cost_forecast': [10.7575, 1.6267, 0.6949, -1.3085, -0.8982, 4.3307, 10.324, 13.3217, 15.7766, 18.8019, 21.7794, 19.8057, 18.4478, 17.4514], 'prod_price_forecast': [-3.398, -12.012, -12.891, -14.781, -14.394, -9.461, -3.807, -0.979, 1.337, 4.191, 7.0, 5.138, 3.857, 2.917], 'load_power_forecast': [297, 200, 400, 200, 800, 2200, 300, 200, 1500, 200, 2800, -2100, 0, 200, 200, 300, 300, 400, 200, 200, 1500, 200, 200, 400, 500, 500, 400, 500, 300, 300, 500, 400, 400, 500, 300, 200, 200, 300, 200, 200, 300, 500, 400, 100, 300, 400, 300, 300], 'prediction_horizon': 14, 'alpha': 1, 'beta': 0, 'soc_init': 0.22, 'soc_final': 0.12}
2024-05-12 09:30:00,354 - web_server - INFO -  >> Setting input data dict
2024-05-12 09:30:00,354 - web_server - INFO - Setting up needed data
2024-05-12 09:30:00,363 - web_server - INFO - Retrieve hass get data method initiated...
2024-05-12 09:30:00,676 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2024-05-12 09:30:01,742 - web_server - ERROR - Exception on /action/naive-mpc-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1463, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 872, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 870, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 855, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/web_server.py", line 108, in action_call
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/command_line.py", line 128, in set_input_data_dict
    df_input_data_dayahead = pd.concat([P_PV_forecast, P_load_forecast], axis=1)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/reshape/concat.py", line 385, in concat
    return op.get_result()
           ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/reshape/concat.py", line 592, in get_result
    df = cons(data, index=index, copy=self.copy)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/frame.py", line 709, in __init__
    mgr = dict_to_mgr(data, index, columns, dtype=dtype, copy=copy, typ=manager)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/internals/construction.py", line 481, in dict_to_mgr
    return arrays_to_mgr(arrays, columns, index, dtype=dtype, typ=typ, consolidate=copy)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/internals/construction.py", line 120, in arrays_to_mgr
    arrays, refs = _homogenize(arrays, index, dtype)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/internals/construction.py", line 589, in _homogenize
    val = val.reindex(index, copy=False)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/series.py", line 4918, in reindex
    return super().reindex(
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/generic.py", line 5360, in reindex
    return self._reindex_axes(
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/generic.py", line 5375, in _reindex_axes
    new_index, indexer = ax.reindex(
                         ^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/indexes/base.py", line 4275, in reindex
    raise ValueError("cannot reindex on an axis with duplicate labels")
ValueError: cannot reindex on an axis with duplicate labels
2024-05-12 09:30:01,749 - web_server - INFO - Passed runtime parameters: {}
2024-05-12 09:30:01,750 - web_server - INFO -  >> Setting input data dict
2024-05-12 09:30:01,750 - web_server - INFO - Setting up needed data
2024-05-12 09:30:01,750 - web_server - INFO -  >> Publishing data...
2024-05-12 09:30:01,750 - web_server - INFO - Publishing data to HASS instance
2024-05-12 09:30:01,756 - web_server - INFO - Successfully posted to sensor.p_pv_forecast = 2667.34
2024-05-12 09:30:01,759 - web_server - INFO - Successfully posted to sensor.p_load_forecast = 300
2024-05-12 09:30:01,763 - web_server - INFO - Successfully posted to sensor.p_deferrable0 = 0.0
2024-05-12 09:30:01,766 - web_server - INFO - Successfully posted to sensor.p_deferrable1 = 2500.0
2024-05-12 09:30:01,768 - web_server - INFO - Successfully posted to sensor.p_batt_forecast = 132.66
[... cut ...]
2024-05-12 09:30:01,776 - web_server - INFO - Successfully posted to sensor.p_grid_forecast = 0.0
2024-05-12 09:30:01,779 - web_server - INFO - Successfully posted to sensor.total_cost_fun_value = 79.19
2024-05-12 09:30:01,781 - web_server - INFO - Successfully posted to sensor.optim_status = Optimal
2024-05-12 09:30:01,784 - web_server - INFO - Successfully posted to sensor.unit_load_cost = 1.6267
2024-05-12 09:30:01,787 - web_server - INFO - Successfully posted to sensor.unit_prod_price = -12.012
```

closed 2024-10-31 by davidusb-geek