davidusb-geek/emhass-add-on #96: Unable to build docker add-on image localy

davidusb-geek, 2024-05-12:
Hi @GeoDerp,
While preparing and testing for the new release v0.9.0 I'm not able to build the docker add-on image locally.
I followed the procedure on `Test.md` which is actually the same exact procedure that I typically do to test locally excepet for the final part. To launch the build I just try to install the add-on as a local add-on on my HA instance.
When doing this the build fails. Could you please try to see if you have the same problem and try to debug this?
Here is the error message on build:
```
Logger: homeassistant.components.hassio
Source: components/hassio/websocket_api.py:135
integration: Home Assistant Supervisor (documentation, issues)
First occurred: 11:04:25 AM (3 occurrences)
Last logged: 11:21:30 AM

Failed to to call /addons/local_emhass/install - The command '/bin/bash -o pipefail -c [[ "${BUILD_ARCH}" == "armhf" || "${BUILD_ARCH}" == "armv7" ]] && pip3 install --index-url=https://www.piwheels.org/simple --no-cache-dir --break-system-packages -r requirements.txt || pip3 install --no-cache-dir --break-system-packages -r requirements.txt' returned a non-zero code: 1
```

GeoDerp, 2024-05-12:
> Hi @GeoDerp,
> While preparing and testing for the new release v0.9.0 I'm not able to build the docker add-on image locally.
> I followed the procedure on `Test.md` which is actually the same exact [...]

Sure. I'll have a look now.

GeoDerp, 2024-05-12:
build_from: #choose between debian or raspbian images
  amd64: ghcr.io/home-assistant/amd64-base-debian:bookworm
  armhf: ghcr.io/home-assistant/armhf-base-raspbian:bookworm
  armv7: ghcr.io/home-assistant/armv7-base-debian:bookworm
  aarch64: ghcr.io/home-assistant/aarch64-base-debian:bookworm
args:
  build_version: addon
  build_repo: https://github.com/davidusb-geek/emhass.git #addon-git mode
  build_branch: master #addon-git mode
codenotary:
  signer: notary@home-assistant.io
  base_image: notary@home-assistant.io

@davidusb-geek 
besides setting requirements.txt:
```
numpy==1.26.4
scipy==1.12.0
pandas<=2.0.3
pvlib>=0.10.2
protobuf>=3.0.0
pytz>=2021.1
requests>=2.25.1
beautifulsoup4>=4.9.3
h5py==3.11.0
pulp>=2.4
pyyaml>=5.4.1
tables<=3.9.1
skforecast==0.12.0
# web server packages
flask>=2.0.3
waitress>=2.1.1
plotly>=5.6.0
#EMHASS
emhass==0.9.0
#git+https://github.com/davidusb-geek/emhass
```
You may need to just re install it again. For me there was something weird going on with deb.debian.org . But it works now???

davidusb-geek, 2024-05-12:
Ok the procedure actually works fine as usual.
But there was a problem with an outdated requirement.txt file.
This is solved, I was able to test everything fine.

closed 2024-05-12 by davidusb-geek