davidusb-geek/emhass-add-on #101: Deferrable load thermal model testin, issue and ideas

martinarva, 2024-08-28:
First of all, thermal model is a really great addition to EMHASS

As winter is slowly approaching I looked into it and i'm planning to use it to control my heat pump which provides floor heating and hot water.

I got it somewhat working, but found some issues 

- Looks like it does not provide longer than 24h lags. I have nordpool dayahead prices and use dynamic prediction period which can be up to 34 hours. Would be nice if thermal model can do the same.
- A bit more challenging is the fact that heat pump power is not continius. It can differ quite significantly. Main reason is outside temperature.  Maybe at first use same logic as with load power prediction to predict how much power heat pump uses. Next step would be if i could feed in my heat pump power sensor, outdoor temperature sensor and sensor that count's how many hours heat pump has worked. This correlation would give a good data to predict next 24-34 hours of power and time needed for heating. Same can apply to hot water.

purcell-lab, 2024-08-29:
I have started a discussion on the use cases for the thermal model:
https://github.com/davidusb-geek/emhass/discussions/340

I would be interested understanding in your base setup.

martinarva, 2024-08-29:
Moved this discussion to https://github.com/davidusb-geek/emhass/discussions/340

closed 2024-08-29 by martinarva