davidusb-geek/emhass-add-on #103: "Battery settings for the add-on incorrect" in HA Supervisor logs

RudolfRendier, 2024-10-15:
Today I noticed my Home Assistant supervisor is complaining about the settings for EMHASS (as shown via 'su logs') 

Version: 0.10.6 - 2024-07-14

I have the following, as from the docs:
https://emhass.readthedocs.io/en/latest/config.html

Pd_max: The maximum discharge power in Watts. Defaults to 1000.
Pc_max: The maximum charge power in Watts. Defaults to 1000.
eta_disch: The discharge efficiency. Defaults to 0.95.
eta_ch: The charge efficiency. Defaults to 0.95.
Enom: The total capacity of the battery stack in Wh. Defaults to 5000.
SOCmin: The minimum allowable battery state of charge. Defaults to 0.3.
SOCmax: The maximum allowable battery state of charge. Defaults to 0.9.
SOCtarget: The desired battery state of charge at the end of each optimization cycle. Defaults to 0.6.

This file lists the same:
https://github.com/davidusb-geek/emhass-add-on/blob/main/emhass/config_emhass.yaml

This file, however, lists a different set of options:
https://github.com/davidusb-geek/emhass-add-on/blob/main/emhass/config.yml

  set_use_battery: "bool?"
  set_battery_dynamic: "bool?" #optional
  battery_dynamic_max: "float(0.0,1.0)?" #optional
  battery_dynamic_min: "float?" #optional
  weight_battery_discharge: "float?" #optional
  weight_battery_charge: "float?" #optional
  battery_discharge_power_max: "int(0,)?" #optional
  battery_charge_power_max: "int(0,)?" #optional
  battery_discharge_efficiency: "float(0.0,1.0)?" #optional
  battery_charge_efficiency: "float(0.0,1.0)?" #optional
  battery_nominal_energy_capacity: "int(0,)?"
  battery_minimum_state_of_charge: "float(0.0,1.0)?" #optional
  battery_maximum_state_of_charge: "float(0.0,1.0)?" #optional
  battery_target_state_of_charge: "float(0.0,1.0)?" #optional

Probably the result of an oversight in a recent refactor. Thanks for EMHASS!

davidusb-geek, 2024-10-31:
Could you please check with the current fresh release if this problem presists?

RudolfRendier, 2024-10-31:
Yes, it still persists. But currently the yaml still contains all the old settings.
I should probably get rid of those manually.

I removed all the settings and the logline is no longer logged.

closed 2024-11-08 by RudolfRendier

JohanAlvedal, 2024-11-11:
> Yes, it still persists. But currently the yaml still contains all the old settings. I should probably get rid of those manually.
> 
> I removed all the settings and the logline is no longer logged.

Could you please explain in more detail what to do?

RudolfRendier, 2024-11-11:
Go the settings for the EMHASS add-on in Home Assistant, not the EMHASS web ui, but the page where you can start/stop/restart/uninstall and select the options to start on boot/auto update etc.

Click the configuration-tab
Click the 3-dots near the top right. And select 'edit in yaml'

I removed everything but the data_path; I don't use the others.
The setting below are still valid
The add-on doesn't require the hass_url, or a long_lived_token. If you use solcast in EMHASS, leave that in as well.

```schema:
  hass_url: "str?" #optional
  long_lived_token: "password?" #optional
  data_path: "list(default|/data/|/share)?" #optional
  time_zone: "match((\\w+)?(\\/)(\\w+))?" #optional
  Latitude: "float?" #optional
  Longitude: "float?" #optional
  Altitude: "float?" #optional
  solcast_api_key: "password?" #optional
  solcast_rooftop_id: "str?" #optional
  solar_forecast_kwp: "float?" #optional
  server_ip: "list(0.0.0.0|127.0.0.1)?" #optional

JohanAlvedal, 2024-11-11:
> Go the settings for the EMHASS add-on in Home Assistant, not the EMHASS web ui, but the page where you can start/stop/restart/uninstall and select the options to start on boot/auto update etc.
> 
> Click the configuration-tab Click the 3-dots near the top right. And select 'edit in yaml'
> 
> I removed everything but the data_path; I don't use the others. The setting below are still valid The add-on doesn't require the hass_url, or a long_lived_token. If you use solcast in EMHASS, leave that in as well.
> 
> ```
>   hass_url: "str?" #optional
>   long_lived_token: "password?" #optional
>   data_path: "list(default|/data/|/share)?" #optional
>   time_zone: "match((\\w+)?(\\/)(\\w+))?" #optional
>   Latitude: "float?" #optional
>   Longitude: "float?" #optional
>   Altitude: "float?" #optional
>   solcast_api_key: "password?" #optional
>   solcast_rooftop_id: "str?" #optional
>   solar_forecast_kwp: "float?" #optional
>   server_ip: "list(0.0.0.0|127.0.0.1)?" #optional
> ```

Thanks a lot for your help.