davidusb-geek/emhass-add-on #105: Can not choose 'mlforecast' in Load forecast method with the web GUI

overas, 2024-11-03:
``Emhass addon v0.11.2 
Hassos 2024.10.4

When I try to select mlforecast in the web GUI there is no option for it. The choices I have are naive and csv. Is this correct?
[image]

If I manually edit the config.json file to use mlforecast and run a daily optimization everything works.
[image]

```2024-11-03 15:41:28,133 - web_server - INFO - Passed runtime parameters: {'load_cost_forecast': [1.018, 1.025, 1.027, 1.015, 1.009, 1.01, 1.007, 0.98, 1.007, 1.01, 1.01, 1.014, 1.014, 1.024, 1.052, 1.268, 1.529, 1.525, 1.498, 1.318, 1.239, 1.218, 1.202, 1.211], 'prod_price_forecast': [0.61, 0.618, 0.62, 0.607, 0.601, 0.603, 0.6, 0.572, 0.599, 0.603, 0.603, 0.606, 0.607, 0.617, 0.644, 0.741, 2.141, 2.102, 1.829, 0.79, 0.711, 0.69, 0.675, 0.683], 'prediction_horizon': 24, 'delta_forecast': 2}
2024-11-03 15:41:28,133 - web_server - INFO -  >> Setting input data dict
2024-11-03 15:41:28,133 - web_server - INFO - Setting up needed data
2024-11-03 15:41:28,135 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2024-11-03 15:41:29,485 - web_server - INFO - Retrieving data from hass for load forecast using method = mlforecaster
2024-11-03 15:41:29,486 - web_server - INFO - Retrieve hass get data method initiated...
2024-11-03 15:41:31,416 - web_server - DEBUG - Number of ML predict forcast data generated (lags_opt): 60
2024-11-03 15:41:31,416 - web_server - DEBUG - Number of forcast dates obtained: 24
2024-11-03 15:41:31,418 - web_server - INFO -  >> Performing dayahead optimization...
2024-11-03 15:41:31,419 - web_server - INFO - Performing day-ahead forecast optimization
2024-11-03 15:41:31,430 - web_server - INFO - Perform optimization for the day-ahead
2024-11-03 15:41:31,576 - web_server - DEBUG - Deferrable load 0: Proposed optimization window: 0 --> 0
2024-11-03 15:41:31,577 - web_server - DEBUG - Deferrable load 0: Validated optimization window: 0 --> 0
2024-11-03 15:41:31,578 - web_server - WARNING - Solver default unknown, using default
Welcome to the CBC MILP Solver ```

bert1111, 2024-11-10:
Just updated and saw this too. Haven't looked into it further.

overas, 2024-11-10:
Maybe this is going to be fixed with this pull reguest: https://github.com/davidusb-geek/emhass/pull/370

davidusb-geek, 2024-12-24:
We will take a look into this