davidusb-geek/emhass-add-on #106: No subfolder 'emhass' in /share/ when saving outside of the default-folder

RudolfRendier, 2024-11-07:
When opting to save files into /share/ there's no emhass subfolder (as indicated by the explanation)

The description below it does mention such a folder:
"data_path (Default default), selecting /data/ will store data in an enclosed folder, only assessable by the addon itself. Select /share/emhass to store data in the mounted share folder, accessible outside of the addon"

Adapting the yaml to /share/emhass is not accepted.

A few other add-ons saving data there all save it in a subfolder with their own name.

davidusb-geek, 2024-12-24:
Hi, yes the files are just saved directly into the `share` folder.
We probably just need to update that description

davidusb-geek, 2024-12-24:
I've just modified this message.
Closing this issue

closed 2024-12-24 by davidusb-geek