davidusb-geek/emhass-add-on #113: EMHASS add=on configuration

renaatdb, 2025-02-26:
Hello,

Is there a reason why I can't change anything in the configuration page. The save button has no effect.
Also YAML and Default button gives no effect.
versie: v0.12.7
Generic x86-64
EMHASS add-on
Core
2025.2.5
Supervisor
2025.02.1
Operating System
14.2

```
[2025-02-26 11:22:44 +0100] [22] [INFO] Obtaining parameters from config.json:
[2025-02-26 11:23:07 +0100] [22] [INFO] EMHASS server online, serving index.html...
[2025-02-26 11:23:09 +0100] [22] [INFO] serving configuration.html...
[2025-02-26 11:23:09 +0100] [22] [INFO] Obtaining parameters from config.json:
[2025-02-26 11:23:21 +0100] [22] [INFO] EMHASS server online, serving index.html...
[2025-02-26 11:23:23 +0100] [22] [INFO] serving configuration.html...
[2025-02-26 11:23:23 +0100] [22] [INFO] Obtaining parameters from config.json:
[2025-02-26 11:25:00 +0100] [22] [INFO] EMHASS server online, serving index.html...
[2025-02-26 11:25:00 +0100] [22] [INFO]  >> Obtaining params: 
[2025-02-26 11:25:00 +0100] [22] [INFO] Passed runtime parameters: {}
[2025-02-26 11:25:00 +0100] [22] [INFO]  >> Setting input data dict
[2025-02-26 11:25:00 +0100] [22] [INFO] Setting up needed data
[2025-02-26 11:25:00 +0100] [22] [INFO]  >> Publishing data...
[2025-02-26 11:25:00 +0100] [22] [INFO] Publishing data to HASS instance
[2025-02-26 11:25:00 +0100] [22] [ERROR] File not found error, run an optimization task first.
[2025-02-26 11:25:02 +0100] [22] [INFO] serving configuration.html...
[2025-02-26 11:25:02 +0100] [22] [INFO] Obtaining parameters from config.json:
```

davidusb-geek, 2025-02-26:
Hi, I don't see any reason.
This is certainly related to your current setup.
Try clearing the browser cache

renaatdb, 2025-02-26:
Re-installed EMASS

clearing the browser cache does noyt help.

```
[2025-02-26 19:43:53 +0100] [22] [INFO] serving configuration.html...
[2025-02-26 19:43:53 +0100] [22] [INFO] Obtaining parameters from config.json:
[2025-02-26 19:45:00 +0100] [22] [INFO]  >> Obtaining params: 
[2025-02-26 19:45:00 +0100] [22] [INFO] Passed runtime parameters: {}
[2025-02-26 19:45:00 +0100] [22] [INFO]  >> Setting input data dict
[2025-02-26 19:45:00 +0100] [22] [INFO] Setting up needed data
[2025-02-26 19:45:00 +0100] [22] [INFO]  >> Publishing data...
[2025-02-26 19:45:00 +0100] [22] [INFO] Publishing data to HASS instance
[2025-02-26 19:45:00 +0100] [22] [ERROR] File not found error, run an optimization task first.
[2025-02-26 19:45:10 +0100] [22] [INFO] EMHASS server online, serving index.html...
[2025-02-26 19:45:10 +0100] [22] [INFO] The data container dictionary is empty... Please launch an optimization task
[2025-02-26 19:45:14 +0100] [22] [INFO] serving configuration.html...
[2025-02-26 19:45:14 +0100] [22] [INFO] Obtaining parameters from config.json:
[2025-02-26 19:45:58 +0100] [21] [INFO] Handling signal: term
[2025-02-26 19:45:59 +0100] [22] [INFO] Worker exiting (pid: 22)
[2025-02-26 19:45:59 +0100] [21] [INFO] Shutting down: Master
   Building emhass @ file:///app
      Built emhass @ file:///app
Uninstalled 1 package in 4ms
Installed 1 package in 1ms
[2025-02-26 19:46:02 +0100] [22] [INFO] Starting gunicorn 23.0.0
[2025-02-26 19:46:02 +0100] [22] [INFO] Listening at: http://0.0.0.0:5000 (22)
[2025-02-26 19:46:02 +0100] [22] [INFO] Using worker: gthread
[2025-02-26 19:46:02 +0100] [23] [INFO] Booting worker with pid: 23
[2025-02-26 19:46:10 +0100] [23] [INFO] Obtaining parameters from config.json:
[2025-02-26 19:46:10 +0100] [23] [INFO] Launching the emhass webserver at: http://0.0.0.0:5000
[2025-02-26 19:46:10 +0100] [23] [INFO] Home Assistant data fetch will be performed using url: http://supervisor/core/api
[2025-02-26 19:46:10 +0100] [23] [INFO] The data path is: /data
[2025-02-26 19:46:10 +0100] [23] [INFO] The logging is: INFO
[2025-02-26 19:46:10 +0100] [23] [INFO] Using core emhass version: 0.12.7
[2025-02-26 19:46:50 +0100] [23] [INFO] EMHASS server online, serving index.html...
[2025-02-26 19:46:50 +0100] [23] [INFO] The data container dictionary is empty... Please launch an optimization task
[2025-02-26 19:46:52 +0100] [23] [INFO] serving configuration.html...
[2025-02-26 19:46:53 +0100] [23] [INFO] Obtaining parameters from config.json:
[2025-02-26 19:50:00 +0100] [23] [INFO]  >> Obtaining params: 
[2025-02-26 19:50:00 +0100] [23] [INFO] Passed runtime parameters: {}
[2025-02-26 19:50:00 +0100] [23] [INFO]  >> Setting input data dict
[2025-02-26 19:50:00 +0100] [23] [INFO] Setting up needed data
[2025-02-26 19:50:00 +0100] [23] [INFO]  >> Publishing data...
[2025-02-26 19:50:00 +0100] [23] [INFO] Publishing data to HASS instance
[2025-02-26 19:50:00 +0100] [23] [ERROR] File not found error, run an optimization task first.
```

michael-salzburg, 2025-03-17:
Hi, I had the same issue.
Check the log output (protocol) and look for an emhass error.

I do not remember my error but the solution was to manually fix the config.json (under /share/config.json) and then the error was gone and I was able to save any new configuration (after restart of the add-on)

renaatdb, 2025-03-17:
> Hi, I had the same issue. Check the log output (protocol) and look for an emhass error.
> 
> I do not remember my error but the solution was to manually fix the config.json (under /share/config.json) and then the error was gone and I was able to save any new configuration (after restart of the add-on)

Sorry, I do not understand: config.json where I find this?

michael-salzburg, 2025-03-23:
1) Find the addon Protocol (see Addons) and look for an error at the time you loaded the EMHASS Configuration page. Normally you should see some errors and find the erroneous parameter.

2) Use ssh or similar to get access to the HA shell.
cd into the directory: /share/ and change the wrong initial parameters (you figured out in the protocol) in config.json

trupel, 2025-03-30:
I had just reinstalled EMHASS, I had the same issue with saving as well.

@michael-salzburg pointed me to the right path, and perhaps my previous config.json was not compatible.

Ended up using terminal, browed to the directory /share and replacing the config.json with [https://[raw.githubusercontent.com/davidusb-geek/emhass/refs/heads/master/src/emhass/data/config_defaults.json](https://raw.githubusercontent.com/davidusb-geek/emhass/refs/heads/master/src/emhass/data/config_defaults.json)](url) and now I am able to save via the UI.

Hope this help.

renaatdb, 2025-04-15:
Sorry, I do not understand: config.json where I find this?

closed 2025-12-10 by davidusb-geek